zephyrproject-rtos/zephyr #77847: Nucleo L552ZEQ TFM-IPC HardFault

faintdono, 2024-08-31:
**Describe the bug**
when flash the STM board with TFM IPC samples I got hardfault
```
[INF] Starting bootloader
[INF] Beginning BL2 provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[INF] PSA Crypto init done, sig_type: EC-P256
[INF] Image index: 1, Swap type: none
[INF] Image index: 0, Swap type: none
[INF] Bootloader chainload address offset: 0x14000
[INF] Jumping to the first image slot
Booting TF-M v2.1.0
[INF] Beginning TF-M provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[Sec Thread] Secure image initializing!
Creating an empty ITS flash layout.
Creating an empty PS flash layout.
[INF][PS] Encryption alg: 0x5500200
[INF][Crypto] Init HW accelerator...
[INF][Crypto] Init HW accelerator... complete.
FATAL ERROR: HardFault
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_l552ze_q/stm32l552xx/ns samples/tfm_integration/tfm_ipc/
2. ./build/tfm/api_ns/regression.sh
3. west flash

**Environment (please complete the following information):**

 - OS: Windows 11
 - Zephyr SDK 0.16.8
 - Zephyr 3.7.99

**Additional context**

norrathep, 2024-11-04:
I also encountered this issue and after some debugging, I found that the initialization for malloc allocation somehow causes this problem (not really sure what the root cause is). But if I limit the maximum malloc allocation (e.g., by adding CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=4096 to samples/tfm_integration/tfm_ipc/prj.conf), this hard fault disappears and the sample can run normally.

Surprisingly, this issue doesn't happen for another board I have tried (nRF9160).

erwango, 2024-11-04:
Thanks @norrathep, great catch. It doesn't to be the case on STM32U5 neither..
So by default `CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=-1`, which means all system RAM is used for this area
IIRC my debug session attempt, the hard fault was happening directly in TF-M secure, before jumping in Zephyr NS application. But probably this isn't the case and issue happens by having this area conflicting with MPU/TF-M settings.

erwango, 2024-11-06:
@norrathep I'll go with your proposal for now. It would be better to understand the underlying issue within TF-M, but this would take much more time and I think having a fix in DV4.0 is welcome.

PR #80977 (2024-11-08): boards: st: stm32l5: Fix TF-M by restricting Libc malloc area
By default, libc malloc allocated area is using all available RAM. For some yet unknown reason, this conflicts with TF-M resulting in a Hard Fault before jumping in the non secure application.

Fow now, define a Libc malloc area defined to 2048 which is the default in some other typical applications (ARMv7 targets enabling USERSPACE).

Fixes #77847

Thanks to @norrathep for the hint.
commit b2c9db76 (2024-11-08): boards: st: stm32l5: Fix TF-M by restricting Libc malloc aera

By default, libc malloc allocated area is using all available RAM.
For some yet unknown reason, this conflicts with TF-M resulting in a
Hard Fault before jumping in the non secure application.

For now, define a Libc malloc area defined to 2048 which is the default in
some other typical applications (ARMv7 targets enabling USERSPACE).

Fixes #77847
closed 2024-11-08 by mmahadevan108