zephyrproject-rtos/zephyr #77848: Add STM32CubeCLT to west debug

glenn-andrews, 2024-08-31:
**Is your enhancement proposal related to a problem? Please describe.**
Currently ST products use OpenOCD or J-Link as runners for `west debug`. https://github.com/zephyrproject-rtos/zephyr/pull/75284 proposes changing the default STM32 `west flash` runner from OpenOCD to the STM32CubeProg utility, but does not allow use of the STM32 GDB server.

**Describe the solution you'd like**
`west debug` to support the STM32CubeCLT GDB server 

**Describe alternatives you've considered**
Keeping with OpenOCD for debug - does not suport latest STM32 parts

**Additional context**
STM32 GDB server is available in the STM32CubeCLT package downloadable here:
https://www.st.com/en/development-tools/stm32cubeclt.html

I'm not keen on requiring an ST account or valid email address to download it, but if we approve https://github.com/zephyrproject-rtos/zephyr/pull/75284 then you'll need one for the default flash runner.

JarmouniA, 2024-08-31:
> but does not allow use of the STM32 GDB server

@glenn-andrews I don't think this is accurate, the changes in #75284 won't affect `west debug` behavior which always defaults to OpenOCD when it's available (meaning `support/openocd.cfg` exists).

JarmouniA, 2024-08-31:
@glenn-andrews The title should be "Add ST-LINK GDB Server to west debug" since STM32CubeCLT is just a software bundle that contains many things.

JarmouniA, 2024-08-31:
For more info about ST-LINK GDB server: https://www.st.com/resource/en/user_manual/um2576-stm32cubeide-stlink-gdb-server-stmicroelectronics.pdf

glenn-andrews, 2024-08-31:
> > but does not allow use of the STM32 GDB server
> 
> @glenn-andrews I don't think this is accurate, the changes in #75284 won't affect `west debug` behavior which defaults to OpenOCD or JLink depending on which is included first in `board.cmake`.

You're correct, I meant to say that we can't use the corresponding STM32 GDB because it is not supported. I'll fix it when I'm home and not trying to edit it on the phone.

JarmouniA, 2024-08-31:
> #75284 proposes changing the default STM32 west flash runner from OpenOCD to the STM32CubeProg utility

@glenn-andrews Actually OpenOCD wasn't default, the selection of the runner to be used with `west flash` was really arbitrary & depended on which runner is mentioned first in `board.cmake` of each board, as explained in the PR, it could be pyOCD, STM32CubeProg, OpenOCD or JLink....

glenn-andrews, 2024-08-31:
> @glenn-andrews Actually OpenOCD wasn't default, the selection of the runner to be used with `west flash` was really arbitrary & depended on which runner is mentioned first in `board.cmake` of each [...]

Better?

leonardo-munoz-pragmafw, 2025-05-15:
Hello @glenn-andrews  @JarmouniA .

I'm interested in contributing to this issue. Could you please let me know if it's still active or requires assistance? I'd be glad to help if it's still relevant.

Thank you!

glenn-andrews, 2025-05-16:
As far as I'm concerned I'd be happy for someone knowledgeable to work on it.

I'd like to see the change made, but tooling is beyond my skills.

closed 2025-06-23 by fabiobaltieri