zephyrproject-rtos/zephyr #77849: Add default runners to west config

glenn-andrews, 2024-08-31:
**Is your enhancement proposal related to a problem? Please describe.**
If you do not have the same runner on your board as the default in `board.cmake` you need to supply the alternative each time.

**Describe the solution you'd like**
1. `west config` options to set the default `west flash` and `west debug` runners that override those in `board.cmake`.
2. Supplying a `--runner` option to the `west` command would override the default, as it does for the default in `board.cmake`.
3. Setting the configuration option to `default` (or the option being missing) would revert to the default in `board.cmake`.

**Describe alternatives you've considered**
Different defaults per board would be nice, but hugely complicated to maintain and administer.

**Additional context**
Prompted by https://github.com/zephyrproject-rtos/zephyr/pull/75284# as it changes the runner, and anyone who does not download the ST software will have to add `-runner openocd` to their commands.

henrikbrixandersen, 2024-09-02:
Do you know you can set `-DBOARD_FLASH_RUNNER=...` and/or `-DBOARD_DEBUG_RUNNER=...` when invoking west?

pdgendt, 2024-09-02:
The proposed `west` aliases in https://github.com/zephyrproject-rtos/west/pull/716 can be useful in this case too:

```shell
west config alias.flash "flash --runner=openocd"
west config alias.debug "debug --runner=openocd"
```

glenn-andrews, 2024-09-02:
> The proposed `west` aliases in [zephyrproject-rtos/west#716](https://github.com/zephyrproject-rtos/west/pull/716) can be useful in this case too:
> 
> ```shell
> west config alias.flash "flash --runner=openocd"
> west config alias.debug "debug --runner=openocd"
> ```

That would do nicely

glenn-andrews, 2024-09-02:
> Do you know you can set `-DBOARD_FLASH_RUNNER=...` and/or `-DBOARD_DEBUG_RUNNER=...` when invoking west?

That seems to just move the problem from having to remember to set it during debug to having to remember to set it during build.

I'm thinking of a situation where I'm working in an STM32 shop and in my daily life I'll always be using a specific tool, or I'm a student with an STM32 development board and don't want to remember specify the assignment-recommended runner for every exercise.

In these situations you want a set-it-and-forget it option for 99% of your work, with an easy override for when you need something different.

pdgendt, 2024-09-02:
> Can you override the alias?

What do you mean? You can override a command with an alias, if that is what you mean.

glenn-andrews, 2024-09-02:
> > Can you override the alias?
> 
> What do you mean? You can override a command with an alias, if that is what you mean.

I mean if I did `west config alias.debug "debug --runner=openocd"` but then for just one test needed to use J-Link, could I do `west debug --runner=jlink` or would the alias expand to `west debug --runner=openocd --runner=jlink`and break because of two runners specified?

pdgendt, 2024-09-02:
> I mean if I did `west config alias.debug "debug --runner=openocd"` but then for just one test needed to use J-Link, could I do `west debug --runner=jlink` or would the alias expand to `west debug --runner=openocd --runner=jlink`and break because of two runners specified?

It will expand to `west debug --runner=openocd --runner=jlink` which should actually work.