zephyrproject-rtos/zephyr #77851: samples: hello_world: mimxrt685_evk_cm33: all application boot fails

hakehuang, 2024-09-01:
**Describe the bug**

A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - > mimxrt685_evk_cm33
 - What have you tried to diagnose or workaround this issue?
 - > git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - b8466e0c951b8a159ed0a0465ed831497668197b

```
b8466e0c951b8a159ed0a0465ed831497668197b is the first bad commit
commit b8466e0c951b8a159ed0a0465ed831497668197b
Author: Vit Stanicek <vit.stanicek@nxp.com>
Date:   Mon Aug 5 10:40:04 2024 +0200

    boards: mimxrt685_evk/mimxrt685s/cm33: Enable DMIC

    Enable DMIC clock in soc.c - attach to chip's audio PLL. Add pinmux
    definitions for the DMIC peripheral. Add nodes to SoC's device tree for
    the DMIC peripheral and its audio channels. Configure the DMIC
    peripheral in board's device tree to enable audio capture.

    Signed-off-by: Vit Stanicek <vit.stanicek@nxp.com>

 .../nxp/mimxrt685_evk/mimxrt685_evk-pinctrl.dtsi   | 10 ++++
 .../mimxrt685_evk_mimxrt685s_cm33.dts              | 27 ++++++++-
 dts/arm/nxp/nxp_rt6xx_common.dtsi                  | 66 ++++++++++++++++++++++
 soc/nxp/imxrt/imxrt6xx/cm33/soc.c                  | 12 ++++
 4 files changed, 114 insertions(+), 1 deletion(-)

```

**To Reproduce**

scripts/twister --device-testing --hardware-map /home/ubuntu/nxp/mimxrt685_evk/map.yaml -T samples/hello_world/

**Expected behavior**

test pass

**Impact**

mimxrt685_evk_cm33 boards

**Logs and console output**

no console output for any applications

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.0-1999-g7cf124b4a96

mmahadevan108, 2024-09-18:
@hakehuang, can this be closed?

mmahadevan108, 2024-09-25:
@hakehuang , can this be closed?

hakehuang, 2024-09-26:
I test in latest version it is OK now.

closed 2024-09-26 by hakehuang