zephyrproject-rtos/zephyr #77863: cmake: sca: sparse: hangs on 'too many errors'

golowanow, 2024-09-02:
**Describe the bug**
Build with `-DZEPHYR_SCA_VARIANT=sparse` can't finish hanging after `error: too many errors`

**To Reproduce**
Run west with `-DZEPHYR_SCA_VARIANT=sparse`, for example:
```
west build -p -b qemu_x86 tests/benchmarks/footprints -DZEPHYR_SCA_VARIANT=sparse
```

**Expected behavior**
Build process completed without hanging.

**Impact**
The `sparse` tool can't be used

**Logs and console output**
```
(.venv) foobar:~/zephyrproject/zephyr$ west build -p -b qemu_x86 tests/benchmarks/footprints -DZEPHYR_SCA_VARIANT=sparse
-- west build: making build dir /home/foobar/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/foobar/zephyrproject/zephyr/tests/benchmarks/footprints
-- CMake version: 3.30.2
-- Found Python3: /home/foobar/zephyrproject/.venv/bin/python3 (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/foobar/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/foobar/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: qemu_x86, qualifiers: atom
-- Found host-tools: zephyr 0.16.8 (/home/foobar/zephyr-sdk)
-- Found toolchain: zephyr 0.16.8 (/home/foobar/zephyr-sdk)
-- Found Dtc: /home/foobar/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/foobar/zephyrproject/zephyr/boards/qemu/x86/qemu_x86.dts
-- Generated zephyr.dts: /home/foobar/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/foobar/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/foobar/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/foobar/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/foobar/zephyrproject/zephyr/boards/qemu/x86/qemu_x86_defconfig'
Merged configuration '/home/foobar/zephyrproject/zephyr/tests/benchmarks/footprints/prj.conf'
Configuration saved to '/home/foobar/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/foobar/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/foobar/zephyr-sdk-0.16.8/x86_64-zephyr-elf/x86_64-zephyr-elf/bin/ld.bfd (found version "2.38")
-- Found SCA: sparse (/usr/bin/cgcc)
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/foobar/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CMake Warning at /home/foobar/zephyrproject/zephyr/subsys/random/CMakeLists.txt:12 (message):

      Warning: CONFIG_TIMER_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

-- Using ccache: /usr/bin/ccache
-- Configuring done (5.6s)
-- Generating done (0.1s)
-- Build files have been written to: /home/foobar/zephyrproject/zephyr/build
-- west build: building application
[1/118] Preparing syscall dependency handling

[2/118] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (/home/foobar/zephyrproject/zephyr), build: v3.7.0-1999-g7cf124b4a964
[8/118] Building C object zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj
/home/foobar/zephyrproject/zephyr/include/zephyr/kernel/internal/mm.h:106:1: error: directive in macro's argument list
...............	
[103/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mmu.c.obj
/home/foobar/zephyrproject/zephyr/include/zephyr/kernel/internal/mm.h:106:1: error: directive in macro's argument list
/home/foobar/zephyrproject/zephyr/include/zephyr/kernel/internal/mm.h:108:1: error: directive in macro's argument list
/home/foobar/zephyrproject/zephyr/include/zephyr/kernel/internal/mm.h:109:1: error: directive in macro's argument list
/home/foobar/zephyrproject/zephyr/include/zephyr/kernel/internal/mm.h:112:1: error: directive in macro's argument list
...	
/home/foobar/zephyrproject/zephyr/kernel/mmu.c:305:17: error: Expected ) in function call
/home/foobar/zephyrproject/zephyr/kernel/mmu.c:305:17: error: got :
/home/foobar/zephyrproject/zephyr/kernel/mmu.c:305:17: error: Expected ; at the end of type declaration
/home/foobar/zephyrproject/zephyr/kernel/mmu.c:305:17: error: too many errors
[104/118] Linking C static library zephyr/kernel/libkernel.a
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain:  zephyr SDK 0.16.8
 - Commit SHA or Version used: v3.7.0-1999-g7cf124b4a964

```
(.venv) foobar:~/zephyrproject/zephyr$ cgcc --version     
gcc-9 (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0                                
```

pdgendt, 2024-09-02:
You can override the maximum number of warnings/errors:

```shell
west build -b qemu_x86 tests/benchmarks/footprints -- -DZEPHYR_SCA_VARIANT=sparse -DCMAKE_C_FLAGS="-fmax-errors=100000"
```

pdgendt, 2024-09-02:
> > You can override the maximum number of warnings/errors:
> 
> well, I've already tried to set large `fmax-errors` (and to 0 as well) - without effect.

Which version are you on? It's should be available from `0.6.3` (https://github.com/lucvoo/sparse/commit/635db41c41a436cf0e8d27a5e70898b5332425ed)

pdgendt, 2024-09-02:
Also make sure to add `--` to pass arguments to CMake/sparse.

golowanow, 2024-09-02:
yeah, I have got too old `sparse` 0.6.1 without max-errors.
However, the build process hangs on 'too many errors' and it might do so even with some large max-errors.
I'll experiment with it.

pdgendt, 2024-09-02:
The manpage says it can be set to `'unlimited'`, however I haven't got it to work.

golowanow, 2024-09-02:
With the latest sparse (0196afe16a50c76) my problem seems resolved - now Zephyr SCA build works without hanging at my dev. environment.

closed 2024-09-02 by golowanow