zephyrproject-rtos/zephyr #77888: stm32: FMC clock is not enabled

ts4ling, 2024-09-02:
**Describe the bug**

I have a custom board with a STM32H562 and it has a display using the FMC. When I use CubeMX I can make it work, but with Zephyr not yet. Luckily I have 2 boards so it's easier to compare.

For both cases it comes down to calling [HAL_SRAM_Init()](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32h5xx/drivers/src/stm32h5xx_hal_sram.c), which calls `HAL_SRAM_MspInit(hsram);`. It's a weak function and CubeMX re-implements it with GPIO stuff **and** `__HAL_RCC_FMC_CLK_ENABLE();`. The Zephyr version enters the weak function, does nothing (it contains nothing) and leaves. It then continues to call `FMC_NORSRAM_Init(hsram->Instance, &(hsram->Init));` and all those other function, but the disabled clock prevents changes to the FMC-registers.

When I enable `FMCEN` within `RCC_AHB4ENR` by hand, before `FMC_NORSRAM_Init(..);` is called, I see pins toggle and the `FMC`-registers contain other values than `0x0`.

It looks like a HAL-issue, but as [hal_stm32](https://github.com/zephyrproject-rtos/hal_stm32) has no issues, I opened one here.

**Expected behavior**

FMC clock should be enabled.

**Impact**

No FMC output.

**Environment (please complete the following information):**

 - OS: Linux
 - Commit SHA or Version used: latest

**Additional context**

My devicetree snippet

```dts
&fmc {
	pinctrl-0 = <&fmc_a15_pg5 &fmc_ne3_pg6 &fmc_nwe_pa9 &fmc_noe_pd4 
						&fmc_d0_pd14 &fmc_d1_pd15 &fmc_d2_pd0 &fmc_d3_pd1
						&fmc_d4_pe7 &fmc_d5_pe8 &fmc_d6_pe9 &fmc_d7_pe10
						&fmc_d8_pe11 &fmc_d9_pe12 &fmc_d10_pe13 &fmc_d11_pe14
						&fmc_d12_pe15 &fmc_d13_pd8 &fmc_d14_pd9 &fmc_d15_pd10>;
	pinctrl-names = "default";
	status = "okay";

	sram {
		compatible = "st,stm32-fmc-nor-psram";//fmc-nor-psram is the correct device
		#address-cells = <1>;
		#size-cells = <0>;

		bank@0 {
			reg = <0x0>;
			st,control = <STM32_FMC_DATA_ADDRESS_MUX_DISABLE
				STM32_FMC_MEMORY_TYPE_SRAM
				STM32_FMC_NORSRAM_MEM_BUS_WIDTH_16
				STM32_FMC_BURST_ACCESS_MODE_DISABLE
				STM32_FMC_WAIT_SIGNAL_POLARITY_LOW
				STM32_FMC_WAIT_TIMING_BEFORE_WS
				STM32_FMC_WRITE_OPERATION_ENABLE
				STM32_FMC_WAIT_SIGNAL_DISABLE
				STM32_FMC_EXTENDED_MODE_DISABLE
				STM32_FMC_ASYNCHRONOUS_WAIT_DISABLE
				STM32_FMC_WRITE_BURST_DISABLE
				STM32_FMC_CONTINUOUS_CLOCK_SYNC_ONLY
				STM32_FMC_WRITE_FIFO_DISABLE
				STM32_FMC_PAGE_SIZE_NONE>;
			st,timing = <3 15 8 0 16 17 STM32_FMC_ACCESS_MODE_A>;
			fmc-mipi-dbi {
				compatible = "st,stm32-fmc-mipi-dbi";
				reset-gpios = <&gpioa 8 GPIO_ACTIVE_LOW>;
//			power-gpios = <&gpioh 0 GPIO_ACTIVE_LOW>;
				register-select-pin = <15>;
				#address-cells = <1>;
				#size-cells = <0>;
				ili9341: lcd-panel@0 {
					compatible = "ilitek,ili9341";
					reg = <0>;
					pixel-format = <ILI9XXX_PIXEL_FORMAT_RGB565>;
					mipi-mode = <MIPI_DBI_MODE_8080_BUS_16_BIT>;  //<-- this mode is currently not supported by the ili9xxx driver
					mipi-max-frequency = <DT_FREQ_M(20)>;
					rotation = <0>;
					width = <320>;
					height = <240>;
				};
			};
		};
	};
};
```

ts4ling, 2024-09-03:
##### CubeMX version

```c
static void HAL_FMC_MspInit(void){
  /* USER CODE BEGIN FMC_MspInit 0 */
  /* USER CODE END FMC_MspInit 0 */
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  if (FMC_Initialized) {
    return;
  }
  FMC_Initialized = 1;

  /* Peripheral clock enable */
  __HAL_RCC_FMC_CLK_ENABLE();

  /** FMC GPIO Configuration
  PE7   ------> FMC_D4
  PE8   ------> FMC_D5
  PE9   ------> FMC_D6
  PE10   ------> FMC_D7
  PE11   ------> FMC_D8
  PE12   ------> FMC_D9
  PE13   ------> FMC_D10
  PE14   ------> FMC_D11
  PE15   ------> FMC_D12
  PD8   ------> FMC_D13
  PD9   ------> FMC_D14
  PD10   ------> FMC_D15
  PD14   ------> FMC_D0
  PD15   ------> FMC_D1
  PG5   ------> FMC_A15
  PG6   ------> FMC_NE3
  PA9   ------> FMC_NWE
  PD0   ------> FMC_D2
  PD1   ------> FMC_D3
  PD4   ------> FMC_NOE
  */
  /* GPIO_InitStruct */
  GPIO_InitStruct.Pin = GPIO_PIN_7|GPIO_PIN_8|GPIO_PIN_9|GPIO_PIN_10
                          |GPIO_PIN_11|GPIO_PIN_12|GPIO_PIN_13|GPIO_PIN_14
                          |GPIO_PIN_15;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.Alternate = GPIO_AF12_FMC;

  HAL_GPIO_Init(GPIOE, &GPIO_InitStruct);

  /* GPIO_InitStruct */
  GPIO_InitStruct.Pin = GPIO_PIN_8|GPIO_PIN_9|GPIO_PIN_10|GPIO_PIN_14
                          |GPIO_PIN_15|GPIO_PIN_0|GPIO_PIN_1|GPIO_PIN_4;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.Alternate = GPIO_AF12_FMC;

  HAL_GPIO_Init(GPIOD, &GPIO_InitStruct);

  /* GPIO_InitStruct */
  GPIO_InitStruct.Pin = GPIO_PIN_5|GPIO_PIN_6;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.Alternate = GPIO_AF12_FMC;

  HAL_GPIO_Init(GPIOG, &GPIO_InitStruct);

  /* GPIO_InitStruct */
  GPIO_InitStruct.Pin = GPIO_PIN_9;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.Alternate = GPIO_AF12_FMC;

  HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

  /* USER CODE BEGIN FMC_MspInit 1 */
  /* USER CODE END FMC_MspInit 1 */
}
```

##### Zephyr version

```c
__weak void HAL_SRAM_MspInit(SRAM_HandleTypeDef *hsram)
{
  /* Prevent unused argument(s) compilation warning */
  UNUSED(hsram);

  /* NOTE : This function Should not be modified, when the callback is needed,
            the HAL_SRAM_MspInit could be implemented in the user file
   */
}
```

mathieuchopstm, 2024-09-12:
Hi @ts4ling,
it appears that the `fmc` is not enabled in the upstream DTSI files for STM32H5, so I assume you have declared it yourself. However, I do not see this declaration in your report, only the board DTS / overlay file.

The FMC driver is implemented such that it does enable the device clock:
https://github.com/zephyrproject-rtos/zephyr/blob/dc5f1bfb3f2eee3e3775175232d508660f5d6efe/drivers/memc/memc_stm32.c#L61-L65

Since these values come from the `clocks` property of the `fmc` node, could you check in your Device Tree that it looks something like `clocks = <&rcc STM32_CLOCK_BUS_AHB4 (1 << 16)>;` (or `0x00010000` instead of `(1 << 16)`)?

ts4ling, 2024-09-13:
Thanks for the solution!

My `build/zephyr/zephyr.dts` contained

```dts
		fmc: memory-controller@47000400 {
			compatible = "st,stm32-fmc";
			status = "okay";
			reg = < 0x47000400 0x400 >;
			clocks = < &rcc 0x94 0x1000 >;
			pinctrl-0 = < &fmc_a15_pg5 &fmc_ne1_pd7 &fmc_nwe_pa9 &fmc_noe_pd4 &fmc_d0_pd14 &fmc_d1_pd15 &fmc_d2_pd0 &fmc_d3_pd1 &fmc_d4_pe7 &fmc_d5_pe8 &fmc_d6_pe9 &fmc_d7_pe10 &fmc_d8_pe11 &fmc_d9_pe12 &fmc_d10_pe13 &fmc_d11_pe14 &fmc_d12_pe15 &fmc_d13_pd8 &fmc_d14_pd9 &fmc_d15_pd10 >;
			pinctrl-names = "default";
// (..)
```

and I have no idea where `clocks = < &rcc 0x94 0x1000 >;` comes from as it's not part of `<st/h5/stm32h562.dtsi>` or `<st/h5/stm32h562zitx-pinctrl.dtsi>` or any other includes of my DTS file.

I adapted my DTS

```DTS
&fmc {
	pinctrl-0 = <&fmc_a15_pg5 &fmc_ne3_pg6 &fmc_nwe_pa9 &fmc_noe_pd4 
						&fmc_d0_pd14 &fmc_d1_pd15 &fmc_d2_pd0 &fmc_d3_pd1
						&fmc_d4_pe7 &fmc_d5_pe8 &fmc_d6_pe9 &fmc_d7_pe10
						&fmc_d8_pe11 &fmc_d9_pe12 &fmc_d10_pe13 &fmc_d11_pe14
						&fmc_d12_pe15 &fmc_d13_pd8 &fmc_d14_pd9 &fmc_d15_pd10>;
	pinctrl-names = "default";
	clocks = <&rcc STM32_CLOCK_BUS_AHB4 0x00010000>;
	status = "okay";
```

as you suggested and now the `FMCEN`-bit is set automatically within `RCC_AHBENR` 😃

closed 2024-09-13 by ts4ling

erwango, 2024-09-13:
@ts4ling Thanks for closing the feedback and issue. Since you've been able to enable FMC on H5, don't hesitate to contribute it if you have some time, this would be a welcome addition.
It's "just" a matter of:
- updating the correct dts/arm/st/h5/*.h5.dtsi file
- enabling on a board if we have a h5 board upstream that supports it.
- making sure it is tested by ci by enabling an existing sample or a test for this board

Overall a good step into taking part into Zephyr development.