zephyrproject-rtos/zephyr #77890: edtlib: wrong handling of "interrupt-map" property

avolkov-1221, 2024-09-02:
**Describe the bug**

It's looks like some bug presented in the implementation of `edtlib._map_interrupt` function. As result `gen-defines.py` script is crashed on the property handling with a completely incomprehensible error, instead of generating headers.

**To Reproduce**

1. appy attached patch.
2. run west build -b ek_ra2l1
3. See error

**Expected behavior**

The `gen-defines.py`  should pass correctly and generate something resonable. Or generate a clear error message.

**Impact**

Any device tree with `interrupt-map` property.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: any
 - Toolchain Zephyr SDK
 - Commit SHA or Version: 7b4e7ff9eda0227762666a39636b5f7729fdfd51

**Additional context**

[bad_interrupt_map.PATCH](https://github.com/user-attachments/files/16839060/bad_interrupt_map.PATCH)

ydamigos, 2024-09-03:
It seems that 'interrupt-controller' property is missing from `icu` node.

avolkov-1221, 2024-09-03:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/77900

avolkov-1221, 2024-09-03:
> It seems that 'interrupt-controller' property is missing from `icu` node.

These are the mutual exclusion properties: 'interrupt-controller' is a property for the root node, while 'interrupt-map' is for the intermediate one.

commit 871afee4 (2025-06-27): edtlib: fix _raw_unit_addr() function

The '_raw_unit_addr()' function was incorrectly written a long time ago.
It tries to look up and use the '#address-cells' of the node containing
the 'interrupts' property itself, instead of using the '#address-cells' of
the interrupt controller's node (or its parent).

As a result, the mapping is successful only if the '#address-cells' value
is the same in both the node and the interrupt controller, all other cases
lead to random errors being generated by the 'gen_defines.py' script.

Also, the function was trying to use the '#address-cells' value as a
li [... cut]
closed 2025-06-27 by kartben