zephyrproject-rtos/zephyr #77894: stm32: support the drive-strength pinctrl node property

cfriedt, 2024-09-02:
**Is your feature request related to a problem? Please describe.**

This is not at all a bug, it's more of a "nice to have". 

When using some peripheral alternate functions (e.g. i2c), it would be really nice to be able to configure the drive strength via Devicetree (as it's done for e.g. [TI cc13xx-cc26xx](https://github.com/zephyrproject-rtos/zephyr/blob/0492235d0dd84ad468acf84a4e3c97a88d98c700/dts/bindings/pinctrl/ti%2Ccc13xx-cc26xx-pinctrl.yaml#L104))

I believe the GPIO drive strength is configured on ST parts using the following bits from [stm32-pinctrl.h](https://github.com/zephyrproject-rtos/zephyr/blob/0492235d0dd84ad468acf84a4e3c97a88d98c700/include/zephyr/dt-bindings/pinctrl/stm32-pinctrl.h#L95-L101).

```cpp
/* GPIO speed */
#define STM32_OSPEEDR_LOW_SPEED		(0x0 << STM32_OSPEEDR_SHIFT)
#define STM32_OSPEEDR_MEDIUM_SPEED	(0x1 << STM32_OSPEEDR_SHIFT)
#define STM32_OSPEEDR_HIGH_SPEED	(0x2 << STM32_OSPEEDR_SHIFT)
#define STM32_OSPEEDR_VERY_HIGH_SPEED	(0x3 << STM32_OSPEEDR_SHIFT)
#define STM32_OSPEEDR_MASK		0x3
#define STM32_OSPEEDR_SHIFT		7
```

**Describe the solution you'd like**
The ability to specify high drive strength with 1 additional, generic devicetree property. Something like the following:

```
			/* I2C_SCL */

			/omit-if-no-ref/ i2c1_scl_pb6: i2c1_scl_pb6 {
				pinmux = <STM32_PINMUX('B', 6, AF4)>;
				bias-pull-up;
				drive-open-drain;
				drive-strength;
			};
```

**Describe alternatives you've considered**
I think the only way to do this currently is to create a custom pinmux with additional bits or'ed together.

```
			/* I2C_SCL */

			/omit-if-no-ref/ i2c1_scl_pb6: i2c1_scl_pb6 {
				pinmux = <(STM32_PINMUX('B', 6, AF4) | STM32_OSPEEDR_VERY_HIGH_SPEED)>;
				bias-pull-up;
				drive-open-drain;
			};
```

**Additional context**

cfriedt, 2024-09-03:
@erwango - oh wow - I just saw that. Would it make sense to change the TI one to use the same terminology?

cfriedt, 2024-09-03:
@vaishnavachath - Was is already discussed at some point about using a common DT binding for drive strength / slew rate in Zephyr?

erwango, 2024-09-04:
> @erwango - oh wow - I just saw that. Would it make sense to change the TI one to use the same terminology?

Maybe. For the record, we've taken this from Linux.

cfriedt, 2024-09-04:
I would guess the same is true for the TI parts, but I haven't checked.

closed 2024-09-06 by cfriedt