zephyrproject-rtos/zephyr #77902: drivers: uart: Assert in event driven nRF UART driver when using power managament

joerchan, 2024-09-03:
**Describe the bug**
When enabling asserts in a build that otherwise appear to be working fine an assert is seen when enabling dynamic device power management and attempting to send data.
The assert happens in nrfx drivers when starting an already started timer for the HW RX counter.

**To Reproduce**
prj.conf:
```
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y

CONFIG_SERIAL=y

CONFIG_NRFX_UARTE3=y
CONFIG_UART_ASYNC_API=y
CONFIG_UART_0_INTERRUPT_DRIVEN=n
CONFIG_UART_0_ASYNC=y
CONFIG_UART_0_NRF_HW_ASYNC=y
CONFIG_UART_0_NRF_HW_ASYNC_TIMER=0
CONFIG_UART_0_NRF_ASYNC_LOW_POWER=y
CONFIG_NRFX_TIMER0=y

CONFIG_ASSERT=y
```
main.c:
```
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/drivers/uart.h>
#include <zephyr/pm/device.h>
#include <zephyr/pm/device_runtime.h>

#include <string.h>

#define UART_DEVICE_NODE DT_CHOSEN(zephyr_shell_uart)
#define MSG_SIZE 32

static const struct device *const uart_dev = DEVICE_DT_GET(UART_DEVICE_NODE);

void _uart_callback(const struct device *dev, struct uart_event *evt, void *user_data)
{
	/* Ignore */
}

void print_uart(char *buf)
{
	int msg_len = strlen(buf);

	uart_tx(uart_dev, buf, msg_len, 0);
}

int main(void)
{
	if (!device_is_ready(uart_dev)) {
		return 0;
	}

	/* configure interrupt and callback to receive data */
	int ret = uart_callback_set(uart_dev, _uart_callback, NULL);

	if (ret < 0) {
		printk("Error setting UART callback: %d\n", ret);
		return 0;
	}

	pm_device_runtime_enable(uart_dev);
	pm_device_runtime_get(uart_dev);

	print_uart("Hello world!\r\n");

	return 0;
}
```

**Expected behavior**

**Impact**
Can't enable asserts, which as explained here: https://github.com/zephyrproject-rtos/zephyr/issues/71130#issuecomment-2222322648 a lot of functions rely on to fail fast.

**Logs and console output**
```
ASSERTION FAIL [m_cb[p_instance->instance_id].state == NRFX_DRV_STATE_INITIALIZED]
 @ WEST_TOPDIR/modules/hal/nordic/nrfx/drivers/src/nrfx_timer.c:210
```

**Environment (please complete the following information):**
Zephyr v3.7.0 or current main branch.
Linux
Zephyr SDK 0.16.4

nordic-piks, 2024-09-12:
Triaged internally

joerchan, 2024-10-16:
Sorry no. It crashed with the application like this on my own board, but on the DK it did not.
Adding `CONFIG_CONSOLE=n` it crashes on the nrf9160 DK (with NS).

I have updated the description.

nordic-krch, 2024-10-23:
I still cannot reproduce it (on 3.7.0). On which commit are you running it? Maybe anything specific in the overlay?

joerchan, 2024-10-29:
I went back and checked it. I'm not sure anymore what I have reproduced it on earlier.
It happens to me on NCS 2.7.0 release.
I therefore went to zephyr commit ea02b93eea3 which is the last common commit between zephyr and the NCS fork.
It is reproducible there.

nordic-krch, 2024-10-31:
Finally, I managed to reproduce it on that commit. I did bisect and it was fixed by 79c93dd5ecfde7b6af6109672b856bbf230bbda1. I think we can close it.

closed 2025-03-10 by nordic-piks