zephyrproject-rtos/zephyr #77919: tests: flash : common: rt10xx: NXP flash does not support negative operation

hakehuang, 2024-09-03:
**Describe the bug**

test_negative_flash_fill cause hard fault for NXP RT series platforms

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1060_evk
 - What have you tried to diagnose or workaround this issue?
 - this fails when the negative test merged
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - 1a48a0174fd519f08703c1e1c0658c545bab3372

**To Reproduce**

scripts/twister -p mimxrt1060_evk -T tests/drivers/flash/common

**Expected behavior**

test pass

**Impact**

flash operation

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-1447-g8617e4684937 ***
Running TESTSUITE flash_driver
===================================================================
Test will run on device is25wp064@0
START - test_flash_erase
 PASS - test_flash_erase in 0.048 seconds
===================================================================
START - test_flash_fill
 PASS - test_flash_fill in 0.042 seconds
===================================================================
START - test_flash_flatten
 PASS - test_flash_flatten in 0.043 seconds
===================================================================
START - test_flash_page_layout
start_offset=0x622000   size=4096       index=1570
page_count=2048
 PASS - test_flash_page_layout in 0.006 seconds
===================================================================
START - test_negative_flash_erase
TEST_FLASH_START = 0x0
TEST_FLASH_SIZE = 0x4000000
 PASS - test_negative_flash_erase in 0.005 seconds
===================================================================
START - test_negative_flash_fill

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:373: flash_driver_test_negative_flash_fill: (rc < 0 is false)
Invalid use of flash_fill returned 0
 FAIL - test_negative_flash_fill in 0.016 seconds
===================================================================
START - test_negative_flash_flatten
 PASS - test_negative_flash_flatten in 0.001 seconds
===================================================================
START - test_negative_flash_read
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x5fffffff
E: r0/a1:  0x80004ab0  r1/a2:  0x5fffffff  r2/a3:  0x600001ff
E: r3/a4:  0x80004aaf r12/ip:  0x800006f1 r14/lr:  0x8000070d
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x6000c5ac
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80004100 (test_negative_flash_read)
E: Halting system
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.0-1447-g8617e4684937

**Additional context**

due to #77895 please reproduce this issue with v3.7.0-1447-g8617e4684937

danieldegrasse, 2024-10-31:
@hakehuang I'm unable to reproduce this issue on cdb9166b810cb7b663: 

```
TESTSUITE flash_driver succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [flash_driver]: pass = 6, fail = 0, skip = 0, total = 6 duration = 0.321 seconds
 - PASS - [flash_driver.test_flash_copy] duration = 0.102 seconds
 - PASS - [flash_driver.test_flash_erase] duration = 0.060 seconds
 - PASS - [flash_driver.test_flash_fill] duration = 0.049 seconds
 - PASS - [flash_driver.test_flash_flatten] duration = 0.065 seconds
 - PASS - [flash_driver.test_flash_page_layout] duration = 0.026 seconds
 - PASS - [flash_driver.test_read_unaligned_address] duration = 0.019 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

Can you check on your side?

hakehuang, 2024-11-01:
> Can you check on your side?

they move this to another test and only Nordic platforms allowed ffd065f93e535d4cd5849e1f17a15717ce7ad173

closed 2024-11-01 by hakehuang