zephyrproject-rtos/zephyr #77922: Add function to get the actual clock frequency in the SPI (and other communucations) API.

flabou, 2024-09-03:
I've been fiddling a lot with SPI lately, and I understand that the `spi-max-frequency` (or the frequency defined in the `spi_config` `frequency` member) is the maximum frequency that can be used by an SPI device, but the actual frequency that will be used can be MUCH lower than that. I think this is driver specific, and there is no clear rule for it. But for instance, I've seen the actual speed can be half the `spi-max-frequency` field. This makes sense I guess because of the prescallers being powers of 2.

Unfortunately, there is no feedback to know the actual used frequency. I only know it can be halved because I used a logic analyzer and an oscilloscope. But there are cases where not being able to reach a minimum frequency would prevent the application from being effective. For example if it's necessary to read high frequency data from an external ADC and a minimum sample rate is required.

Since the driver is responsible for setting the frequency lower than `spi-max-frequency`, one would assume that it would be aware of the actual frequency used. 

So wouldn't we be good to have a extra API function to read the actual frequency? 
Alternatively, maybe the driver could also update the `spi_config` `frequency`  member to the correct value, but I guess it's easier to forget implementing that in a driver than a given function.

This would also make debugging easier than with oscilloscope, even though it's always better to confirm as there can be other clock configuration issues up stream.

I'm mentioning SPI because that is what I am using now, but I think it's also valid for other communication interfaces.

jeppenodgaard, 2024-09-09:
I also encountered issues because I did not consider the potentially large clock frequency difference. 
I thought about maybe allow/require either spi-max-frequency or spi-exact-frequency, but I am not sure I like the increased code complexity.
I also considered a log warning if the actual clock does not match spi-max-frequency.

flabou, 2024-09-16:
I don't think a log warning would be good because it would almost always hit. The set of possible frequencies is usually pretty small compared to the configuration of `spi-max-frequency` which can be any integer from 1 to several MHz value. The actual frequency depends on the source clock configuration and how some register that will often configure a prescaler. The name `spi-max-frequency` makes a lot of sense in that regard.

I think a way to be able to read the actual frequency, and maybe clearly indicating in the documentation that `spi-max-frequency` and the the member `frequency` of `spi_config` are only aspirational, would be good.

jeppenodgaard, 2024-09-18:
> I think a way to be able to read the actual frequency [...]

How would you use this function / what is your use case for reading it programmatically?

My suggestions is based on the LED strip SPI driver, which needs a precise clock since the bit time needs to be known. However the usage of SPI in that driver is probably bit abnormal.

But the `spi-max-frequency` might seem to have cause some confusion there.
E.g. `nucleo_h743zi` uses a `pll1_q_ck` as base clock which runs at 480 MHz. The SPI clock will then be 3.75 MHz (480 MHz / 128 prescaler), and not 4 MHz.

nucleo_h743zi.overlay:
```
&arduino_spi {
	led_strip: ws2812@0 {
...
		spi-max-frequency = <DT_FREQ_M(4)>;
....
        };
};

```

flabou, 2025-01-03:
Sorry for very late reply. 
The use case is that you can read back the actual clock programmatically, and then you do whatever you want with it. For instance, the user can decide to trigger a log message if the clock is 10% lower than intended. In many case, this is more useful at the development / debug stage.

tbursztyka, 2025-01-04:
the frequency being set in spi_config **MUST** be followed by the SPI controller driver. Unless it cannot support it, then it would need to return -EINVAL.

If you get a different frequency than requested: that's a bug in the controller driver. 

spi-max-frequency is, afaik, the only field we could use to express the frequency we want to set (note: this field if not invented for zephyr, it has been existing before in linux, thus the reuse). In any case, it should not be interpreted as: it is a maximum, but something/somewhere can alter it. Because there is nothing that enables that behavior anywhere. What is configured at build time must be used as-is at run time.