zephyrproject-rtos/zephyr #77956: [Backport v3.6-branch] Failed to backport #74949

zephyrbot, 2024-09-03:
This issue was created automatically because the backport of #74949 to `v3.6-branch` failed.

The backport to `v3.6-branch` failed:
```
Auto-merging lib/utils/utf8.c
[backport-74949-to-v3.6-branch 61e0f024d8e] zephyr: Add zero-len check for utf8_trunc
 Author: Emil Gydesen <emil.gydesen@nordicsemi.no>
 Date: Tue Jun 25 13:08:08 2024 +0200
 1 file changed, 8 insertions(+), 1 deletion(-)
Auto-merging tests/unit/util/main.c
CONFLICT (content): Merge conflict in tests/unit/util/main.c
error: could not apply 1fc67d22d18... tests: unit: util: Add unit test of utf8_trunc and utf8_lcpy
hint: After resolving the conflicts, mark them with
hint: "git add/rm <pathspec>", then run
hint: "git cherry-pick --continue".
hint: You can instead skip this commit with "git cherry-pick --skip".
hint: To abort and get back to the state before "git cherry-pick",
hint: run "git cherry-pick --abort".
hint: Disable this message with "git config advice.mergeConflict false"
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v3.6-branch v3.6-branch
# Navigate to the new working tree
cd .worktrees/backport-v3.6-branch
# Create a new branch
git switch --create backport-74949-to-v3.6-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x 1fc67d22d18c912cecbe49abf9acc4474e8786a6~2..1fc67d22d18c912cecbe49abf9acc4474e8786a6
# Push it to GitHub
git push --set-upstream origin backport-74949-to-v3.6-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v3.6-branch
```
Then, create a pull request where the `base` branch is `v3.6-branch` and the `compare`/`head` branch is `backport-74949-to-v3.6-branch`.

henrikbrixandersen, 2024-09-23:
@Thalley Can this be closed now that https://github.com/zephyrproject-rtos/zephyr/pull/78286 was merged?

Thalley, 2024-09-23:
> @Thalley Can this be closed now that #78286 was merged?

Yes, together with https://github.com/zephyrproject-rtos/zephyr/issues/77957. 

The PR should have closed these issues; not sure why it didn't.

closed 2024-09-23 by Thalley