zephyrproject-rtos/zephyr #77968: [backport]  Bluetooth: RFCOMM: check the validity of received frame

ceolin, 2024-09-03:
Missing validation fixed in #74640 must be backported.

henrikbrixandersen, 2024-09-04:
For some reason, a backport PR was not automatically created.

ceolin, 2024-09-13:
@lylezhu2012 can you take a look please ?

lylezhu2012, 2024-09-18:
> @lylezhu2012 can you take a look please ?

What do I need to do?

henrikbrixandersen, 2024-09-23:
Anybody working on this?

Thalley, 2024-09-23:
> Anybody working on this?

I don't think so. Ideally the maintainer, @lylezhu2012, would take this up, but I don't think the backporting procedure is defined well enough for people unfamiliar with the process to easily do it.

henrikbrixandersen, 2024-09-23:
> I don't think so. Ideally the maintainer, @lylezhu2012, would take this up, but I don't think the backporting procedure is defined well enough for people unfamiliar with the process to easily do it.

Okay. I have created a manual backport PR here: https://github.com/zephyrproject-rtos/zephyr/pull/78860

henrikbrixandersen, 2024-09-26:
Fixed in #78860.

closed 2024-09-26 by henrikbrixandersen