zephyrproject-rtos/zephyr #77970: tests: pwm: led_pwm: pwm test fail

hakehuang, 2024-09-04:
**Describe the bug**

led_pwm fails with `Increasing brightness gradually`

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
- [ ] mimxrt1060_evk
- [ ] frdm_kw41z
- [ ] mimxrt1170_evk/mimxrt1176/cm7
- [ ] twr_ke18f
- [ ] vmu_rt1170/mimxrt1176/cm7
- [ ] frdm_ke15z

 - What have you tried to diagnose or workaround this issue?
 - NA, this is test and driver support status issue
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - not a regression.

**To Reproduce**

scripts/twister -p mimxrt1060_evk -T samples/drivers/led_pwm

**Expected behavior**

test can pass

**Impact**

supporting for pwm features

**Logs and console output**

```
2024-09-03 03:50:09,713 - twister - DEBUG - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/shared/disk/nxp/mimxrt1060_evk/temp/mimxrt1060_evk_twister_test_only/twister-out/mimxrt1060_evk/samples/drivers/led_pwm/sample.drivers.led.led_pwm', '--runner', 'linkserver', '--probe=022900000582988000000000000000000000000097969905']
2024-09-03 03:50:16,171 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build v3.7.0-1987-g1540bd7d4930 ***
2024-09-03 03:50:16,177 - twister - DEBUG - DEVICE: [00:00:00.005,000] [0m<inf> main: Testing LED 0 - no label[0m
2024-09-03 03:50:16,177 - twister - DEBUG - HARNESS:Console:EXPECTED(1/7):'Testing LED \d+'
2024-09-03 03:50:16,181 - twister - DEBUG - DEVICE: [00:00:00.010,000] [0m<inf> main:   Turned on[0m
2024-09-03 03:50:16,181 - twister - DEBUG - HARNESS:Console:EXPECTED(2/7):'Turned on'
2024-09-03 03:50:17,186 - twister - DEBUG - DEVICE: [00:00:01.015,000] [0m<inf> main:   Turned off[0m
2024-09-03 03:50:17,186 - twister - DEBUG - HARNESS:Console:EXPECTED(3/7):'Turned off'
2024-09-03 03:50:17,440 - twister - DEBUG - -- west flash: using runner linkserver
-- runners.linkserver: LinkServer: /usr/local/LinkServer/LinkServer, version v1.5.30

2024-09-03 03:50:18,192 - twister - DEBUG - DEVICE: [00:00:02.020,000] [0m<inf> main:   Increasing brightness gradually[0m
2024-09-03 03:50:18,192 - twister - DEBUG - HARNESS:Console:EXPECTED(4/7):'Increasing brightness gradually'
2024-09-03 03:50:20,221 - twister - DEBUG - DEVICE: [00:00:04.046,000] [1;31m<err> pwm_mcux: Too long period (234375), adjust pwm prescaler![0m
2024-09-03 03:50:20,225 - twister - DEBUG - DEVICE: [00:00:04.054,000] [1;31m<err> main: err=-22[0m
2024-09-03 03:50:20,231 - twister - DEBUG - DEVICE: [00:00:04.058,000] [0m<inf> main: Testing LED 0 - no label[0m
2024-09-03 03:50:20,235 - twister - DEBUG - DEVICE: [00:00:04.064,000] [0m<inf> main:   Turned on[0m
2024-09-03 03:50:21,240 - twister - DEBUG - DEVICE: [00:00:05.069,000] [0m<inf> main:   Turned off[0m
2024-09-03 03:50:22,246 - twister - DEBUG - DEVICE: [00:00:06.073,000] [0m<inf> main:   Increasing brightness gradually[0m
2024-09-03 03:50:24,274 - twister - DEBUG - DEVICE: [00:00:08.100,000] [1;31m<err> pwm_mcux: Too long period (234375), adjust pwm prescaler![0m
2024-09-03 03:50:24,279 - twister - DEBUG - DEVICE: [00:00:08.108,000] [1;31m<err> main: err=-22[0m
2024-09-03 03:50:24,284 - twister - DEBUG - DEVICE: [00:00:08.112,000] [0m<inf> main: Testing LED 0 - no label[0m
2024-09-03 03:50:24,289 - twister - DEBUG - DEVICE: [00:00:08.118,000] [0m<inf> main:   Turned on[0m
2024-09-03 03:50:25,293 - twister - DEBUG - DEVICE: [00:00:09.122,000] [0m<inf> main:   Turned off[0m
2024-09-03 03:50:26,300 - twister - DEBUG - DEVICE: [00:00:10.127,000] [0m<inf> main:   Increasing brightness gradually[0m
2024-09-03 03:50:28,328 - twister - DEBUG - DEVICE: [00:00:12.153,000] [1;31m<err> pwm_mcux: Too long period (234375), adjust pwm prescaler![0m
2024-09-03 03:50:28,332 - twister - DEBUG - DEVICE: [00:00:12.161,000] [1;31m<err> main: err=-22[0m
2024-09-03 03:50:28,338 - twister - DEBUG - DEVICE: [00:00:12.166,000] [0m<inf> main: Testing LED 0 - no label[0m
2024-09-03 03:50:28,342 - twister - DEBUG - DEVICE: [00:00:12.171,000] [0m<inf> main:   Turned on[0m
2024-09-03 03:50:29,347 - twister - DEBUG - DEVICE: [00:00:13.176,000] [0m<inf> main:   Turned off[0m
2024-09-03 03:50:30,353 - twister - DEBUG - DEVICE: [00:00:14.180,000] [0m<inf> main:   Increasing brightness gradually[0m
2024-09-03 03:50:32,381 - twister - DEBUG - DEVICE: [00:00:16.207,000] [1;31m<err> pwm_mcux: Too long period (234375), adjust pwm prescaler![0m
2024-09-03 03:50:32,386 - twister - DEBUG - DEVICE: [00:00:16.215,000] [1;31m<err> main: err=-22[0m
2024-09-03 03:50:32,391 - twister - DEBUG - DEVICE: [00:00:16.219,000] [0m<inf> main: Testing LED 0 - no label[0m
2024-09-03 03:50:32,396 - twister - DEBUG - DEVICE: [00:00:16.225,000] [0m<inf> main:   Turned on[0m
2024-09-03 03:50:33,400 - twister - DEBUG - DEVICE: [00:00:17.229,000] [0m<inf> main:   Turned off[0m
2024-09-03 03:50:34,407 - twister - DEBUG - DEVICE: [00:00:18.234,000] [0m<inf> main:   Increasing brightness gradually[0m
2024-09-03 03:50:36,435 - twister - DEBUG - DEVICE: [00:00:20.260,000] [1;31m<err> pwm_mcux: Too long period (234375), adjust pwm prescaler![0m
2024-09-03 03:50:36,439 - twister - DEBUG - DEVICE: [00:00:20.268,000] [1;31m<err> main: err=-22[0m
2024-09-03 03:50:36,445 - twister - DEBUG - DEVICE: [00:00:20.273,000] [0m<inf> main: Testing LED 0 - no label[0m
2024-09-03 03:50:36,450 - twister - DEBUG - DEVICE: [00:00:20.278,000] [0m<inf> main:   Turned on[0m
2024-09-03 03:50:37,440 - twister - DEBUG - Timed out while monitoring serial output on mimxrt1060_evk
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.0-1987-g1540bd7d4930

nordic-piks, 2024-10-18:
It fails also at nrf platforms - e.g, nrf52840dk/nrf52840:
```
*** Booting Zephyr OS build v3.7.0-4578-g829c03bcdca3 ***
[00:00:00.386,627] <inf> main: Testing LED 0 - no label
[00:00:00.392,517] <inf> main:   Turned on
[00:00:01.397,308] <inf> main:   Turned off
[00:00:02.402,191] <inf> main:   Increasing brightness gradually
[00:00:04.429,168] <inf> main:   Blinking on: 100 msec, off: 100 msec
[00:00:09.436,401] <err> pwm_nrfx: Prescaler for period_cycles 32000000 not found.
[00:00:09.444,976] <err> main: err=-22
[00:00:09.449,615] <inf> main:   Cycle period not supported - on: 1000  msec, off: 1000 msec
[00:00:14.458,923] <inf> main:   Turned off, loop end
[00:00:14.464,630] <inf> main: Testing LED 0 - no label
[00:00:14.470,520] <inf> main:   Turned on
[00:00:15.475,311] <inf> main:   Turned off
[00:00:16.480,194] <inf> main:   Increasing brightness gradually
[00:00:18.507,171] <inf> main:   Blinking on: 100 msec, off: 100 msec
```
(but we were not executing those tests for a while - thus reporting lately)

dleach02, 2024-10-24:
raised priority to medium due to cross platform failure detection

dleach02, 2024-10-29:
just FYI but this error occurs all the way back to v3.5

dleach02, 2024-10-30:
The problem is with the call to led_blink() using CONFIG_BLINK_DELAY_SHORT. The value is set to 100 msec which causes a calculation that overflows the 16bit register for setting the PWM in pwm_mcux.c (and likely in Nordic's driver as well). What I'm unsure of is if we should know the maximum BLINK rate that can be set by the pwm as part of the test or not? 

Someone converted the sample to use Kconfig for this and when they did this they added a board specific config to work around the error being reported (e8afe6607a04b4dc3). I can fix this for our boards by doing the same thing or just generically change the setting to something much smaller but this seems like it needs a more generic solution? 

@henrikbrixandersen (PWM) 
@simonguinot @Mani-Sadhasivam (LED)

michal-smola, 2024-10-31:
Please have a look at PR https://github.com/zephyrproject-rtos/zephyr/pull/80605. It should be fixing the issue.

nordic-piks, 2024-11-07:
Triaged internally

dleach02, 2024-11-07:
> Triaged internally

@nordic-piks did you guys come to some conclusions?

nordic-piks, 2024-11-07:
> > Triaged internally
> 
> @nordic-piks did you guys come to some conclusions?

We did not discussed details, just checked if issue has owners etc.
However, I just checked the latest results, there is improvement in pass rate (after https://github.com/zephyrproject-rtos/zephyr/pull/80605)
The are remaining failures at nrf9160dk@0.14.0/nrf9160/ns and nrf5340dk/nrf5340/cpuapp/ns,
test seems to hang:
```
*** Booting Zephyr OS build v4.0.0-rc2-112-g4300a4c33b0f ***
[00:00:00.424,072] <inf> main: Testing LED 0 - no label
[00:00:00.429,718] <inf> main:   Turned on
[00:00:01.434,326] <inf> main:   Turned off
[00:00:02.438,995] <inf> main:   Increasing brightness gradually
[00:00:04.465,667] <inf> main:   Blinking on: 100 msec, off: 100 msec
[00:00:09.472,564] <err> pwm_nrfx: Prescaler for period_cycles 32000000 not found.
[00:00:09.480,773] <err> main: err=-22
[00:00:09.485,229] <inf> main:   Cycle period not supported - on: 1000  msec, off: 1000 msec
```

simonguinot, 2025-08-16:
Hi @decsny.

Blinking delays are now configurable for the `led_pwm` sample since https://github.com/zephyrproject-rtos/zephyr/commit/6a84631eec9619c333d85cb784750ca74c531b23.

So these options can be used to adjust the blinking periods for platform. Here is an example: https://github.com/zephyrproject-rtos/zephyr/commit/85e1353624454cceb81edca5686dd4a99f313ee6.

Unfortunately I don't own the platforms mentioned in this issue. So I am unable to fix it...

hakehuang, 2025-10-27:
> [@hakehuang](https://github.com/hakehuang) [@nordic-piks](https://github.com/nordic-piks) can you still reproduce?

@jeppenodgaard your log shows you do not execute any of them only build. and  pre my testing the issue is still there.

this is your log (build only)
```
 2 selected test cases not executed: 2 not run (built only).
```

this is my latest test log
```
*** Booting Zephyr OS build v4.2.0-6179-g169fd6a4a98e ***
[00:00:00.005,000] <inf> main: Testing LED 0 - no label
[00:00:00.010,000] <inf> main:   Turned on
[00:00:01.015,000] <inf> main:   Turned off
[00:00:02.020,000] <inf> main:   Increasing brightness gradually
[00:00:02.047,000] <err> main: err=-16 brightness=2

[00:00:02.052,000] <inf> main: Testing LED 0 - no label
[00:00:02.058,000] <inf> main:   Turned on
[00:00:03.063,000] <inf> main:   Turned off
[00:00:04.067,000] <inf> main:   Increasing brightness gradually
[00:00:04.084,000] <err> main: err=-16 brightness=1

[00:00:04.090,000] <inf> main: Testing LED 0 - no label
[00:00:04.096,000] <inf> main:   Turned on
[00:00:05.100,000] <inf> main:   Turned off
[00:00:06.105,000] <inf> main:   Increasing brightness gradually
[00:00:06.122,000] <err> main: err=-16 brightness=1

[00:00:06.127,000] <inf> main: Testing LED 0 - no label
[00:00:06.133,000] <inf> main:   Turned on
[00:00:07.138,000] <inf> main:   Turned off
[00:00:08.142,000] <inf> main:   Increasing brightness gradually
[00:00:08.170,000] <err> main: err=-16 brightness=2

[00:00:08.175,000] <inf> main: Testing LED 0 - no label
[00:00:08.181,000] <inf> main:   Turned on
[00:00:09.185,000] <inf> main:   Turned off
[00:00:10.190,000] <inf> main:   Increasing brightness gradually
[00:00:10.207,000] <err> main: err=-16 brightness=1

[00:00:10.212,000] <inf> main: Testing LED 0 - no label
[00:00:10.218,000] <inf> main:   Turned on
[00:00:11.223,000] <inf> main:   Turned off
[00:00:12.227,000] <inf> main:   Increasing brightness gradually
[00:00:12.245,000] <err> main: err=-16 brightness=1

[00:00:12.250,000] <inf> main: Testing LED 0 - no label
[00:00:12.256,000] <inf> main:   Turned on
[00:00:13.260,000] <inf> main:   Turned off
[00:00:14.265,000] <inf> main:   Increasing brightness gradually
[00:00:14.282,000] <err> main: err=-16 brightness=1

[00:00:14.287,000] <inf> main: Testing LED 0 - no label
[00:00:14.293,000] <inf> main:   Turned on
[00:00:15.298,000] <inf> main:   Turned off
[00:00:16.302,000] <inf> main:   Increasing brightness gradually
[00:00:16.330,000] <err> main: err=-16 brightness=2

[00:00:16.335,000] <inf> main: Testing LED 0 - no label
[00:00:16.341,000] <inf> main:   Turned on
[00:00:17.345,000] <inf> main:   Turned off
[00:00:18.350,000] <inf> main:   Increasing brightness gradually
[00:00:18.367,000] <err> main: err=-16 brightness=1

[00:00:18.373,000] <inf> main: Testing LED 0 - no label
[00:00:18.378,000] <inf> main:   Turned on
[00:00:19.383,000] <inf> main:   Turned off

```