zephyrproject-rtos/zephyr #77972: mgmt: mcumgr: handle merged TF-M applications

JordanYates, 2024-09-04:
**Is your enhancement proposal related to a problem? Please describe.**

The current image management implementation cannot handle TF-M applications that use a merged secure and non-secure application from the perspective of the bootloader (`TFM_MCUBOOT_IMAGE_NUMBER == 1`).

The flash area logic only refer to the secure partitions:
https://github.com/zephyrproject-rtos/zephyr/blob/ca48767be48ca6a7536edf7b1d59f3246046d40e/subsys/mgmt/mcumgr/grp/img_mgmt/src/zephyr_img_mgmt.c#L29-L34

https://github.com/zephyrproject-rtos/zephyr/blob/ca48767be48ca6a7536edf7b1d59f3246046d40e/boards/nordic/nrf9151dk/nrf9151dk_nrf9151_common.dtsi#L216-L243

This results in the mcumgr commands trying to read the image header from the start of the secure flash (correct, but currently triggers a SecureFault) and the image trailer from the end of secure flash (incorrect, also triggers a SecureFault).

**Describe the solution you'd like**

The image management handlers should be updated to pull information from the correct flash areas (distinct from the slot ID) when built with TFM. This breaks the current assumptions that a single slot ID has a single corresponding flash area ID.

nordicjm, 2024-09-04:
Not sure I fully understand the bug, the image you want to upload that is merged, is that one contagious area with just one MCUboot header/trailer/signature for the whole area, or 2? If it is 2, you cannot upload it as a single file. Or do you mean that the _ns partitions are not used by MCUmgr?

JordanYates, 2024-09-04:
The image is one contiguous area, with a single header and trailer. In essence, `slot0_partition + slot0_ns_partition` (as required by `TFM_MCUBOOT_IMAGE_NUMBER == 1`).
But, based on the devicetree partitioning, all the flash area operations are on the flash area corresponding with `slot0_partition`.
Hopefully it is obvious why any image operations would fail given that.

In fact I suspect (but have not tested) that this is broken for `TFM_MCUBOOT_IMAGE_NUMBER == 2` as well, given that it is looking at the secure partition, not the non-secure partition.

JordanYates, 2024-09-04:
My plan on resolving this downstream is to tweak how the partitions are defined so that mcumgr gets a handle to the complete flash area, while still preserving the secure/nonsecure partitioning for other tooling:
```
		slot0_partition: partition@10000 {
			label = "image-0";
			reg = <0x00010000 0x000F0000>;
		};
		slot0_s_partition: subpartition@10000 {
			label = "image-0-secure";
			reg = <0x00010000 0x00030000>;
		};
		slot0_ns_partition: subpartition@40000 {
			label = "image-0-nonsecure";
			reg = <0x00040000 0xC0000>;
		};
```
The `subpartition` node name is required to avoid defining two nodes with the same name (`partition@10000`).
Note this sort of change would be made much easier upstream with #74707 (Which I note you have approved 👍).

de-nordic, 2025-01-31:
I think we should separate partitioning from board definitions and provide something like partitioning profiles, for example: default, mcuboot and tfm, that would be given to a build as an option.

JordanYates, 2025-01-31:
> I think we should separate partitioning from board definitions and provide something like partitioning profiles, for example: default, mcuboot and tfm, that would be given to a build as an option.

Where would these profiles live, and how would they handle a secondary image that lives on external flash?

de-nordic, 2025-01-31:
> 
> Where would these profiles live, and how would they handle a secondary image that lives on external flash?

I had two ideas for that, one was that every board defines these profiles that it is supposed to work with, as overlays - probably a lot of work.
Second was to define a templates for them and boards would have configurations for templates.

I have already off-line talk with @nordicjm regarding above and see how hard that would be to implement.

One of the reasons I wanted to have at least the mcuboot and default profile was so that both mcuboot and non-mcuboot builds would use `zephyr,code-partition`  selected partition, but it's definition would just change; this would solve the problem where mcuboot build of an app is aware of how it is limited to a certain partition, while non-mcuboot builds ignore partitioning and can span into other partitions, like storage.

nordicjm, 2025-05-20:
@JordanYates been playing with this yesterday and been getting the secure fault you observe (however, using 2 images not 1) which makes it impossible to mark an image for update in the secure update slot, how would you expect this to work?

JordanYates, 2025-05-20:
> @JordanYates been playing with this yesterday and been getting the secure fault you observe (however, using 2 images not 1) which makes it impossible to mark an image for update in the secure update slot, how would you expect this to work?

The faults with 2 images are a different problem than this issue, since as you observe it would require writing to secure flash. The only way I could see that working is by deferring the write to the secure image through some PSA shim, but that is more complicated than solving the merged image case.

The merged case should be able to progress towards an upstream solution with the addition of #86012

nordicjm, 2025-05-20:
Not following this, if you have `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER` = 1 then you would have s and ns combined so would be using slot0 and slot1, when reading slot0 you would fault because you're in secure mode. If you have `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER` = 2 then s and ns would be different, however there is actually a disjoint here because you can have `CONFIG_MCUMGR_GRP_IMG_UPDATABLE_IMAGE_NUMBER` = 1 or 2, if it's equal to 1 then in theory you can only see the non-secure image (note: this assumes, wrongly I think, that the first image is the non-secure one, I assume it should be the secure one?) and can update that (though is that really intentional? I don't use TF-M so I'm not sure if you would only update the non-secure part without the secure part) - this, oddly, would actually work, however if you set this to 2 then it would fault again because for the other image, you would try accessing the secure are which would fault.

So I guess I am asking for the following combinations what would you expect to be each image in MCUmgr (secure, or non-secure, or combined) and how you would expect to use it given that it would fault - for:

- `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER` = 1
- `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER` = 2 and `CONFIG_MCUMGR_GRP_IMG_UPDATABLE_IMAGE_NUMBER` = 1
- `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER` = 2 and `CONFIG_MCUMGR_GRP_IMG_UPDATABLE_IMAGE_NUMBER` = 2

JordanYates, 2025-05-20:
> Not following this, if you have CONFIG_TFM_MCUBOOT_IMAGE_NUMBER = 1 then you would have s and ns combined so would be using slot0 and slot1, when reading slot0 you would fault because you're in non-secure mode trying to read the secure image.

You would, if you didn't have my commits to make that make it work :) (77d3277f44f4e106949f7e7e61769dc077c11548, 9dc7bc7b26121e510a007665667d62d4af9b114a)
Note that you could take a stricter view than I do of the secure memory and restrict the read access down to just the MCUboot header like I had originally, I have other reasons to make the entire program memory readable.

I don't use `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER = 2` and have no plans to do so, co-ordinating multiple images that may fail at different points isn't worth any theoretical savings to me (and likely never would be, since I update based on binary diffs, so it doesn't take me any longer to write an image if the secure application stays the same)

nordicjm, 2026-09-16:
should be fixed with https://github.com/zephyrproject-rtos/zephyr/pull/118016/

closed 2026-09-16 by nordicjm