zephyrproject-rtos/zephyr #77983: Multiple ICE40 instances on the same SPI master

benediktibk, 2024-09-04:
**Describe the bug**
When multiple ICE40 instances on the same SPI master are enabled the build will fail, as there will be multiple instances of the same pinctrl-config object within fpga_ice40.c.

**To Reproduce**
1. CONFIG_PINCTRL=y
2. enable two instances of `lattice,ice40-fpga` on the same master.
3. west build -> BOOM 

**Expected behavior**
It should be possible to use two ICE40 with PINCTRL on the same SPI master.

tbursztyka, 2024-09-04:
Could you give more detailed steps on how to reproduce? I miss the board and the sample/test to use.

benediktibk, 2024-09-05:
> Could you give more detailed steps on how to reproduce? I miss the board and the sample/test to use.

With the current setup the corresponding test doesn't trigger the issue, as there is a workaround for it in place. But if you set `CONFIG_PINCTRL=y` in `zephyr/tests/drivers/build_all/fpga/prj.conf` you can trigger the issue with `west build --board nucleo_f746zg zephyr/tests/drivers/build_all/fpga`.

cfriedt, 2024-09-09:
This issue also came up when the ice40 driver was added. But at the time, there was no clean way to program multiple nodes on the same SPI bus, since it was not possible to reset to a default SPI pinctrl state.

Ultimately, the SPI API will likely need to be changed to fix this bug.

benediktibk, 2024-11-28:
After #80854 has been merged the problem is reduced to the driver for lattice,ice40-fpga-bitbang. Therefore less severe, but still existing.

PR #87900 (2025-04-24): Enable the usage of multiple ICE40 in bitbang mode on the same SPI peripheral
This will fix #77983, but I'm still uncertain if we should actually merge it. Because at least for the hardware which I have available deinit is not yet implemented, so I wasn't even able to test it.

With this PR we will basically change the situation for some devices from the state "you can have only one ICE40" to "the ICE40 is not supported". But, at least in such cases there is now a clear path forward: Implement deinit for the SPI peripheral on this device. Considering this, as well as that this bitbang version  [... cut]
commit 21c1a9ef (2025-04-24): drivers: fpga: use device_deinit in ICE40 bitbang driver

Replace reapplying the original pin configuration via pinctrl
in the ICE40 bitbang driver with a device_deinit/device_init.
Fixes #77983.
closed 2025-04-24 by kartben