zephyrproject-rtos/zephyr #78010: MQTT-SN Client library Gateway Search Support

SpontaneousDuck, 2024-09-04:
**Is your enhancement proposal related to a problem? Please describe.**
Our current system design can contain multiple MQTT-SN Gateways in a network and relies on ADVERTISE, SEARCHGW and GWINFO messages to locate the nearest gateway for connection. These messages allow the gateway to change and us not hardcode edge devices to a gateway.

**Describe the solution you'd like**
Support for "Gateway discovery using ADVERTISE, SEARCHGW and GWINFO messages". This is currently listed as unsupported in the documentation. This would probably include some redesign of the transport interface for MQTT-SN as well as it currently has no concept of broadcast/addressed messages and connecting/disconnecting from addresses. 

**Describe alternatives you've considered**
One workaround would be exposing the MQTT-SN message encoding directly to the user to allow searching completely separately and then using the client library to connect once a gateway has been found. This would mean user would need to operate the transport separately outside of the library as well through the library.

**Additional context**

Encoding/decoding the ADVERTISE, SEARCHGW and GWINFO messages is already supported internally to the MQTT-SN Client library. These are not used by the exposed APIs available now though. The "transport" interface is designed in a way where the transport layer connection to the Gateway is established at client initialization time. This interface will need to be redesigned to support the "broadcast" and "addressed" message states in MQTT-SN. This process is described in section 6.1 of the MQTT-SN spec [link](https://groups.oasis-open.org/higherlogic/ws/public/download/66091/MQTT-SN_spec_v1.2.pdf). 
1. Upon initialization a client usually has no awareness of any gateways it will the "broadcast" messages to anyone listening that it is looking for a gateway. 
2. Once receiving a GWINFO message, the client will then use the CONNECT message to "bind" to a selected Gateway. 

The Client API flow would be mqtt_sn_client_init -> "mqtt_sn_client_search" -> mqtt_sn_connect.  

There will also need to be some work in the receive handler to handle ADVERTISE and GWINFO packets.

rlubos, 2024-09-05:
@SpontaneousDuck Are you willing to work on this feature?

SpontaneousDuck, 2024-09-05:
Yes! It may take me few weeks but I should be able to do it. I'll set up a PR once I have something started.

commit aa9c9228 (2024-11-16): net: mqtt-sn: Add Gateway Advertisement and Discovery process support

Fixes: #78010
This commit implements the "Gateway Advertisement and Discovery" process
defined in section 6.1 of the MQTT-SN specification.
This includes breaking changes to the transport interface and the default
included UDP interface implementation as support for UDP multicast
messages is added as implemented by the Paho MQTT-SN Gateway.
commit cc33fd68 (2024-11-16): net: mqtt-sn: Update MQTT-SN Unit Tests

Fixes: #78010
This commit adds tests for the "Gateway Advertisement and Discovery"
process. This includes tests for handling SEARCHGW, GWINFO, and ADVERTISE
as well as manually adding a Gateway for use. Gateway pruning when
missing ADVERTISE messages is also tested.
commit e19838d1 (2024-11-16): net: mqtt-sn: Update MQTT-SN Publisher Sample with Gateway Discovery

Fixes: #78010
This commit adds the "Gateway Advertisement and Discovery" process
to the MQTT-SN Publisher Sample application.
closed 2024-11-16 by nashif