zephyrproject-rtos/zephyr #78012: STM32 erasing flash partition causes hard fault crash if CONFIG_NETWORKING and CONFIG_LWM2M are enabled

davidalo, 2024-09-04:
**Describe the bug**
STM32 erasing flash partition causes hard fault crash if CONFIG_NETWORKING and CONFIG_LWM2M are enabled

**To Reproduce**

- Sample program:
```
#include <zephyr/drivers/flash.h>
#include <zephyr/storage/flash_map.h>
#include "zephyr/sys/printk.h"

#define STORAGE_PARTITION			storage_partition
#define STORAGE_PARTITION_DEVICE	FIXED_PARTITION_DEVICE(STORAGE_PARTITION)
#define STORAGE_PARTITION_OFFSET	FIXED_PARTITION_OFFSET(STORAGE_PARTITION)
#define STORAGE_PARTITION_ID		FIXED_PARTITION_ID(STORAGE_PARTITION)

int main(void)
{
	int rc;
	if (!device_is_ready(STORAGE_PARTITION_DEVICE)) {
		printk("Flash device %s is not ready\n", STORAGE_PARTITION_DEVICE->name);
		return 0;
	}

	const struct flash_area *fa;

	rc = flash_area_open(STORAGE_PARTITION_ID, &fa);
	if (rc) {
		printk("flash_area_open() fail, error %d\n", rc);
		return rc;
	}

	rc = flash_area_erase(fa, 0, fa->fa_size);
	if (rc) {
		printk("flash_area_erase() fail, error %d\n", rc);
		return rc;
	}

	printk("partition has been erased\n");
	
	return 0;
}
```

- Configuration:
```
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y

CONFIG_NETWORKING=y
CONFIG_LWM2M=y
```

If I disable either CONFIG_NETWORKING or CONFIG_LWM2M it works as expected.

**Expected behavior**

**Impact**

**Logs and console output**
- GDB output of hard fault:

```
(gdb) backtrace
#0  z_arm_fatal_error (reason=36, esf=esf@entry=0x200264d8 <z_interrupt_stacks+1944>) at /app/zephyr/arch/arm/core/fatal.c:79
#1  0x080020b0 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /app/zephyr/arch/arm/core/cortex_m/fault.c:1157
#2  0x08002178 in z_arm_usage_fault () at /app/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#3  <signal handler called>
#4  z_arm_fatal_error (reason=36, esf=esf@entry=0x20026540 <z_interrupt_stacks+2048>) at /app/zephyr/arch/arm/core/fatal.c:79
#5  0x080020b0 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /app/zephyr/arch/arm/core/cortex_m/fault.c:1157
#6  0x08002178 in z_arm_usage_fault () at /app/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#7  <signal handler called>
#8  flash_stm32_wait_flash_idle (dev=dev@entry=0x8018ed8 <__device_dts_ord_105>) at /app/build/zephyr/include/generated/zephyr/syscalls/kernel.h:608
#9  0x0800c9b6 in erase_sector (sector=<optimized out>, dev=0x8018ed8 <__device_dts_ord_105>) at /app/zephyr/drivers/flash/flash_stm32f7x.c:111
#10 flash_stm32_block_erase_loop (dev=dev@entry=0x8018ed8 <__device_dts_ord_105>, offset=offset@entry=65536, len=len@entry=65536) at /app/zephyr/drivers/flash/flash_stm32f7x.c:138
#11 0x0800c84c in flash_stm32_erase (len=65536, offset=65536, dev=0x8018ed8 <__device_dts_ord_105>) at /app/zephyr/drivers/flash/flash_stm32.c:207
#12 flash_stm32_erase (dev=0x8018ed8 <__device_dts_ord_105>, offset=65536, len=65536) at /app/zephyr/drivers/flash/flash_stm32.c:186
#13 0x08001a60 in main () at /app/zephyr/samples/subsys/nvs/src/main.c:26
``` 

**Environment (please complete the following information):**

 - Zephyr v3.7.0
 - Board: Nucleo F767ZI

**Additional context**

davidalo, 2024-09-04:
Crashes here (`flash_stm32f7x.c  erase_sector`):

```
	regs->CR = (regs->CR & ~(FLASH_CR_PSIZE | FLASH_CR_SNB)) |
		   FLASH_PSIZE_BYTE |
		   FLASH_CR_SER |
		   (sector << FLASH_CR_SNB_Pos) |
		   FLASH_CR_STRT;
```

davidalo, 2024-09-04:
Backtrace  with `CONFIG_DEBUG=y

```
(gdb) backtrace
#0  fault_handle (esf=esf@entry=0x20026320 <z_interrupt_stacks+1904>, fault=3, recoverable=recoverable@entry=0x20026307 <z_interrupt_stacks+1879>)
    at /app/zephyr/arch/arm/core/cortex_m/fault.c:843
#1  0x0800223c in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /app/zephyr/arch/arm/core/cortex_m/fault.c:1119
#2  0x080022cc in z_arm_usage_fault () at /app/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#3  <signal handler called>
#4  fault_handle (esf=esf@entry=0x20026380 <z_interrupt_stacks+2000>, fault=6, recoverable=recoverable@entry=0x20026367 <z_interrupt_stacks+1975>)
    at /app/zephyr/arch/arm/core/cortex_m/fault.c:843
#5  0x0800223c in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /app/zephyr/arch/arm/core/cortex_m/fault.c:1119
#6  0x080022cc in z_arm_usage_fault () at /app/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#7  <signal handler called>
#8  sys_clock_announce (ticks=775) at /app/zephyr/kernel/timeout.c:211
#9  0x0800f4c8 in sys_clock_isr () at /app/zephyr/drivers/timer/cortex_m_systick.c:214
#10 <signal handler called>
#11 z_impl_k_uptime_ticks () at /app/zephyr/kernel/timeout.c:281
#12 0x0800eb54 in k_uptime_ticks () at /app/build/zephyr/include/generated/zephyr/syscalls/kernel.h:609
#13 k_uptime_get () at /app/zephyr/include/zephyr/kernel.h:1830
#14 flash_stm32_wait_flash_idle (dev=dev@entry=0x801f594 <__device_dts_ord_105>) at /app/zephyr/drivers/flash/flash_stm32.c:110
#15 0x0800ec90 in erase_sector (dev=dev@entry=0x801f594 <__device_dts_ord_105>, sector=sector@entry=2) at /app/zephyr/drivers/flash/flash_stm32f7x.c:111
#16 0x0801d4e6 in flash_stm32_block_erase_loop (dev=dev@entry=0x801f594 <__device_dts_ord_105>, offset=offset@entry=65536, len=len@entry=65536)
    at /app/zephyr/drivers/flash/flash_stm32f7x.c:138
#17 0x0801d402 in flash_stm32_erase (dev=0x801f594 <__device_dts_ord_105>, offset=65536, len=65536) at /app/zephyr/drivers/flash/flash_stm32.c:209
#18 0x08012456 in z_impl_flash_erase (size=<optimized out>, offset=<optimized out>, dev=<optimized out>) at /app/zephyr/include/zephyr/drivers/flash.h:318
#19 flash_erase (dev=<optimized out>, offset=<optimized out>, size=<optimized out>) at /app/build/zephyr/include/generated/zephyr/syscalls/flash.h:89
#20 0x08012480 in flash_area_erase (fa=<optimized out>, off=off@entry=0, len=<optimized out>) at /app/zephyr/subsys/storage/flash_map/flash_map.c:82
#21 0x08001988 in main () at /app/zephyr/samples/subsys/nvs/src/main.c:30
```

davidalo, 2024-09-05:
Using a post kernel function after stm32_flash_init and before net_init does not help.

FRASTM, 2024-09-25:
with CONFIG_NETWORKING=y  and CONFIG_DEBUG=n
- the sector is erased at address 0x08010000 (storage partition) after flashing (which is not a mass erase from west flash)
- the sector is erased if the storage partition is  partition@100000  (in place of slot1)

FRASTM, 2024-10-24:
--> sample exceeding the 64K (0x10000) mcuboot,  partition is erased at  sector at address 0x10000
can you try Moving  the sector to erase at 0x20000 or so  (update partitioning)

erwango, 2024-11-06:
> --> sample exceeding the 64K (0x10000) mcuboot, partition is erased at sector at address 0x10000 can you try Moving the sector to erase at 0x20000 or so (update partitioning)

@davidalo Did you had a try ?

davidalo, 2024-11-11:
Hello, 
Sorry for my late reply. I will give it a try this week.

davidalo, 2024-11-11:
Moving `storage_partition` to address 0x20000 sector 4 with 128 KB seems to work, no hard fault.

```
		storage_partition: partition@20000 {
			label = "storage";
			reg = <0x00020000 DT_SIZE_K(128)>;
		};
```

erwango, 2024-11-12:
> le application is erasing **sector 2** from 0x801 0000 - 0x801 7FFF, which correspnds to the "storage partition"
> Are we sure that this sector does not contains code ?
> In other words, does the sample fit the boot_partition of 64K ? If so the sample could be erasing i

Ok, thanks for the update.
Unfortunately, there is nothing that can be fixed in board configuration as "storage" partition can't be moved to 0x20000 on a permanent basis. Zephyr storage partitions are supposed to be used by FS subsystems which requires 2 sectors of maximum 64K. On a F767ZI based board, the only option is at 0x10000.
To be compatible with a large application as yours, you'd need to use a bootloader (mcuboot) which would jump to an application in a large slot after 0x40000 (For instance `slot0_partition` as defined in this board).

As a consequence, I'm closing this issue.