zephyrproject-rtos/zephyr #78020: edtlib: wrong endianess in many functions

avolkov-1221, 2024-09-04:
**Describe the bug**

Meanwhile the Zephyr [documentation](https://docs.zephyrproject.org/latest/build/dts/intro-syntax-structure.html#writing-property-values) is clear enough about the 64 bits integers:

> 64-bit integers are written as two 32-bit cells in big-endian order. The value 0xaaaa0000bbbb1111 would be written <0xaaaa0000 0xbbbb1111>.

Many functions, like `_init_regs`, assume the opposite and treat values as little-endian (using [:4*addr_len] in Python semantics). As a result, declarations like the following:

 ```
parent {
    #address-cell = <1>
    node-1@1 {
        regs = <0 1>
    }
}
```
are handled incorrectly. Specifically, the `reg` value should be 1, but almost all parts of edtlib will treat the `reg`'s option value as 0. As if it were written:
 ```
parent {
    #address-cell = <1>
    node-1@1 {
        regs = <1 0>
    }
}
```
**To Reproduce**

Use any edtlib based script with the above snippets, and check the result.

**Expected behavior**

The 64-bit values (and above) should be handled as big-endian, not little- one. Otherwise, device trees for Zephyr and other software will become incompatible.

**Impact**

All edtlib based software.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: any
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: any up to and including the 3.7.99

**Additional context**

57300, 2024-09-05:
> ```
> parent {
>    #address-cell = <1>;
>    node-1@1 {
>        regs = <0 1>;
>    };
> };
> ```
> 
> 
>     
>   
> 
> are handled incorrectly. Specifically, the `reg` value should be 1, but the function will treat the `reg`'s property value as 0. As if it had been written:

It's 0 because `#address-cells = <1>`. The second `reg` cell (1) would be the size (assuming implicit `#size-cells = <1>`).
Try setting `#address-cells = <2>` to represent 64-bit addresses.

avolkov-1221, 2024-09-05:
> > declarations like the following:
> > ```
> > parent {
> >    #address-cell = <1>;
> >    node-1@1 {
> >        regs = <0 1>;
> >    };
> > };
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > are handled incorrectly. Specifically, the `reg` value should be 1, but the function will treat the `reg`'s property value as 0. As if it had been written:
> 
> It's 0 because `#address-cells = <1>`. The second `reg` cell (1) would be the size (assuming implicit `#size-cells = <1>`). Try setting `#address-cells = <2>` to represent 64-bit addresses.

I know. It's not a bug, it was an easiest way to demonstrate the incorrect work of the library. In case of big endian, it should be 1, not 0.

mmahadevan108, 2024-09-10:
@avolkov-1221 , can you submit a PR for this issue?

mmahadevan108, 2024-09-12:
@decsny @galak , do you have the bandwidth to look into this.

avolkov-1221, 2024-09-13:
> @avolkov-1221 , can you submit a PR for this issue?

It's https://github.com/zephyrproject-rtos/zephyr/pull/77900 PR. It's under development.

decsny, 2024-09-17:
I will say, it looks like that function is only used for interrupt mapping, and I don't think the rest of the DT tooling treats addresses like little endian, because there are many examples of big endian multi-cell addresses being actively used in zephyr such as in I2C and I3C addresses which are 96 bits and written in big endian format and resulting in a big endian unit address.

ydamigos, 2025-06-27:
This issue should be closed. PR [#91011](https://github.com/zephyrproject-rtos/zephyr/pull/91011) fixed it.

closed 2025-06-27 by pdgendt