zephyrproject-rtos/zephyr #78050: "PWM" on teensy41

Chethan-TS, 2024-09-05:
**unable to configure PWM on teensy 4.1**

i am trying the "samples/basic/servo_motor" example for teensy41 board, below have attached the overlay for teensy41 but when i build the same i get error.

chethan@Chethan:~/GCB_rnd/BLDC/teensy_test$ west build -p always -b teensy41
-- west build: making build dir /home/chethan/GCB_rnd/BLDC/teensy_test/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/chethan/GCB_rnd/BLDC/teensy_test
-- CMake version: 3.29.6
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/chethan/.cache/zephyr
-- Zephyr version: 3.5.0 (/home/chethan/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: teensy41
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/chethan/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/home/chethan/zephyr-sdk-0.16.5-1)
-- Found Dtc: /home/chethan/zephyr-sdk-0.16.5-1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/chethan/zephyrproject/zephyr/boards/arm/teensy4/teensy41.dts
-- Found devicetree overlay: /home/chethan/GCB_rnd/BLDC/teensy_test/app.overlay
node '/soc/i2c@403f0000/mpu6050@68' compatible 'inv,mpu6050' has unknown vendor prefix 'inv'
-- Generated zephyr.dts: /home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/dts.cmake

warning: The choice symbol MPU6050_TRIGGER_NONE (defined at drivers/sensor/mpu6050/Kconfig:22) was
selected (set =y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MPU6050_TRIGGER_NONE and/or look up
MPU6050_TRIGGER_NONE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/chethan/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/chethan/zephyrproject/zephyr/boards/arm/teensy4/teensy41_defconfig'
Merged configuration '/home/chethan/GCB_rnd/BLDC/teensy_test/prj.conf'
Configuration saved to '/home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/.config'
Kconfig header saved to '/home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/chethan/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/chethan/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Load components for MIMXRT1062:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpspi component is included.
driver_lpuart component is included.
driver_igpio component is included.
driver_pwm component is included.
driver_cache_armv7_m7 component is included.
-- Configuring done (4.2s)
-- Generating done (0.1s)
-- Build files have been written to: /home/chethan/GCB_rnd/BLDC/teensy_test/build
-- west build: building application
[1/141] Preparing syscall dependency handling

[2/141] Generating include/generated/version.h
-- Zephyr version: 3.5.0 (/home/chethan/zephyrproject/zephyr), build: zephyr-v3.5.0
[106/141] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/chethan/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DCPU_MIMXRT1062DVJ6A -DKERNEL -DNDEBUG -DPICOLIBC_DOUBLE_PRINTF_SCANF -DXIP_BOOT_HEADER_DCD_ENABLE=1 -DXIP_BOOT_HEADER_ENABLE=1 -DXIP_EXTERNAL_FLASH -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/chethan/zephyrproject/zephyr/include -I/home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/include/generated -I/home/chethan/zephyrproject/zephyr/soc/arm/nxp_imx/rt -I/home/chethan/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/chethan/zephyrproject/zephyr/modules/cmsis/. -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1062 -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1062/drivers -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1062/xip -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/lpspi -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/lpuart -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/igpio -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/pwm -I/home/chethan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7 -I/home/chethan/zephyrproject/zephyr/modules/hal_nxp/. -fno-strict-aliasing -Os -imacros /home/chethan/GCB_rnd/BLDC/teensy_test/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/chethan/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/chethan/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/chethan/GCB_rnd/BLDC/teensy_test=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/chethan/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/chethan/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/chethan/GCB_rnd/BLDC/teensy_test/src/main.c
/home/chethan/GCB_rnd/BLDC/teensy_test/src/main.c: In function 'process_mpu6050':
/home/chethan/GCB_rnd/BLDC/teensy_test/src/main.c:59:33: warning: passing argument 2 of 'uart_poll_out' makes integer from pointer without a cast [-Wint-conversion]
   59 |         uart_poll_out(uart_dev, buf); // Sending data via UART
      |                                 ^~~
      |                                 |
[... 46 lines cut ...]
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/home/chethan/zephyrproject/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/home/chethan/zephyrproject/zephyr/include/zephyr/device.h:280:9: note: in expansion of macro 'COND_CODE_1'
  280 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(compat),                         \
      |         ^~~~~~~~~~~
/home/chethan/zephyrproject/zephyr/include/zephyr/device.h:282:22: note: in expansion of macro 'ZERO_OR_COMPILE_ERROR'
  282 |                     (ZERO_OR_COMPILE_ERROR(0)))
      |                      ^~~~~~~~~~~~~~~~~~~~~
/home/chethan/GCB_rnd/BLDC/teensy_test/src/main.c:85:43: note: in expansion of macro 'DEVICE_DT_GET_ONE'
   85 |     const struct device *const uart_dev = DEVICE_DT_GET_ONE(uart1);
      |                                           ^~~~~~~~~~~~~~~~~
[119/141] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/chethan/GCB_rnd/BLDC/teensy_test/build
chethan@Chethan:~/GCB_rnd/BLDC/teensy_test$ 

i would be grateful if someone assist me on the same. 

* Page: `amples/basic/servo_motor`
* Version: 3.5.0

[overlay.txt](https://github.com/user-attachments/files/16892167/overlay.txt)

henrikbrixandersen, 2024-09-10:
> * Version: 3.5.0

Zephyr version 3.5.0 is no longer supported. Can you reproduce this on either v3.7.0 or current `main` branch?

bdkrae, 2024-09-10:
With current `main` branch the overlay file `teensy41.overlay` for `samples/basic/servo_motor` should look something like this:

```
/ {
	servo: servo {
		compatible = "pwm-servo";
		pwms = <&flexpwm2_pwm2 1 PWM_MSEC(20) PWM_POLARITY_NORMAL>;
		min-pulse = <PWM_USEC(700)>;
		max-pulse = <PWM_USEC(2500)>;
	};
};

&pinctrl {
	pinmux_flexpwm_ch1: pinmux_flexpwm_ch1 {
		group0 {
			pinmux = <&iomuxc_gpio_b0_11_flexpwm2_pwmb2>;
			drive-strength = "r0-7";
			slew-rate = "fast";
			nxp,speed = "200-mhz";
		};
	};
};

&flexpwm2_pwm2 {
	status = "okay";
	pinctrl-0 = <&pinmux_flexpwm_ch1>;
	pinctrl-names = "default";
	nxp,prescaler = <64>;
};
```

FlexPWM peripheral is not active per default for teensy board, so you have to do that in this file. Also you need a pinmapping, in this example its mapped to `gpio_b0_11` which you will find on boards `pin 9`.

mmahadevan108, 2024-09-25:
@Chethan-TS did you get a chance to try the suggestion from @bdkrae

Chethan-TS, 2024-09-26:
thanks for the response @bdkrae .

yes, @mmahadevan108 i did try with the current **main** branch . in his particular overlay, flexpwm2_pwmb2 has been used and it worked well and fine for the above mentioned example , when i try to use other available pwm's only few worked.

i do not know the reason, it would be helpful it i get clarification on the same.

 
[image]

bdkrae, 2024-09-30:
Here is an extended example for `teensy41.overlay`. Maybe this helps to clarify the principle. If you still experience problems please provide more details on which pins it fails.

```
/ {
	servo: servo {
		compatible = "pwm-servo";
		// pwms = <&flexpwm2_pwm0 1 PWM_MSEC(20) PWM_POLARITY_NORMAL>; // flexpwm2 ch0 output B -> teensy pin 33
		// pwms = <&flexpwm2_pwm0 0 PWM_MSEC(20) PWM_POLARITY_NORMAL>; // flexpwm2 ch0 output A -> teensy pin 4
		// pwms = <&flexpwm2_pwm2 0 PWM_MSEC(20) PWM_POLARITY_NORMAL>; // flexpwm2 ch2 output A -> teensy pin 6
		// pwms = <&flexpwm4_pwm2 1 PWM_MSEC(20) PWM_POLARITY_NORMAL>; // flexpwm4 ch2 output B -> teensy pin 3
		pwms = <&flexpwm3_pwm1 1 PWM_MSEC(20) PWM_POLARITY_NORMAL>; // flexpwm3 ch1 output B -> teensy pin 28
		min-pulse = <PWM_USEC(700)>;
		max-pulse = <PWM_USEC(2500)>;
	};
};

&pinctrl {
    pinmux_flexpwm2_ch0: pinmux_flexpwm2_ch0 {
        group0 {
            pinmux = <&iomuxc_gpio_emc_06_flexpwm2_pwma0>, //output A
                    <&iomuxc_gpio_emc_07_flexpwm2_pwmb0>; //output B
            drive-strength = "r0-7";
            slew-rate = "fast";
            nxp,speed = "200-mhz";
        };
    };
	pinmux_flexpwm2_ch2: pinmux_flexpwm2_ch2 {
		group0 {
			pinmux = <&iomuxc_gpio_b0_10_flexpwm2_pwma2>, //output A
                    <&iomuxc_gpio_b0_11_flexpwm2_pwmb2>; //output B
			drive-strength = "r0-7";
			slew-rate = "fast";
			nxp,speed = "200-mhz";
		};
	};
	pinmux_flexpwm4_ch2: pinmux_flexpwm4_ch2 {
		group0 {
			pinmux = <&iomuxc_gpio_emc_04_flexpwm4_pwma2>, //output A
                    <&iomuxc_gpio_emc_05_flexpwm4_pwmb2>; //output B
			drive-strength = "r0-7";
			slew-rate = "fast";
			nxp,speed = "200-mhz";
		};
	};
	pinmux_flexpwm3_ch1: pinmux_flexpwm3_ch1 {
		group0 {
			pinmux = <&iomuxc_gpio_emc_32_flexpwm3_pwmb1>; //output A
			drive-strength = "r0-7";
			slew-rate = "fast";
			nxp,speed = "200-mhz";
		};
	};
};

&flexpwm2_pwm0 {
	status = "okay";
	pinctrl-0 = <&pinmux_flexpwm2_ch0>;
	pinctrl-names = "default";
	nxp,prescaler = <64>;
};

&flexpwm2_pwm2 {
	status = "okay";
	pinctrl-0 = <&pinmux_flexpwm2_ch2>;
	pinctrl-names = "default";
	nxp,prescaler = <64>;
};

&flexpwm4_pwm2 {
	status = "okay";
	pinctrl-0 = <&pinmux_flexpwm4_ch2>;
	pinctrl-names = "default";
	nxp,prescaler = <64>;
};

&flexpwm3_pwm1 {
	status = "okay";
	pinctrl-0 = <&pinmux_flexpwm3_ch1>;
	pinctrl-names = "default";
	nxp,prescaler = <64>;
};
```

mmahadevan108, 2024-11-06:
@Chethan-TS , can we close this issue?

henrikbrixandersen, 2024-12-29:
Closing due to lack of response.

closed 2024-12-29 by henrikbrixandersen