zephyrproject-rtos/zephyr #78056: check_compliance.py test fails if environment have not synced a extra vendors

sylvioalves, 2024-09-05:
**Describe the bug**
Before #77386, `./script/ci/check_compliance.py` would run without the need of extra HAL modules. After that PR, the script now requires that vendors with blobs, as such as Espressif, STM32, Infineon and NXP to be synced locally.
This degrades CI performance as it can't be optimized for custom CIs. 

**To Reproduce**
1) Sync with latest main
2) `west update`
3) Delete one of the hal folder vendors describe above: `rm -rf zephyr/modules/hal/xxxx`
4) `./script/ci/check_compliance.py`

**Expected behavior**
Compliance check to work without any issues.

**Impact**

**Logs and console output**
If stm32 hal is removed for example:
```
warning: undefined symbol ZEPHYR_HAL_STM32_MODULE_BLOBS:

- Referenced at drivers/bluetooth/hci/Kconfig:70:

config BT_STM32WBA
    bool
    default y
    select HAS_STM32LIB
    select BT_HCI_SET_PUBLIC_ADDR
    depends on DT_HAS_ST_HCI_STM32WBA_ENABLED && ZEPHYR_HAL_STM32_MODULE_BLOBS && BT_HCI && BT_DRIVERS
    help
      ST STM32WBA HCI Bluetooth interface
Running KconfigBasicNoModules tests in /home/sylvio/espressif/zephyr/zephyr ...
warning: undefined symbol ZEPHYR_HAL_STM32_MODULE_BLOBS:
```

**Environment (please complete the following information):**
 - OS: Linux
 - 0.16.8
 -  17545c17a508fff54e0f937a76b6abc938c60c44

tejlmand, 2024-09-05:
whether this issue can be configured a bug or not depends one whether projects from the west.yml are expected to be present when running compliance or not.

If it is allowed to remove a project as described, then we will need to create a dummy Kconfig setting (`config ZEPHYR_<name>_MODULE[_BLOBS]`) for each module which could be present.
Such Kconfig might be consider by some as cargo cult, ref: https://github.com/zephyrproject-rtos/zephyr/pull/77419/commits/be477e5efc7652f81911a5d893f5d44d44846d4a

Another downside is that not all Zephyr modules may have a corresponding Kconfig file, because it is possible to have a Zephyr module with CMake files, board roots, bin blobs but not a standard Kconfig file, in which case there will be no clear location for such dummy Kconfig.

One option could be to create the symbols based on all projects from the manifest, cloned or not cloned, but that will just kick the ball further down, because whatabout the usecase where the user removes the project from the manifest (instead of removing it in the filesystem)

jhedberg, 2024-09-05:
> One option could be to create the symbols based on all projects from the manifest, cloned or not cloned, but that will just kick the ball further down, because whatabout the usecase where the user removes the project from the manifest (instead of removing it in the filesystem)

An additional problem that this has, is that it's not possible to say from the manifest whether the module declares any blobs or not. For that you have to have the actual module and look at its `zephyr/module.yml`.

jhedberg, 2024-09-05:
> An additional problem that this has, is that it's not possible to say from the manifest whether the module declares any blobs or not. For that you have to have the actual module and look at its `zephyr/module.yml`.

To be clear, for the blobs Kconfig option there are three possibilities:
1. It's not defined at all (i.e. the module doesn't declare blobs)
2. It's defined but not enabled (the module declares blobs but they haven't been fetched)
3. It's defined and enabled (the module declares blobs and they have been fetched)

jhedberg, 2024-09-05:
@sylvioalves btw, we have this same exact issue with CI in our downstream tree. For now I ended up passing `-e KconfigBasicNoModules` to the compliance script to suppress the warning.

sylvioalves, 2024-09-05:
> @sylvioalves btw, we have this same exact issue with CI in our downstream tree. For now I ended up passing `-e KconfigBasicNoModules` to the compliance script to suppress the warning.

Right, I can do the same here instead of fetching those extra HAL modules.

tejlmand, 2024-09-06:
> An additional problem that this has, is that it's not possible to say from the manifest whether the module declares any blobs or not. For that you have to have the actual module and look at its `zephyr/module.yml`.

That could technically be solved by always declaring `ZEPHYR_<name>_MODULE_BLOBS` with no default (==`n`) for any modules and only modules which actually has blobs both  defined and fetched will result in `default y`.

> It's not defined at all (i.e. the module doesn't declare blobs)

Is this variant important to know ?
And you cannot distinguish this from the module not being present :wink: 

> It's defined and enabled (the module declares blobs and they have been fetched)

I don't think this is correct, iirc it's enough that one blob has been fetched for this symbol to be defined, meaning the value of this setting does not indicate that all the blobs from the module has been fetched, only that, as minimum, one blob has been fetched.

jhedberg, 2024-09-06:
> > It's not defined at all (i.e. the module doesn't declare blobs)
> 
> Is this variant important to know ? And you cannot distinguish this from the module not being present 😉

I'm not opposed to removing it. I had two things in mind originally:
- Less "bloat" in the space of Kconfig symbols
- Ideological issues, i.e. some fully open source module maintainers may not like the implication that there "could" be binary blobs for their module (due to the existence of this Kconfig option for their module)

> I don't think this is correct, iirc it's enough that one blob has been fetched for this symbol to be defined, meaning the value of this setting does not indicate that all the blobs from the module [...]

Sure, I don't disagree here. I guess I should have written more accuratly "and some have been fetched".

jhedberg, 2024-09-11:
@tejlmand I think basing the Kconfig symbol generation on the manifest rather than module.yml might be the better approach here, however to avoid generating the BLOBS symbol for modules that don't have blobs, do you think it might make sense to have a boolean flag in west.yml to indicate up-front whether the module provides blobs or not?

jhedberg, 2024-09-12:
> @tejlmand I think basing the Kconfig symbol generation on the manifest rather than module.yml might be the better approach here, however to avoid generating the BLOBS symbol for modules that don't [...]

After thinking a bit more about this, using west.yml isn't really possible, since the Kconfig option names also depend on what the module declares as its name in module.yml. I created #78316 as a solution that should work, however we may need to improve the documentation as well, to make it clear that modules need to do this. I've seen the arguments regarding "cargo cult" and modules that don't even use Kconfig, however in the absence of a universal solution I think it's better to have at least something that makes sure downstream compliance checking can fully be used again.

tejlmand, 2024-09-12:
> After thinking a bit more about this, using west.yml isn't really possible, since the Kconfig option names also depend on what the module declares as its name in module.yml.

a very good point which I had not considered.

> I created https://github.com/zephyrproject-rtos/zephyr/pull/78316 as a solution that should work, however we may need to improve the documentation as well, to make it clear that modules need to do this.

:+1: 

> I've seen the arguments regarding "cargo cult" and modules that don't even use Kconfig, however in the absence of a universal solution I think it's better to have at least something that makes sure downstream compliance checking can fully be used again.

I agree, and if this is properly reasoned and documented, then I guess it can no longer be described as cargo-cult.
And for module which doesn't use Kconfig, then it's really only an issue if the Zephyr Kconfig tree want to be able to depend on such module. If there are no Kconfig references to the module, then there is no problem.

PR #78316 (2024-09-16): modules: Kconfig: Provide default definitions for modules with blobs
There are HCI drivers which reference the module-specific Kconfig symbols which are auto-generated when the modules are present. Provide default definitions for these symbols in case the modules are missing, so that compliance checking doesn't stumble over seemingly non-existent Kconfig options.

Fixes #78056
closed 2024-09-16 by aescolar