zephyrproject-rtos/zephyr #78066: K_SPINLOCK() on single core soc

tswaehn, 2024-09-05:
I actually do not understand and may need a quick guess, where to look.

this compiles without errors

```
	uint32_t c;

	uint32_t irq = arch_irq_lock();
	c = test();
	arch_irq_unlock(irq);
	printf("incrementing %d\n", c++);

```

but this:

```
	struct k_spinlock my_lock;

	K_SPINLOCK(&my_lock) {
		c = test();
	}

	printf("incrementing %d\n", c++);

```
fails - as far as I understand that should be the same as above. 

```
error: 'c' may be used uninitialized
```

tswaehn, 2024-09-05:
It turned out hat compiling with` -Og` produced the bug above. actually I dont understand.

tswaehn, 2024-09-05:
Actually, shouldn't it be irrelavant which optimization level ?

peter-mitsis, 2024-09-06:
So far, I have been unable to reproduce this.  Which compiler (and version) are you using? Also, what board are you building for?

Thinking aloud ... at its core, K_SPINLOCK is a for-loop construct. _If_ the compiler is generating the warning you described, my initial thoughts are that it is determining that the body of the for-loop might not execute. For that to occur, its local spinlock key ___i.key_ would have a non-initialized value which could then potential be non-zero. However, the way it is constructed, it should have an initial value of 0.

This makes me think that either ...
  a) my understanding of how the spinlock key structure is initialized is flawed, or
  b) there is a problem in the toolchain being used.

tswaehn, 2024-09-06:
Using RISCV toolchain for qemu 

board: qemu_riscv32/qemu_virt_riscv32

my main looks like this:
```
int32_t test(){
  return 42;
}

int main() {
  uint32_t count;

  struct k_spinlock my_lock;

  K_SPINLOCK(&my_lock) { 
    count = test(); 
  }

  while(1) {
  }
}
```

andyross, 2024-09-06:
Likely[1] this is going to be down to -Og disabling whatever pass was folding/eliding/removing the loop test, so it doesn't know for sure that the loop will execute at least once, thus the assignment to `count` inside the K_SPINLOCK() can't be proven to be reached, thus the warning.

Obviously the workaround for the specific code in question is just to initialize `count` at declaration time so the warning can't happen.  That may not be acceptable in all cases, but it's at least something to get you unblocked.

Beyond that... I guess my gut says this isn't worth fixing unless we want to enable a "debug build" at that optimization level as an official feature.  The workaround is trivial.

But if we do, the form would be to refactor the test expression so that it's always guaranteed to be true the first time it's called.

[1] The other possibility is a toolchain bug, but that's pretty rare.

peter-mitsis, 2024-09-09:
Additional info:

A little extra digging suggests that the reported behavior was introduced somewhere in GCC version 12, as I am not seeing it on GCC version 11. Furthermore, I am not seeing this behavior on any optimization level except -Og (I tried using O0, O1, O2, O3, Oz, Os and Og).

I am thinking that this is a toolchain issue, as it makes little sense to me as to why _only_ that optimization should generate the warning/error. In the meantime, my recommendation would be to employ the suggested workaround.

tswaehn, 2024-09-10:
it took me some time to figure out that the issue comes from compiling with -Og. so just to avoid anybody else wastes a lot of time (like me). is it possible to add something to cmake => "please dont use -Og, there might be issues with the riscv toolchain" ?