zephyrproject-rtos/zephyr #78080: mcumgr: `MCUMGR_GRP_OS_BOOTLOADER_INFO` TF-M support

JordanYates, 2024-09-06:
**Is your feature request related to a problem? Please describe.**

The `MCUMGR_GRP_OS_BOOTLOADER_INFO` option is not supported for applications built with TF-M, even though TF-M internally uses MCUboot as the bootloader (for BL2).

https://github.com/zephyrproject-rtos/zephyr/blob/34982bb00b57486aeb83a6cb3bb7433791302aed/subsys/mgmt/mcumgr/grp/os_mgmt/Kconfig#L197-L205 

**Describe the solution you'd like**

`MCUMGR_GRP_OS_BOOTLOADER_INFO` supported for applications built with TF-M.

**Additional context**

The function implementation currently depends heavily on the MCUboot Kconfig options, which exist in a different form in the TF-M build system and are not currently available to the Zephyr application.
https://github.com/zephyrproject-rtos/zephyr/blob/34982bb00b57486aeb83a6cb3bb7433791302aed/subsys/mgmt/mcumgr/grp/os_mgmt/src/os_mgmt.c#L426-L442

nordicjm, 2024-09-06:
Unlikely that will be supported as TF-M has their own saved data function https://git.trustedfirmware.org/plugins/gitiles/TF-M/trusted-firmware-m/+/eb8ff0db7d657b77abcd0262d5bf7f38eb1e1cdc/bl2/src/shared_data.c#166 so a zephyr one cannot be used which saves this data

JordanYates, 2024-09-06:
I'm not sure what the relevance of that is? `MCUMGR_GRP_OS_BOOTLOADER_INFO` (at least currently) simply returns the upgrade mode supported by the bootloader, which is available in the TF-M `MCUBOOT_UPGRADE_STRATEGY` CMake variable if not explicitly set.

nordicjm, 2024-09-06:
Must be dreaming, thought `blinfo_lookup()` was integrated into MCUmgr but apparently not...

nordicjm, 2025-01-31:
It is not possible to get this, the TF-M build system creates that after the build is started, it is not available during configuration time, so cmake cannot access that information unfortunately

nordicjm, 2025-01-31:
See here, configuring sample: `zconf nrf5340dk//cpuapp/ns -DEXTRA_CONF_FILE="overlay-bt.conf" -DSB_CONFIG_BOOTLOADER_NONE=y`:
```
grep -r "OVERWRITE_ONLY" .
grep -r "MCUBOOT_UPGRADE_STRATEGY" .
```
then start the build, cancel, and see:
```
grep -r "MCUBOOT_UPGRADE_STRATEGY" .
./smp_svr/tfm/api_ns/cmake/spe_config.cmake:set(MCUBOOT_UPGRADE_STRATEGY               OVERWRITE_ONLY)
./smp_svr/tfm/CMakeCache.txt:MCUBOOT_UPGRADE_STRATEGY:STRING=OVERWRITE_ONLY
./smp_svr/tfm/CMakeCache.txt://STRINGS property for variable: MCUBOOT_UPGRADE_STRATEGY
./smp_svr/tfm/CMakeCache.txt:MCUBOOT_UPGRADE_STRATEGY-STRINGS:INTERNAL=OVERWRITE_ONLY;SWAP_USING_SCRATCH;SWAP_USING_MOVE;DIRECT_XIP;RAM_LOAD
grep -r "OVERWRITE_ONLY" .
./smp_svr/tfm/build.ninja:  DEFINES = -DBL1_HEADER_SIZE=0x000 -DBL1_TRAILER_SIZE=0x000 -DBL2 -DBL2_HEADER_SIZE=0x400 -DBL2_TRAILER_SIZE=0x400 -DCONFIG_TFM_CONNECTION_POOL_ENABLE -DDAUTH_CHIP_DEFAULT -DMCUBOOT_BUILTIN_KEY -DMCUBOOT_FIH_PROFILE_OFF -DMCUBOOT_IMAGE_NUMBER=2 -DMCUBOOT_OVERWRITE_ONLY -DMCUBOOT_SIGN_EC256 -DNRF5340_XXAA_APPLICATION -DNRF53_SERIES -DNRF_NS_SECONDARY -DNRF_NS_STORAGE -DNRF_SKIP_FICR_NS_COPY_TO_RAM -DOTP_WRITEABLE -DPLATFORM_DEFAULT_BL1 -DPLATFORM_DEFAULT_NV_COUNTERS -DPLATFORM_DEFAULT_OTP -DPLATFORM_NS_NV_COUNTERS=0 -DPS_CRYPTO_AEAD_ALG_GCM -DPS_ENCRYPTION -DSECURE_UART1 -DTFM_DUMMY_PROVISIONING -DTFM_PARTITION_CRYPTO -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_PARTITION_NS_AGENT_TZ -DTFM_PARTITION_PLATFORM -DTFM_PARTITION_PROTECTED_STORAGE -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_SP_LOG_RAW_ENABLED
```

closed 2025-01-31 by nordicjm