zephyrproject-rtos/zephyr #78084: HAL module request: hal_realtek

zjian-zhang, 2024-09-06:
## Origin

This is a repository containing the Realtek SDK, which currently includes the Ameba SOC series. In the future, it will also include drivers for other SOCs in our company.
https://github.com/zjian-zhang/hal_realtek

## Purpose

Reuse an existing hardware abstraction layer from the Ameba-AIoT SDK to support the drivers for AmebaDplus SoCs. 

## Mode of integration

The module should be integrated as an external HAL module. The proposed name is hal_realtek.

## Maintainership

hal_realtek will be maintained by for now:
@zjian-zhang

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/78036

## Description

[ameba-rtos](https://github.com/Ameba-AIoT/ameba-rtos ) is the development framework for Realtek SoCs supported on Linux, which contains the necessary software and tools to develop firmware for the AmebaDplus. This includes register files, peripheral drivers, system startup files, documentation, various utilities, third-party libraries and a toolchain.

The [hal_realtek](https://github.com/zjian-zhang/hal_realtek) is a pared-down version of the ameba-rtos that contains only the components needed for Zephyr; namely, register files and bare metal peripheral drivers. Utilities, toolchains, and other components not needed by Zephyr are excluded. The hal_realtek will be an artifact generated by a GitHub Actions workflow implemented in the ameba-rtos repository.

## Dependencies

none

## Revision

33388add80e47f2130996688d82e8f3afdd20841

## License

Apache-2.0 license

nordicjm, 2024-09-06:
Binary files are not allowed

zjian-zhang, 2024-09-20:
I have deleted the Binary files and will create a new Issue: Binary blobs for adding the binary blobs.

keith-zephyr, 2024-09-25:
I see `hal_realtek` includes some cmsis files.  Can the `hal_realtek` module use the one of the existing CMSIS modules instead?
* https://github.com/zephyrproject-rtos/cmsis 
* https://github.com/zephyrproject-rtos/CMSIS_6

zjian-zhang, 2024-09-25:
> I see `hal_realtek` includes some cmsis files. Can the `hal_realtek` module use the one of the existing CMSIS modules instead?
> 
> * https://github.com/zephyrproject-rtos/cmsis
> * https://github.com/zephyrproject-rtos/CMSIS_6

Thanks for your reply.
1. The `hal_realtek` repository is a copy (not a fork) of the [ameba-rtos repository](https://github.com/Ameba-AIoT/ameba-rtos/tree/master/component/soc/amebadplus/cmsis), which means the entire amebadplus folder was copied.
2. The `hal_realtek` use the same [core_cm55.h](https://github.com/zephyrproject-rtos/cmsis/blob/master/CMSIS/Core/Include/core_cm55.h) as https://github.com/zephyrproject-rtos/cmsis, but our soc has a cache module in core_cm23, so the file [core_cm23_km0.h](https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/cmsis/core_cm23_km0.h) is not identical and should be included.

If `hal_realtek` cannot include some cmsis files, we can remove it, and delete them when copying from ameba-rtos repository. waiting for your reply.

keith-zephyr, 2024-09-25:
> Thanks for your reply.
>
> 1. The `hal_realtek` repository is a copy (not a fork) of the [ameba-rtos repository](https://github.com/Ameba-AIoT/ameba-rtos/tree/master/component/soc/amebadplus/cmsis), [...]

I think it would be preferred to use the existing  https://github.com/zephyrproject-rtos/cmsis module, and remove the CMSIS provided headers from `hal_realtek`.

However, I do see some other HALs pull in a local copy of CMSIS content.  I see the [NXP HAL](https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/CMSIS) also makes a copy of many CMSIS files.  @dleach02  - do you know the history of NXP's inclusion of CMSIS?

zjian-zhang, 2024-09-26:
> I think it would be preferred to use the existing https://github.com/zephyrproject-rtos/cmsis module, and remove the CMSIS provided headers from `hal_realtek`.
>
> However, I do see some other HALs [...]

I have already removed the CMSIS provided headers from hal_realtek.

zjian-zhang, 2024-09-29:
Please help to check again if there are any problems. I have other blobs to submit based on this repository, thanks.

nordicjm, 2024-09-30:
I don't want to see another NXP-like hal added personally, so can you explain what any of this is and why it's needed in your hal:

- https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/app/monitor/ram/monitor_km0.c what is this?
- https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/misc/ameba_ota.c why do you need ota files? How is this useful for zephyr when mcuboot is the bootloader that zephyr uses?
- https://github.com/zjian-zhang/hal_realtek/tree/main/ameba/amebadplus/source/swlib/libnosys what are any of these files, why are any of them needed?
- https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/swlib/log.c zephyr has a logging subsystem, why is there some custom log thing here?
- https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/usrcfg/ameba_bootcfg.c zephyr devices are configured through devicetree, why are you doing it in a .h file?
- https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/usrcfg/ameba_flashcfg.c as above
- https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/usrcfg/include/ameba_userheapcfg.h head is zephyr controlled by Kconfig, so it's definitely not 0
- https://github.com/zjian-zhang/hal_realtek/tree/main/ameba/amebadplus/ld why are there linker files for adding mbedtls functions or c stdio functions when zephyr deals with these?
- https://github.com/zjian-zhang/hal_realtek/tree/main/ameba/common/os_wrapper/source you are using zephyr functions and defines from your hal, your hal should have no knowledge of zephyr at all, it just needs to be the hal

gmarull, 2024-09-30:
For reference: Zephyr glue code is **NOT** allowed in hal repos, otherwise any change in Zephyr can become a multirepo nightmare.

zjian-zhang, 2024-10-02:
> I don't want to see another NXP-like hal added personally, so can you explain what any of this is and why it's needed in your hal:
> 
> * https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/app/monitor/ram/monitor_km0.c what is this?
the whole folder was copied from [ameba-rtos repository](https://github.com/Ameba-AIoT/ameba-rtos/tree/master/component/soc/amebadplus/cmsis), so this file is here. 
> * https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/misc/ameba_ota.c why do you need ota files? How is this useful for zephyr when mcuboot is the bootloader that zephyr uses?
as above.
> * https://github.com/zjian-zhang/hal_realtek/tree/main/ameba/amebadplus/source/swlib/libnosys what are any of these files, why are any of them needed?
do not need them now, they were used for zero warning in our sdk when build with no OS.
> * https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/swlib/log.c zephyr has a logging subsystem, why is there some custom log thing here?
the driver code use some log api like RTK_LOGx, and we do not want to changed to log api from zephyr. 
> * https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/usrcfg/ameba_bootcfg.c zephyr devices are configured through devicetree, why are you doing it in a .h file?
we will use our bootloader, not mucboot, so the ameba_bootcfg.c used for config out bootloader can be here? or every configuration for our bootloader must through devicetree like zephyr.bin?  
> * https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/usrcfg/ameba_flashcfg.c as above
> * https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/usrcfg/include/ameba_userheapcfg.h heap is zephyr controlled by Kconfig, so it's definitely not 0
it was not uesd in zephyr, and I have removed it. 
> * https://github.com/zjian-zhang/hal_realtek/tree/main/ameba/amebadplus/ld why are there linker files for adding mbedtls functions or c stdio functions when zephyr deals with these?
the ld files are symbols of the rom code in asic, and mbedtls or c stdio functions are used in our drivers, the drives are also in the rom, so mbedtls or c stdio functions must in rom. we list all of symbols of rom and provide with  PROVIDE(xxx = xxxx);，so they are all weak symbol, and does not impact zephyr's strong symbol.
> * https://github.com/zjian-zhang/hal_realtek/tree/main/ameba/common/os_wrapper/source you are using zephyr functions and defines from your hal, your hal should have no knowledge of zephyr at all, it just needs to be the hal
I have move the os wrapper to zephyr/soc/realtek/ameba, am i right? our driver need some service from OS, and we implement the glue code to support multi OS.

zjian-zhang, 2024-10-02:
> For reference: Zephyr glue code is **NOT** allowed in hal repos, otherwise any change in Zephyr can become a multirepo nightmare.

I have move the os wrapper to zephyr/soc/realtek/ameba, am i right? I refer to other hal_xxx and implement the glue code here, and now I do not where to put it.

zjian-zhang, 2024-12-02:
Please help to check again if there are any problems. thanks.

zjian-zhang, 2024-12-24:
Sorry to bother.
Please tell me how you will deal with this request and the schedule, and are there any other things I should offer?
Thanks.

zjian-zhang, 2025-02-17:
Hi @zephyrproject-rtos/tsc and @nashif, 

I hope this message finds you well. I wanted to kindly follow up on [Issue #78084](https://github.com/zephyrproject-rtos/zephyr/issues/78084), which was submitted on Sep 6, 2024. This issue reuses an existing hardware abstraction layer from the Ameba-AIoT SDK to support the drivers for AmebaDplus SoCs.

Since the approval of this issue is required for merging [PR #78036](https://github.com/zephyrproject-rtos/zephyr/pull/78036), could the TSC prioritize reviewing this at your earliest convenience?

Please let me know if further details are needed. Thank you for your time and guidance!

zjian-zhang, 2025-05-12:
Please help to check the issue. thanks a lot.

carlescufi, 2025-08-06:
@zjian-zhang the vote today was postponed due to these statements included in many of the source code files, which are incompatible with the Apache v2 License:

```
  * This module is a confidential and proprietary property of RealTek and
  * possession or use of this module requires written permission of RealTek.
```

zjian-zhang, 2025-08-14:
> > [@zjian-zhang](https://github.com/zjian-zhang) any updates on this? are you able to remove the text I mentioned above?
> 
> Yes, these contents conflict with Apache 2.0. We plan to remove them. Due to a colleague being on leave, this part of the update hasn't been completed yet. We expect to submit the update tomorrow.

The text that was incompatible with the Apache v2 License has been removed. Please review the changes and let me know if you need any further modifications. Thank you for your attention to this matter.

zjian-zhang, 2025-08-15:
> [@zjian-zhang](https://github.com/zjian-zhang) Why isn't the HAL repo hosted by https://github.com/Ameba-AIoT?

When I submitted PR https://github.com/zephyrproject-rtos/zephyr/pull/78036 on September 6, 2024, I did not have administrator privileges for https://github.com/Ameba-AIoT. Therefore, the repository was not created under https://github.com/Ameba-AIoT at that time. Current and future submissions will be made through Ameba-AIoT. For example, https://github.com/zephyrproject-rtos/zephyr/pull/91933 is a PR provided from https://github.com/Ameba-AIoT/zephyr/tree/rtl8721d. After the TSC creates the hal_realtek repository in the Zephyr project, we will use https://github.com/Ameba-AIoT/nuwa_hal_realtek for tracking and maintenance.

zjian-zhang, 2025-09-15:
Please let us know if there are any additional concerns or feedback we should address. We greatly appreciate your time and review, and we look forward to landing this HAL repository in the mainline at your earliest convenience. @carlescufi

ubieda, 2025-09-17:
> After the TSC creates the hal_realtek repository in the Zephyr project, we will use [Ameba-AIoT/nuwa_hal_realtek](https://github.com/Ameba-AIoT/nuwa_hal_realtek) for tracking and maintenance.

@zjian-zhang could you please clarify what's the model repo to be submitted for review? If it's [Ameba-AIoT/nuwa_hal_realtek](https://github.com/Ameba-AIoT/nuwa_hal_realtek), it needs to align with the HAL module requirements for submission (including the integration approach for binary blobs) before getting approved.

zjian-zhang, 2025-09-30:
> > Questions/comments sent directly to Realtek:
> > 
> > 1. Is the individual that submitted the change, [@zjian-zhang](https://github.com/zjian-zhang), employed by Realtek? The TSC just wants to clarify who will be responsible for maintaining the hal_realtek module.
> > 2. Earlier versions of the submission included copyright language that is not compatible with open source licenses ([example](https://github.com/zjian-zhang/hal_realtek/blob/e3d9fcdfe0f8e65d5ba25bc36196f3b2f3047f2b/ameba/amebad/source/fwlib/include/ameba.h#L1-L8)). Later versions of the module removed the text indicating that the code is proprietary property of Realtek.  I want to make sure Realtek approved this change to the copyright language and that Realtek has generally approved releasing all this code under the Apache 2.0 license.
> > 3. The [Zephyr issue](https://github.com/zephyrproject-rtos/zephyr/issues/78084) requesting the hal indicates that the origin of the code is [zjian-zhang/hal_realtek](https://github.com/zjian-zhang/hal_realtek), however, there is a later comment on the Zephyr issue that indicates the origin of the code will be [Ameba-AIoT/nuwa_hal_realtek](https://github.com/Ameba-AIoT/nuwa_hal_realtek) .  Can Realtek clarify what the correct origin will be?
> > 4. If [Ameba-AIoT/nuwa_hal_realtek](https://github.com/Ameba-AIoT/nuwa_hal_realtek)  is the origin of this code, that repository needs updates to fix the copyright notices and also adds binary blobs that don't follow Zephyr's guidelines
> 
> thank you [@keith-zephyr](https://github.com/keith-zephyr) for the summary.
> 
> In addition to that, the source has code the reimplements and duplicate C library functions implmented in zephyr and external C libraries, see https://github.com/zjian-zhang/hal_realtek/blob/main/ameba/amebadplus/source/swlib/sscanf_minimal.c. Why is this necessary to be part of the HAL?

sscanf_minimal.c was introduced to reduce stack usage during function calls. This is achieved by supporting only a subset of format specifiers. Additionally, the `_sscanf_ss` function—used by the code in [Ameba-AIoT/nuwa_hal_realtek](https://github.com/Ameba-AIoT/nuwa_hal_realtek) (i.e., the .a binary blobs that will be provided later)—depends on this minimal implementation.

zjian-zhang, 2025-11-10:
> From the description:
> 
> > The [hal_realtek](https://github.com/zjian-zhang/hal_realtek) is a pared-down version of the ameba-rtos that contains only the components needed for Zephyr; namely, register files and bare metal peripheral drivers. Utilities, toolchains, and other components not needed by Zephyr are excluded.
> 
> That does not seem to be correct, as there still are: CMSIS Cortex-M33 Core Peripheral Access Layer Header File - core_cm33.h (core_cm33_km4.h) swlib - what contains various tools that are unlikely to be needed, including sscanf_minimal, logging, malloc/free/printf wrapper for gcc compiler
> 
> I do not think a HAL should have this kind of components, and we should not open the door to components like this.

Thank you for your feedback. Please allow me to clarify the points raised:

- **Custom Cortex-M33 Core**: The AmebaD SoC uses a customized version of the Cortex-M33 core (with a 2-way D-Cache), which requires specific modifications to the standard CMSIS header. Hence, the core_cm33_km4.h file is necessary and cannot be replaced by the generic core_cm33.h.
- **sscanf_minimal.c**: As previously noted, this minimal implementation was introduced to reduce stack usage and supports only a limited set of format specifiers. However, since the current HAL integration does not yet utilize `_sscanf_ss`, we will remove `sscanf_minimal.c` from the HAL module for now to align with Zephyr’s expectations.
- **Logging**: The logging utilities are actively used within the HAL drivers themselves (e.g., for debug and error reporting during driver initialization or operation). Therefore, they are essential to the functionality of `hal_realtek` and cannot be removed at this stage.
- **malloc/free/printf wrappers**: These have already been removed from the HAL module. The `printf` wrapper was originally added to prevent log interleaving in multi-threaded scenarios, but we acknowledge that such functionality belongs at a higher layer and is not appropriate for a HAL.

We aim to keep hal_realtek as minimal and focused as possible—containing only register definitions and bare-metal peripheral drivers—and appreciate your guidance in helping us meet Zephyr’s architectural standards.

ZhiyuanTang17, 2025-11-19:
> [@zjian-zhang](https://github.com/zjian-zhang) There is also the [rtkconnectivity](https://github.com/rtkconnectivity) GitHub organization, which maintains the SDK for the Bee series MCUs ([RTL87X2G](https://www.realmcu.com/en/Home/Product/RTL8762G-RTL877xG-Series) and [RTL8752H](https://www.realmcu.com/en/Home/Products/RTL8752H-Series)), if I understand correctly.
> 
> Will the content of [rtkconnectivity/hal_realtek](https://github.com/rtkconnectivity/hal_realtek) repo eventually be merged into the planned Zephyr `hal_realtek` repo alongside the `ameba` code? If not, we should probably consider using more distinct names to avoid confusion for Zephyr users.

Hi @martinjaeger and Zephyr Memebers:

Thank you for your mention on our team's work. I am Zhiyuan, a software development engineer in Realtek's Bluetooth department, and I am currently responsible for maintaining [rtkconnectivity/hal_realtek](https://github.com/rtkconnectivity/hal_realtek) repo. Our team has been involved in Zephyr downstream development for quite some time, and we are also eager for the Bee series SoCs to be integrated into the Zephyr mainline.

Both the Ameba and Bee series SoCs are products of Realtek, with the Ameba series focusing on Wi-Fi and the Bee series emphasizing Bluetooth. @zjian-zhang is our colleague from the Ameba team. We hope to share the same **hal_realtek** repo.

Currently, our team is preparing the HAL content (reviewing licenses, code coverity, etc.). Once we are ready, we will also submit a GitHub issue to request that Zephyr members and the TSC review it. At that time, we will add a 'bee' directory in the root of hal_realtek to manage our HAL code alongside 'ameba'.

If you have any questions, please feel free to bring them up.

P.S. Regarding the integration of the Realtek Bee series into the Zephyr mainline, we sincerely hope to receive feedback and suggestions from the Zephyr community.

Zhiyuan

nashif, 2025-11-22:
note that this repo was created as a fork, so you might want to think about how it gets updated going forward.

zjian-zhang, 2025-11-22:
I’d like to kindly clarify the origin of the hal_realtek repository.

The correct upstream source is **not** `rtkconnectivity/hal_realtek`, but rather our official repository: https://github.com/Ameba-AIoT/nuwa_hal_realtek

The current `zephyrproject-rtos/hal_realtek` appears to have been mistakenly marked as “forked from rtkconnectivity/hal_realtek.”

We would appreciate it if this could be corrected to point to `Ameba-AIoT/nuwa_hal_realtek`, as outlined in the original request: https://github.com/zephyrproject-rtos/zephyr/issues/78084#issue-2509888107

zjian-zhang, 2025-11-25:
I’m trying to contribute the `Ameba-AIoT/nuwa_hal_realtek` to Zephyr. However, the repository https://github.com/zephyrproject-rtos/hal_realtek **appears to be empty** and **does not show a "Fork" button**, making it **impossible** to submit a pull request via the standard GitHub workflow.

Could you please help check why the "Fork" button is missing for this repository? @nashif

martinjaeger, 2025-11-25:
You should be able to add the new repo as an additional remote to your local repo and push the current state directly to that remote. Afterwards you create fork of the repo in the Zephyr organization and use pull requests for further changes.

zjian-zhang, 2025-11-25:
> You should be able to add the new repo as an additional remote to your local repo and push the current state directly to that remote. Afterwards you create fork of the repo in the Zephyr organization and use pull requests for further changes.

I understand the proposed flow, but forking isn’t possible—the repository at https://github.com/zephyrproject-rtos/hal_realtek doesn’t show a "Fork" button.

zjian-zhang, 2025-11-26:
Thank you for your guidance! To better understand the process, I’d like to confirm a few points based on Zephyr’s documentation:

1. According to [Individual Roles & Responsibilities](https://docs.zephyrproject.org/latest/develop/modules.html#individual-roles-responsibilities), the **module owner** is effectively the **maintainer** of the module repository—is this understanding correct?
2. For pull requests targeting `zephyrproject-rtos/hal_realtek`, after approval by the module owner, the PR would still need to be merged by the Release Engineering team (not the module owner), consistent with [the Roles / Permissions policy](https://docs.zephyrproject.org/latest/project/project_roles.html#roles-permissions)—is that accurate?
3. Even if the `zephyrproject-rtos/hal_realtek` repository were set up as a GitHub fork, would synchronization with upstream still require a pull request? The [Requirements for allowed practices](https://docs.zephyrproject.org/latest/develop/modules.html#requirements-for-allowed-practices) clearly state:
> “Changes to the main branch of a module repository, including synchronization with upstream code base, may only be applied via pull requests.”
>
> Since using GitHub’s “Sync fork” button updates the branch directly **without** creating a PR, such an approach would not comply with this policy—correct?
4. If the repo were a proper fork, who would be responsible for initiating the sync from upstream to the downstream (here the downstream is `zephyrproject-rtos/hal_realtek`)—the Administrator, the Module Owner, or the Merger (i.e., Release Engineering team)?
5. Is there a plan to automate this synchronization via a script or GitHub Actions workflow that could periodically update the module from upstream—potentially leveraging GitHub’s “Sync fork” capability if the repository is set up as a proper fork?

fabiobaltieri, 2025-11-26:
> 1. According to [Individual Roles & Responsibilities](https://docs.zephyrproject.org/latest/develop/modules.html#individual-roles-responsibilities), the **module owner** is effectively the **maintainer** of the module repository—is this understanding correct?

That's right.

>     2. For pull requests targeting `zephyrproject-rtos/hal_realtek`, after approval by the module owner, the PR would still need to be merged by the Release Engineering team (not the module owner, which consistent with [the Roles / Permissions policy](https://docs.zephyrproject.org/latest/project/project_roles.html#roles-permissions)) —is that accurate?

No, you'll be able to merge PR against hal_realtek yourself, but the release team is going to be the one merging the PRs in west.yml that that changes what the module points to.

>     4. Even if the `zephyrproject-rtos/hal_realtek` repository were set up as a GitHub fork, would synchronization with upstream still require a pull request? The [Requirements for allowed practices](https://docs.zephyrproject.org/latest/develop/modules.html#requirements-for-allowed-practices) clearly state:> 

Well if it was setup as a fork and you were tracking the upstream branch directly you could use the "sync fork" button. If it's not setup as a fork you need to open pull requests but also https://github.com/zephyrproject-rtos/infrastructure/pull/266 is in progress and once that's done it'll keep the repo in sync automatically regardless of whether it's been created as a fork or not. So you could do development on your upstream with your own process and then changes will get pulled automatically if that's what you want.

If the plan is to use `hal_realtek` to host code from two different upstreams though that's obviously not going to work and you'll have to do PRs instead.

> > “Changes to the main branch of a module repository, including synchronization with upstream code base, may only be applied via pull requests.”
> > Since using GitHub’s “Sync fork” button updates the branch directly **without** creating a PR, such an approach would not comply with this policy—correct?

That does not sound right to me, there's few modules that are synced without PRs, github.com/zephyrproject-rtos/sof/ is what I use as reference usually.

>     4. If the repo were a proper fork, who would be responsible for initiating the sync from upstream to the downstream (here the downstream is `zephyrproject-rtos/hal_realtek`)—the Administrator, the Module Owner, or the Merger (i.e., Release Engineering team)?

See my comment above, module owner  but ideally it'd be automated.

>     5. Is there a plan to automate this synchronization via a script or GitHub Actions workflow that could periodically update the module from upstream—potentially leveraging GitHub’s “Sync fork” capability if the repository is set up as a proper fork?

Yeah https://github.com/zephyrproject-rtos/infrastructure/pull/266, does not use github "sync fork" so it'll work even if the repository has been created as a standalone thing.

zjian-zhang, 2025-11-27:
Understood—so the `zephyrproject-rtos/hal_realtek` repository does **not** need to be set up as a GitHub fork. Thank you for your reply! @fabiobaltieri 

> I can do the initial push from Ameba-AIoT/nuwa_hal_realtek if needed

Could you please perform the initial push from `Ameba-AIoT/nuwa_hal_realtek`, using the `rtl8721d` (default) branch, when convenient? @nashif 

After the repository is populated, I’ll update the manifest file with the correct repository URL and maintainer information.