zephyrproject-rtos/zephyr #78099: Crash/segfault on all nrf5340 bsim simulations during IPC init

aescolar, 2024-09-06:
**Describe the bug**
Most simulations are crashing for the nrf5340_bsim/nrf5340/cpuapp target during IPC init due to a segfault

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure you can build the bsim targets: https://docs.zephyrproject.org/latest/boards/native/nrf_bsim/doc/nrf52_bsim.html#building-and-running
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf5340bsim/nrf5340/cpuapp  -DAPP_DIR=../samples/bluetooth/peripheral_hr/ ../share/sysbuild
3. ninja
4. valgrind ./zephyr/zephyr.exe -nosim
5. See error

**Expected behavior**
No crashes

**Impact**
main CI failing for all BT development

**Logs and console output**
```
==3262026==    at 0x8054006: pbuf_init (pbuf.c:68)
==3262026==    by 0x8053C4D: icmsg_open (icmsg.c:273)
==3262026==    by 0x80531BD: open (ipc_icbmsg.c:969)
==3262026==    by 0x80512F3: ipc_service_open_instance (ipc_service.c:38)
==3262026==    by 0x807DA95: bt_ipc_open (ipc.c:323)
==3262026==    by 0x805B69D: bt_hci_open (bluetooth.h:113)
==3262026==    by 0x8061621: bt_enable (hci_core.c:4396)
==3262026==    by 0x804B46A: main (main.c:193)
==3262026==  Address 0x20070004 is not stack'd, malloc'd or (recently) free'd
```

**Environment (please complete the following information):**

 - OS: Linux, CI, Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): Host gcc 11/12
 - Commit SHA or Version used: main as of now 34982bb00b57486aeb83a6cb3bb7433791302aed

**Additional context**

aescolar, 2024-09-06:
CC @dchat-nordic

aescolar, 2024-09-06:
CC @dchat-nordic @Thalley @cvinayak @kruithofa given the amount of macrology around this initialization, It is looking to me that reverting this commit first and then sending calmly a fix is the best way forwards.
So far it looks like a fix for this is hardly trivial, so I don't like hurrying it as a hotfix.

aescolar, 2024-09-06:
Background:
Unfortunately changes in DTS are not triggering bsim tests today which resulted in this break not being caught.

The reason for the break is that the icbmsg backend is not yet supported in the nrf5340bsim target. Its configuration initialization is not proper.

aescolar, 2024-09-06:
CC @doki-nordic @carlocaione

Thalley, 2024-09-06:
Thank you for investigating. I concur with reverting the commit

aescolar, 2024-09-09:
Lowering priority and removing regression as hotfix got merged

closed 2024-09-26 by nashif