zephyrproject-rtos/zephyr #78113: stm32h7: Cannot connect with J-Link after flashing blinky

nono313, 2024-09-06:
**Describe the bug**

If I build the blinky sample for nucleo_h723zg using stm32cubeprogrammer runner, and then I connect a J-Link to flash it again through the MIPI10 connector: west flash fails.

**To Reproduce**
1. Build blinky for nucleo_h723zg
2. Flash using jlink runner
3. Power cycle
4. Try to flash again using jlink runner
5. Try to connect using `JLinkExe -device STM32H723ZG -if swd -speed 4000`
6. See error

**Expected behavior**
We should be able to flash multiple times and to connect using J-Link in any circumstances.

**Impact**
J-Link in unusable, even using hardware reset (with `Rst 2`
ST-Link works if I force hardware reset.

**Logs and console output**

```
J-Link>connect
Device "STM32H723ZG" selected.

Connecting to target via SWD
ConfigTargetSettings() start
ConfigTargetSettings() end - Took 6us
InitTarget() start
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
Can not attach to CPU. Trying connect under reset.
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
Timeout while waiting for CPU to halt.
Connecting to CPU via connect under reset failed.
InitTarget() end - Took 659ms
Connect failed. Resetting via Reset pin and trying again.
ConfigTargetSettings() start
ConfigTargetSettings() end - Took 18us
InitTarget() start
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
Can not attach to CPU. Trying connect under reset.
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
Timeout while waiting for CPU to halt.
Connecting to CPU via connect under reset failed.
InitTarget() end - Took 656ms
Error occurred: Could not connect to the target device.
For troubleshooting steps visit: https://wiki.segger.com/J-Link_Troubleshooting
```

**Environment (please complete the following information):**

 - OS: Linux Ubuntu 24.04.1
 - Toolchain: Zephyr SDK 0.16.5-1
 - Zephyr 3.7

**Additional context**

If I connect with STM32CubeProgrammer without hardware reset (for exemple selecting software reset), I also get an error (`No STM32 target found! If your product embeds Debug Authentication, please perform a discovery using Debug Authentication`)

If I add `HAL_DBGMCU_EnableDBGSleepMode();` in the main() of the blinky app, I am able to connect and I get this logs:

```
$ JLinkExe -device STM32H723ZG -if swd -speed 4000
SEGGER J-Link Commander V7.98f (Compiled Aug 29 2024 10:19:38)
DLL version V7.98f, compiled Aug 29 2024 10:19:12

Connecting to J-Link via USB...O.K.
Firmware: J-Link V12 compiled Aug 22 2024 13:27:55
Hardware version: V12.00
J-Link uptime (since boot): 0d 02h 52m 17s
S/N: 602004954
License(s): RDI, FlashBP, FlashDL, JFlash, GDB
USB speed mode: High speed (480 MBit/s)
VTref=3.296V

Type "connect" to establish a target connection, '?' for help
J-Link>connect
Device "STM32H723ZG" selected.

Connecting to target via SWD
ConfigTargetSettings() start
ConfigTargetSettings() end - Took 6us
InitTarget() start
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
Error while checking CPU state.
Can not attach to CPU. Trying connect under reset.
SWD selected. Executing JTAG -> SWD switching sequence.
DAP initialized successfully.
InitTarget() end - Took 1.56s
Found SW-DP with ID 0x6BA02477
DPIDR: 0x6BA02477
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[3]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x84770001, ADDR: 0x00000000)
AP[1]: AHB-AP (IDR: 0x84770001, ADDR: 0x01000000)
AP[2]: APB-AP (IDR: 0x54770002, ADDR: 0x02000000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FE000
CPUID register: 0x411FC272. Implementer code: 0x41 (ARM)
Cache: L1 I/D-cache present
Found Cortex-M7 r1p2, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ E00FE000
[0][0]: E00FF000 CID B105100D PID 000BB4C7 ROM Table
ROMTbl[1] @ E00FF000
[1][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[1][1]: E0001000 CID B105E00D PID 000BB002 DWT
[1][2]: E0002000 CID B105E00D PID 000BB00E FPB-M7
[1][3]: E0000000 CID B105E00D PID 000BB001 ITM
[0][1]: E0041000 CID B105900D PID 001BB975 ETM-M7
[0][2]: E0043000 CID B105900D PID 004BB906 CTI
I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way
D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
Memory zones:
  Zone: "Default" Description: Default access mode
Cortex-M7 identified.
```

erwango, 2024-09-09:
@nono313 We're not using J-Link, so it will be hard to provide support for us. I'm letting the point open, but assigning back to you.

erwango, 2024-09-09:
@nono313 Are we sure this problem is zephyr specific and could not be seen using a STM32Cube project ?

nono313, 2024-09-09:
@erwango yes when I build a simple project generated with Cube MX I do not have any issue with St-link in software reset or with j-Link

erwango, 2024-09-17:
See https://github.com/zephyrproject-rtos/zephyr/pull/78567

erwango, 2024-09-20:
@nono313 Did you had a chance to test https://github.com/zephyrproject-rtos/zephyr/pull/78567 ?
NB: It requires `CONFIG_STM32_ENABLE_DEBUG_SLEEP_STOP=y`

nono313, 2024-09-20:
@erwango I just checked and it works well with the Kconfig value. What I don't understand is why this is needed, knowing as I do not do anything to put the MCU in sleep or stop mode.

erwango, 2024-09-20:
> @erwango I just checked and it works well with the Kconfig value. What I don't understand is why this is needed, knowing as I do not do anything to put the MCU in sleep or stop mode.

I didn't checked but I'd expect it goes to "sleep" mode automatically in idle.

nono313, 2024-09-20:
Power is not a topic at all on my project so if some low power features are enabled without setting anything explicitely that may be an issue.

erwango, 2024-09-20:
> Power is not a topic at all on my project so if some low power features are enabled without setting anything explicitely that may be an issue.

If I'm reading H7 programming manual (PM0253.pdf) correctly, this is the default C-M7 behavior:
> The wait for interrupt instruction, WFI, causes immediate entry to sleep mode unless the wakeup condition is true [..] When the processor executes a WFI instruction it stops executing instructions and enters sleep mode.

nono313, 2024-09-20:
Without enabling any power management stuff I wouldn't expect a WFI to be called but maybe I'm wrong there.

PR #78567 (2024-09-24): stm32h7: Allows flashing with j-link
Fixes #78113
closed 2024-09-24 by mmahadevan108