zephyrproject-rtos/zephyr #78129: Autocomplete, submit partial string to use with SHELL_DYNAMIC_CMD_CREATE

koffes, 2024-09-08:
**Is your enhancement proposal related to a problem? Please describe.**

The autocomplete function in the shell is a great feature, but does rely on the system having a static list of all possible commands. If the commands are few, this is fine.
However, if you want to autocomplete e.g. a list of files on an SD card, this requires a huge array of `char paths_and_filenames[MAX_PATHS][MAX_PATH_LEN]`

**Describe the solution you'd like**

When autocompleting filepaths, the function `static void <file_paths_get>(size_t idx, struct shell_static_entry *entry)` is used. The idx parameters iterates from 0 to N and all filepaths must be provided. If this function in addition could take the partial string which is already entered (such as "/foo/bar/") then just the foo folder with subfolders would have to be iterated on an SD card to generate the autocomplete strings.
Alternatively, the user function could just provide the files and folders under "/foo/bar/" and no subfolders to save time.

FYI @nordic-krch

jakub-uC, 2024-09-09:
@koffes  but the shell has already the possibility to autocomplete dynamic commands. We are using this feature for example in the LOG subsystem. Or maybe I didn't understand your comment?

jakub-uC, 2024-09-09:
no, autocomplete will also work on dynamic command. The only catch is that there are no root dynamic commands.

jakub-uC, 2024-09-09:
[image]

nordic-krch, 2024-09-10:
The issue is that dynamic command function when doing the autocomplete does not get current partial command.

Dynamic command is get:
https://github.com/zephyrproject-rtos/zephyr/blob/cb6c78491027925c86c70946abeaf04145c9fc4d/subsys/shell/shell.c#L338
And then it is checked if it matches current partial command:
https://github.com/zephyrproject-rtos/zephyr/blob/cb6c78491027925c86c70946abeaf04145c9fc4d/subsys/shell/shell.c#L340

So in case of files if your partial file path is `/foo/f` `z_shell_cmd_get` must still return all files in the file system because it is not aware of that current partial path. I wonder if we extend `z_shell_cmd_get` and dynamic command getter to also get `incompl_cmd` as one of the parameters then in case of file paths that could be used to limit files returned by the getter.