zephyrproject-rtos/zephyr #78156: The number of isr_exit in the trace log is less than the number of isr_enter

qhe952823334, 2024-09-09:
I am using Zephyr v3.7 on the RISC-V 64 platform with tracing, and analyzing the logs using Babeltrace and Trace Compass. I have observed that the number of isr_exit events is frequently less than the number of isr_enter events. I can confirm that my code does not remain in any interrupt handler for an extended period, so I suspect that isr_exit is under-recorded.
[image]

 Additionally, I noticed that all instances where isr_exit is missing occur around thread_switch_in/out events

My code runs without any errors.  Is this a normal phenomenon?

alexpaschoaletto, 2024-09-27:
Although not really in the same circumstance as you, I have encountered problems with asymmetry between thread switch events as well. It is currently under investigation. You might want to read #76057 to grab the details. [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/76057#issuecomment-2236545540) is what solved in my case.

nashif, 2025-03-14:
dupljicate of https://github.com/zephyrproject-rtos/zephyr/issues/76057

closed 2025-03-14 by nashif