zephyrproject-rtos/zephyr #78157: deferred logging blocks the whole log system

andreas-chmielewski-gcx, 2024-09-09:
**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

zephyr-version:
VERSION_MAJOR = 3
VERSION_MINOR = 5
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =

prj.conf
CONFIG_LOG=y
CONFIG_LOG_MODE_DEFERRED=y
CONFIG_LOG_SPEED=n
CONFIG_LOG_BUFFER_SIZE=50000
CONFIG_LOG_PROCESS_THREAD_SLEEP_MS=10

mmahadevan108, 2024-09-10:
@andreas-chmielewski-gcx is this issue reproducible with the latest Zephyr upstream `main` branch.

andreas-chmielewski-gcx, 2024-09-30:
I can not test this easily as we are not using the latest zephyr version but v3.5.99-ncs1-1 (using nrf sdk)

jussivastimo, 2024-10-15:
I have same issue with v3.7.0-3885-g7773fe8dd60

closed 2024-12-30 by github-actions[bot]