zephyrproject-rtos/zephyr #78169: Sensing subsystem: SPI timeout when reading data from sensor

mimok, 2024-09-09:
**Describe the bug**

I'm using a Thingy91 board (nrf9160) and I want to read samples from the on-board ADXL362 acceleremoter using the new sensing subsystem. I updated my dts overlay accordingly.

Every time, the subsystem tries to read a sample from the sensor, there is a communication timeout on the SPI bus.

**To Reproduce**

I believe this behavior can be reproduced with any  *real* board (not the native simulator) and any *real* accelerometer. 

**Expected behavior**

Data should be read from the sensor.

**Impact**

Sensing subsystem is not working properly on my device and i cannot use it.

**Origin of the bug**

By digging into the code, i can see that the function `phy_3d_sensor_submit()` is a blocking function while it should be a non blocking function as explained in the [documentation ](https://docs.zephyrproject.org/latest/hardware/peripherals/sensor/index.html#implementing-read-and-decode). Indeed, it calls the blocking  `sensor_sample_fetch_chan()` function. 

In addition, there is a [`TODO` comment](https://github.com/zephyrproject-rtos/zephyr/blob/468eb56cf242eedba62006ee758700ee6148763f/subsys/sensing/sensor_mgmt.c#L288) in the file `sensor_mgmt.c` showing that the problem is actually known (or at least expected).

I did a quick test by putting the call to `sensor_read()` in a work executed by the system work queue and it works. That shows that the problem actually comes from calling a blocking function from the ISR handler `sensing_sensor_polling_timer()`.

I know this part of the code has been updated between v3.6.0 (that i'm using) and v3.7.0 but i believe the problem is still there.

**Environment (please complete the following information):**

I'm using a Thingy91 board (nrf9160) with sensor ADXL362.
I'm using Nordic SDK v2.7.0 based on Zephyr OS v3.6.0

closed 2024-11-25 by github-actions[bot]

mimok, 2024-12-03:
@dball387 In my opinion, the sensing subsystem is not "production ready". So I developped my own alternative to the sensing subsystem using the old sensor API and the system work queue. I'm only using few sensors therefore it was not very complex.

yperess, 2025-01-28:
@mimok the sensing subsystem is not production ready. I'll bring this up at the next sensor working group

mimok, 2025-01-29:
@yperess, maybe it would be better to not merge in the main branch features that are not ready tu use or at least put a warning in the documentation (this is not a criticism, just my opinion :-) ).

yperess, 2025-01-30:
> [@yperess](https://github.com/yperess), maybe it would be better to not merge in the main branch features that are not ready tu use or at least put a warning in the documentation (this is not a criticism, just my opinion :-) ).

I agree, there was a discussion a while back in the process working group and in the TSC about how we can add maturity labels to the doxygen for APIs. I'm not sure what happened with that, but it's worth bringing up again.