zephyrproject-rtos/zephyr #78177: NXP: driver: nxp,flexio-spi: Lockup / bus faults

matt-wood-ct, 2024-09-09:
**Describe the bug**
`nxp,flexio-spi` appears to not work on the imxrt10xx series, I had tried on my actual project board (imxrt1040 based) to use it in place of the LPSPI to avoid some undesirable behaviour from the HW SPI, but using the flexio version of the driver simply did not boot on my board. After some stepping through on a JTAG debugger I found it was stuck in an endless ISR loop on the first attempted SPI transaction.

After this issue I went to the basic `tests/drivers/spi/spi_controller_peripheral` and wrote a quick overlay for the `teensy41` board to validate if the driver works in a basic setup on supported imxrt based hardware. With the attached overlay [teensy41.overlay.txt](https://github.com/user-attachments/files/16931052/teensy41.overlay.txt) which replaces lpspi3 with a flexio clone the test faults, log:  
```
*** Booting Zephyr OS build v3.7.0-1712-geaf92a1494cc ***
Running TESTSUITE spi_controller_peripheral
===================================================================
START - test_basic
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x20300000
E: r0/a1:  0x202800f0  r1/a2:  0x00000001  r2/a3:  0xc0000000
E: r3/a4:  0x20300000 r12/ip:  0x0000ea60 r14/lr:  0x60006ced
E:  xpsr:  0x6100006b
E: Faulting instruction address (r15/pc): 0x6000bbae
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20280348 (idle)
E: Halting system

```

Also tried on latest main branch, same result:  
```
*** Booting Zephyr OS build v3.7.0-2409-ga46cef4422d9 ***
Running TESTSUITE spi_controller_peripheral
===================================================================
START - test_basic
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x20300000
E: r0/a1:  0x202800f0  r1/a2:  0x00000001  r2/a3:  0xc0000000
E: r3/a4:  0x20300000 r12/ip:  0x0000ea60 r14/lr:  0x60006ced
E:  xpsr:  0x6100006b
E: Faulting instruction address (r15/pc): 0x6000bbae
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20280348 (idle)
E: Halting system
```

**To Reproduce**

1. Add overlay to `tests\drivers\spi\spi_controller_peripheral\boards`
2. Build for `teensy41` target
3. Flash using teensy tool
4. View logs on UART output (pin no1)

**Expected behavior**

**Impact**
FlexIO SPI not usable on all IMXRT10xx platforms

**Logs and console output**
See above

**Environment (please complete the following information):**

 - Windows
 - Toolchain Zephyr SDK
 - a46cef4422d9

**Additional context**
On my imxrt1040 based custom board the SPI driver gets stuck while initialising an SPI NOR flash (`jedec,spi-nor`) on the first SPI `transact` call. It does not fault but just gets stuck here and never makes any progress, it doesn't even timeout:  
[image]

matt-wood-ct, 2024-09-10:
Update, I noticed I made a mistake in the overlay file on the teensy test, I referenced the wrong flexio bank, updated overlay:
[teensy41.overlay.txt](https://github.com/user-attachments/files/16941469/teensy41.overlay.txt)  
With this corrected the test still bus faults but I do see some clock and MOSI activity:

[image]

mmahadevan108, 2025-02-17:
@matt-wood-ct , I have started to look at this issue. I noticed that spi_controller_peripheral test has not been enabled for FlexIO-SPI. Have you tried `tests/drivers/spi/spi_loopback/`. This has been tested on `mimxrt1064_evk`.

mmahadevan108, 2025-02-24:
@matt-wood-ct , this PR https://github.com/zephyrproject-rtos/zephyr/pull/86166 should address the Flexio-SPI related failures that you were seeing.

mmahadevan108, 2025-02-25:
@matt-wood-ct, I tested with the SPI loopback test on MIMXRT1040-EVK board with PR https://github.com/zephyrproject-rtos/zephyr/pull/86166. The test works successfully.

matt-wood-ct, 2025-02-25:
> [@matt-wood-ct](https://github.com/matt-wood-ct), I tested with the SPI loopback test on MIMXRT1040-EVK board with PR [#86166](https://github.com/zephyrproject-rtos/zephyr/pull/86166). The test works successfully.

Awesome, great work man

closed 2025-02-27 by kartben