zephyrproject-rtos/zephyr #78189: PRIq macro incorrectly converting Qs of form .0X, .00X, etc.

laurenmurphyx64, 2024-09-09:
**Describe the bug**

When given a fixed-point number approximating a decimal of the form ".0X" e.g. "23.08" / "23.008" instead of giving the correct approximation, the PRIq macro gives a decimal of the form ".X" e.g. "23.79986" / "23.7995"

- Discovered on `frdm_k64f`, reproducible on all qemus
- Identical result with negative numbers

```
-23.08 -> -23.79986
.008 -> 0.7995
```

- (TODO: Test if regression)

**To Reproduce**

```
#include <stdio.h>
#include <zephyr/types.h>
#include <zephyr/dsp/print_format.h>
#include <zephyr/sys/util.h>
#include <math.h>

#define SHIFT 16
#define RES 100

int main(void)
{
	int32_t read = 2308; // to be divided by 100 to get actual reading
	double readconv = 23.08;

	int32_t readq = read * pow(2, 31 - SHIFT);
	int32_t resq = RES * pow(2, 31 - SHIFT);

	int32_t resultq = 
		(int32_t)((((int64_t) readq) << (31 - SHIFT)) / ((int64_t) resq));

	int32_t doubleresultq = (int32_t) (readconv * pow(2, 31 - SHIFT));

	printf("read: %d 0x%x;\n", read, read);
	printf("read PRIq: %s%d.%d;\n", PRIq_arg((int32_t) readq, 6, SHIFT));
	printf("readq: %lld 0x%llx\n", readq, readq);
	printf("res: %d 0x%x;\n", RES, RES);
	printf("res PRIq: %s%d.%d;\n", PRIq_arg((int32_t) resq, 6, SHIFT));
	printf("resq: %lld 0x%llx\n", resq, resq);
	printf("resultq: %d 0x%x;\n", resultq, resultq);
	printf("result PRIq: %s%d.%d;\n", PRIq_arg(resultq, 6, SHIFT));
	printf("doubleresultq: %d 0x%x;\n", doubleresultq, doubleresultq);
	printf("doubleresult PRIq: %s%d.%d;\n", PRIq_arg(doubleresultq, 6, SHIFT));

	return 0;
}
```

`west build -b qemu_x65 samples/hello_world -t run`

**Expected behavior**

```
23.08 
0x 000b 8a3d
0000 0000 0000 1011 1 . 000 1010 0011 1101
10111 = 23
000 1010 0011 1101 = 2^-4 + 2^-6, etc with -10, -11, -12, -13, -14, -16 = 0.080032349
23.080032349
```

**Impact**
Showstopper, since new sensor API uses fixed point numbers

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain: Zephyr SDK
 - SHA: latest main (`a46cef4`)

laurenmurphyx64, 2024-09-09:
@yperess Am I totally missing something here?

thedjnK, 2024-09-09:
You are printing as a floating point number, so the numbers you print are expected, floating point has a limited precision (and depends on if you are using single or double precision)

yperess, 2025-01-28:
I'll take a look

yperess, 2025-04-02:
It seems that picolibc doesn't support the `%06d` format so you're missing a leading 0. I'm still looking for a good solution for this.

keith-packard, 2025-04-02:
Picolibc supports `%06d`; I just verified by hacking samples/hello_world:
```
int main(void)
{
	printf("Hello World! %06d %s\n", 12, CONFIG_BOARD_TARGET);

	return 0;
}
```
If you set `CONFIG_PICOLIBC_IO_MINIMAL`, then you'll get a version which lacks all sorts of functionality (including zero-filling) so maybe you've select that?

yperess, 2025-07-30:
Can you verify that https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CBPRINTF_N_SPECIFIER is enabled?