zephyrproject-rtos/zephyr #78214: clang-format and checkpatch have contradictory requirements

nordic-krch, 2024-09-10:
**Describe the bug**

After running `clang-format` on `drivers/serial/uart_nrfx_uarte.c` (#78179) I'm getting checkpatch errors.

There are 2 types of errors:
- `SPACING: space prohibited before that close parenthesis ')'` clang-format added space after `VAL,` in a statement like below
 ```
COND_CODE_1(FLAG, (.elem = VAL, ), ())
```
- `TABSTOP: Statements should start on a tabstop` clang-format reformatted macro with code generation in a way that is not accepted by checkpath.

**To Reproduce**

1. run clang-format on `drivers/serial/uart_nrfx_uarte.c` and then run checkpatch
2. See error

**Expected behavior**

Running clang-format shall not violate checkpatch rules.

**Impact**

Red CI after running clang-format. Manual code adjustment must be used which discourage user from using clang-format.

**Environment (please complete the following information):**

 - OS: Linux
 - Commit SHA 056d5a354bc

pdgendt, 2024-09-10:
FYI `clang-format` reports checkpatch issues on the "notice" level, these aren't blocking PRs.

nordic-krch, 2024-09-10:
yes, but if i run `clang-format` on a file and commit that I will get red CI because checkpatch reports those as errors. https://docs.zephyrproject.org/latest/contribute/guidelines.html#clang-format suggests to use it. On the other hand it states that
> When there are differences between the [Coding Style](https://docs.zephyrproject.org/latest/contribute/guidelines.html#id2) guidelines and the formatting generated by code formatting tools, the [Coding Style](https://docs.zephyrproject.org/latest/contribute/guidelines.html#id2) guidelines take precedence.

but it would be best to configure it in a way so that there are as few differences as possible.

nordic-krch, 2024-09-10:
If it's not recommended to run clang-format on entire existing file (https://github.com/zephyrproject-rtos/zephyr/pull/78179#issuecomment-2340255925) then maybe it's worth adding it to the guidelines section about `clang-format`. It's not clear how to treat those notices left by clang-format in the PR? Is it obligatory to fix them?

keith-zephyr, 2025-01-15:
This is addressed by https://github.com/zephyrproject-rtos/zephyr/pull/83374

ithinuel, 2025-02-04:
Trying to apply clang-format to some of the arm architecture files, I hit several cases of mismatchTrying to apply clang-format to some of the arm architecture files, I hit several cases of mismatched requirements between Clang-Format and checkpatch.

For example on current main (at https://github.com/zephyrproject-rtos/zephyr/commit/52b8e3d3706f7503d6e4e36413c2f78a2ac95480) running clang-format leaves checkpatch outputing:

```
tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:140: ERROR:SPACING: space prohibited after that '&&' (ctx:WxW)
#140: FILE: tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:140:
+	zassert_true((src->s16 == dst->s16) && (src->s17 == dst->s17) && (src->s18 == dst->s18) &&
 	                                    ^

tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:176: WARNING:BLOCK_COMMENT_STYLE: Block comments use a trailing */ on a separate line
#176: FILE: tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:176:
+	 * it is required by the test */

tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c total: 1 errors, 1 warnings, 663 lines checked
```

Running `checkpatch.pl …  --fix-inplace` removes the space after the first `&&` on that line
But then `check_compliance` fails because Clang-format disagrees with this missing space.

Another case where the thow cannot be satisfied is on `include/zephry/arch/arm/mpu/arm_mpu_v8.h:238` where clang-format makes the line 102 characters long instead of the 100 limit, but I guess this one is more on clang-format’s side.

keith-zephyr, 2025-02-05:
> For what it’s worth, I discovered the directives to locally disable clang format:

In my opinion - disabling clang-format for short code blocks is a good approach to resolve such conflicts.  Overall, clang-format does a decent, but not perfect job.

closed 2025-04-21 by github-actions[bot]