zephyrproject-rtos/zephyr #78223: Bluetooth: Host: Assertion in bt_hci_evt_create() with scanning and extended advertising in progress

cvinayak, 2024-09-10:
**Describe the bug**
Assertion in bt_hci_evt_create() with scanning and extended advertising in progress.

```
[2024-09-10 12:30:07.812] uart:~$ [DEVICE]: 7C:10:41:3A:2C:E2 (random), AD evt type 0, RSSI -96  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 us), SID: 0xff
[2024-09-10 12:30:07.828] uart:~$ [DEVICE]: 68:72:C3:9A:1E:47 (public), AD evt type 3, RSSI -78  C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 us), SID: 0xff
[2024-09-10 12:30:07.845] uart:~$ [DEVICE]: 50:7C:98:1D:BF:56 (random), AD evt type 0, RSSI -59  C:1 S:1 D:9  C:1 S:1 D:0 SR:0 E:0 Prim: ASSERTION FAIL [buf] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_common.c:19
[2024-09-10 12:30:07.866] uart:~$ [00:45:10.745,148] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x200034f8
[2024-09-10 12:30:07.877] uart:~$ [00:45:10.745,156] <err> os: r3/a4:  0x00000003 r12/ip:  0x20003458 r14/lr:  0x00011309
[2024-09-10 12:30:07.889] uart:~$ [00:45:10.745,162] <err> os:  xpsr:  0x01000000
[2024-09-10 12:30:07.897] uart:~$ [00:45:10.745,167] <err> os: Faulting instruction address (r15/pc): 0x00011316
[2024-09-10 12:30:07.907] uart:~$ [00:45:10.745,188] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[2024-09-10 12:30:07.918] uart:~$ [00:45:10.745,209] <err> os: Current thread: 0x200034f8 (sysworkq)
[2024-09-10 12:30:07.927] uart:~$ [00:45:10.819,969] <err> os: Halting system
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir -p build/shell; cd build/shell
2. cmake -GNinja -DBOARD=nrf54l15pdk/nrf54l15/cpuapp -DCONFIG_BT_CTLR_PHY_CODED=y ../../tests/bluetooth/shell
3. ninja
4. ninja flash
5. open terminal (ex, minicom -D /dev/ttyACM{n})
6. `bt init`
7. `bt adv-create conn-nscan ext-adv`
8. `bt adv-data 020933`
9. `bt adv-start`
10. `bt scan on coded`
11. Let the application run for a duration in multiples of 15 mins
12. The said assertion can be observed after 15/30/45/... mins when RPA expires

**Expected behavior**
Scanning and Extended/Legacy Advertising should continue for eternity without assertions.

**Impact**
Showstopper

**Logs and console output**
[hci_common_assert.txt](https://github.com/user-attachments/files/16944159/hci_common_assert.txt)

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA: a3f71f13bfb

**Additional context**
NA

jori-nordic, 2024-09-10:
Easier repro: https://github.com/jori-nordic/zephyr/commit/899271d8268c1eda8d7f1169ff8020a54bf32069

```bash
cd zephyr/tests/bsim/bluetooth/host/scan/start_stop
./compile.sh && ./test_scripts/start_stop.sh
```

[image]

jori-nordic, 2024-09-10:
[image]

[image]

alwa-nordic, 2024-09-10:
> @cvinayak, is this a regression?

Yes probably another victim of https://github.com/zephyrproject-rtos/zephyr/pull/71697

alwa-nordic, 2024-09-10:
> > Easier repro: [jori-nordic@899271d](https://github.com/jori-nordic/zephyr/commit/899271d8268c1eda8d7f1169ff8020a54bf32069)
> > ```shell
> > cd zephyr/tests/bsim/bluetooth/host/scan/start_stop
> > ./compile.sh && ./test_scripts/start_stop.sh
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ![image](https://private-user-images.githubusercontent.com/68428270/366130780-36b9bcbf-9b3a-4086-abd9-b38d96014fdd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5ODY3MjcsIm5iZiI6MTcyNTk4NjQyNywicGF0aCI6Ii82ODQyODI3MC8zNjYxMzA3ODAtMzZiOWJjYmYtOWIzYS00MDg2LWFiZDktYjM4ZDk2MDE0ZmRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEwVDE2NDAyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2MDFmYTI0ZTUyNzI4MGYxNTAzMmUzNWNhOGRhZGYyZWE5ODc3MWJmMDJlMzBhZjNjOWY1YWE0YjQ4OGIxNzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RRpxEdXut4IQiafE9q4bbrFyz56p8G78Q7VOwZLyuLA)
> 
> The call is clearly from `rpa_timeout`

The back trace shows control passes through `hci_driver_send`, at which point it's in the controller.

cvinayak, 2024-09-10:
Put it this way, why is host calling Controller functions directly? and why from a system work queue?

alwa-nordic, 2024-09-10:
> Put it this way, why is host calling Controller functions directly? and why from a system work queue?

Good question.. The host is calling the registered `bt_hci_driver_api.send`, which is `hci_driver_send`. `bt_hci_driver_api.send` should probably not be allowed to block, and preferably not use much stack either.

alwa-nordic, 2024-09-10:
> and why from a system work queue?

We got rid of the TX thread in the Host.

cvinayak, 2024-09-10:
> > and why from a system work queue?
> 
> We got rid of the TX thread in the Host.

Yes, in commit 28be8909a6c5f0525ddbb401e181dd438a57019f

which also states:
```
   There is an edge-case, where we call `bt_hci_cmd_send_sync()` from the
    syswq, stalling the system. The proposed mitigation is to attempt to drain
    the command queue from within `bt_hci_cmd_send_sync()`.
    
    My spidey sense tingles however, and it would be better to just remove the
    capability of calling this fn from the syswq. But doing this requires
    refactoring a bunch of synchronous procedures in the stack (e.g. stack
    init, connection establishment, address setting etc), dragging in more
    work. I will do it, but in a later patch.
```

Was this `later patch`  done?

jori-nordic, 2024-09-10:
We weren't trying to flame, the comments I made were made in jest :)

I will clean up the test and make a PR tomorrow (only for the test, not a fix). We can then try to brainstorm a temporary workaround to this particular issue.

The issue is in neither host or controller IMO, but rather in the interface between the two. What this assert really tells us, is that it detected a potential deadlock, likely due to https://github.com/zephyrproject-rtos/zephyr/issues/50786

I will be stepping away from host development for a while, so the `later patch` will not come from me. But rest assured that an async command interface is [being](https://github.com/zephyrproject-rtos/zephyr/issues/74642#issuecomment-2190534513) [discussed](https://github.com/zephyrproject-rtos/zephyr/issues/77241#issuecomment-2304400678). As soon as we have an async interface, we will be able to refactor those two problematic places in the host and submit the patch as promised.

Ensuring controller is re-entrant or not should not be the responsibility of the host. Controller can take a mutex if it's not re-entrant. 
In our case though, no commands are sent from other threads than the system workqueue so it should be fine. What aleks said with "deleted the TX thread" is that we deleted the thread (ie the resource), but not the single-tx-context behavior.

cvinayak, 2024-09-10:
> Ensuring controller is re-entrant or not should not be the responsibility of the host.

True. But for historical reasons the functional interface of the Controller took advantage of the then co-operative tx thread to ensure a single execution context call of Controller functions.

 > What aleks said with "deleted the TX thread" is that we deleted the thread (ie the resource), but not the single-tx-context behavior.
 
 Good to know (for now).

alwa-nordic, 2024-09-10:
> The issue is in neither host or controller IMO, but rather in the interface between the two.

I agree and think we need to discuss and formalize the interface.

The Host design is moving towards asynchronous functions and using the system work queue. I would prefer not to have a thread.

Likewise, I expect most HCI drivers will eventually be implemented using the system work queue, or no thread at all. So, I think the right place to manage a thread for the HCI driver is in the HCI driver itself.

The new HCI driver interface `zephyr/drivers/bluetooth.h` is meant to eventually follow whatever the standard Zephyr driver concept is. To that end, what context can interact with the driver should be the same as for any Zephyr syscall. I believe syscalls should be available from the system work queue.

We could make a HCI driver shim to manage a thread for the actual HCI driver and ensure `bt_hci_send` is asynchronous and safe to call from the system work queue by offloading the call into `drv.send` to the thread. I'm not sure how this would fit into the new HCI driver interface.

@jhedberg, what do you think?

mmahadevan108, 2024-09-12:
If https://github.com/zephyrproject-rtos/zephyr/pull/78294 is fixing this issue, then please link by adding  `Fixes` to the PR description.

jori-nordic, 2024-09-12:
it does not. It's only a test that reproduces it.

PR #78397 (2024-09-17): Bluetooth: buf: Put command complete/status in sync buf pool
Why is it ok to use the sync pool?

Because command complete/status is processed in prio: that means on the same stack as the `bt_recv()` call from the driver.

Why does it fix the issue?

Because the complete/status event goes into a pool that is guaranteed to have one free buffer any time `bt_recv()` is not executing.

Since the driver is the one calling bt_recv(), it (hopefully) will finish one `bt_recv()` before starting another one.

Fixes #78223
commit 6d5cce66 (2024-09-17): Bluetooth: buf: Put command complete/status in sync buf pool

Why is it ok to use the sync pool?

Because command complete/status is processed in prio: that means on the
same stack as the `bt_recv()` call from the driver.

Why does it fix the issue?

Because the complete/status event goes into a pool that is guaranteed to
have one free buffer any time `bt_recv()` is not executing.

Since the driver is the one calling bt_recv(), it (hopefully) will
finish one `bt_recv()` before starting another one.

Fixes #78223
closed 2024-09-17 by fabiobaltieri