zephyrproject-rtos/zephyr #78225: Add FlashDB as external module

karthi012, 2024-09-10:
## Origin

FlashDB: Database for micorcontroller
https://github.com/armink/FlashDB

## Purpose
Zephyr RTOS does not include a built-in database. Instead, **FlashDB**, an external module, offers database support for Zephyr. FlashDB is a highly lightweight database designed for microcontrollers, featuring a minimal memory footprint.

## Mode of integration
FlashDB is integrated as an external module and is already maintained by the community, with ongoing updates for features and bug fixes.

Name of the module: FlashDB
## Maintainership
@parthitce  as a Maintainer. @karthi012 adding myself as collaborators.

## Pull Request
We haven’t submitted any pull requests to the mainline Zephyr yet. This is the initial stage of discussions regarding database support for Zephyr. You can find the reference implementation for Zephyr here: https://github.com/linumiz/rixx/commit/d54a84701b10436837c2febf9b8dd6a0bbeb54c6.

## Description
FlashDB is an ultra-lightweight database designed for data storage in embedded devices. It leverages the features of flash memory to ensure reliability and performance while minimizing resource consumption and extending the lifespan of flash devices.

FlashDB offers two modes of database engines:
 **Key-Value Database**: Stores data as key-value pairs, with each key serving as a unique identifier.
**Time Series Database**: Organizes data in a time series format, providing high query performance.

For a more detailed description, refer to the FlashDB documentation here: https://github.com/armink/FlashDB/blob/master/README.md."

What is its primary functionality?
FlashDB is a lightweight database designed for data storage using flash devices.

What problem are you trying to solve?
Zephyr RTOS is primarily used for IoT applications, it often needs to manage large volumes of data flows. However, Zephyr lacks a suitable database solution for handling this data. FlashDB aims to address this gap effectively.

Why is this the right component to solve it?

1. FlashDB has a small memory footprint.
2. It does not use heap memory allocations.
3. It includes wear leveling to extend the life of the flash memory.

## Dependencies

What other components does this package depend on?
FlashDB operates in two modes:

1.     **Flash Mode
2.     POSIX File Mode**

Current Support: FlashDB operates in **POSIX File Mode.**
**POSIX File Mode**: Utilizes Zephyr's **POSIX file system** implementation for data storage.
**Flash Mode**: Implementation is required and will use the **Zephyr Flash API**; it is not yet supported.

## Revision
Recently, FlashDB introduced Zephyr module support in a reference implementation, with the initial commit available here: https://github.com/armink/FlashDB/commit/523fd2052d13baf72fd1d8e177c2a162a03392be. This marks the beginning of FlashDB's integration with Zephyr."

## License
Apache License 2.0

nashif, 2024-09-11:
without direct dependency of zephyr on the code in the module, this should be added as external module, i.e. 

https://docs.zephyrproject.org/latest/develop/manifest/index.html#external-projects-modules

nashif, 2024-09-11:
> POSIX filesystem implementation or Flash API are dependencies, if so, it is not allowed to add as a external module.

where is this dependency? I do not see any subsystem in zephyr right now and in your PR that would make it depend on the database...

parthitce, 2024-09-11:
@nashif Thanks for the clarification and details provided during the TSC call today. To summarize,
- Adding a new external module in Zephyr (west.yml and optional.yml) means there  is a consumer for the module internally in the Zephyr itself
- Suggestion is to add Flash API as well to FlashDB

As of now there is no direct dependencies in Zephyr for FlashDB and it's only for the application developer usecase, this issue can be closed. As mentioned in the initial message, existing port for Flash Database using Posix with examples are available in https://github.com/linumiz/rixx.

@karthi012 Close it if you don't have more questions or points to add.

karthi012, 2024-09-14:
@nashif @parthitce Thanks for your response.

closed 2024-09-14 by karthi012