zephyrproject-rtos/zephyr #78226: ld.lld: error: unable to find library -lstdc++

niedzwiecki-dawid, 2024-09-10:
After this [commit](https://github.com/zephyrproject-rtos/zephyr/commit/9a9e252d1677c047929f5602eace209123fafe3e), I get an error `ld.lld: error: unable to find library -lstdc++`. The [PR](https://github.com/zephyrproject-rtos/zephyr/pull/77887) says `No functional change expected` so I assume it is a bug.

Simply `-lstdc++` flag has been added to my project which uses llvm.

Thanks in advance, sorry if I missed something.

mmahadevan108, 2024-09-10:
@niedzwiecki-dawid, can you provide some steps to reproduce the failure. Kindly use the issue reporting template to provide this inforamtion.

keith-zephyr, 2024-09-10:
In our downstream project, we're using our own llvm toolchain. Our project is using the `CONFIG_LIBCXX_LIBCPP` option.

Without https://github.com/zephyrproject-rtos/zephyr/pull/77887, the LINK_LIBRARIES variable includes `-lc++`.  

However, with https://github.com/zephyrproject-rtos/zephyr/pull/77887 applied, the LINK_LIBRARIES variable includes `-lstdc++`.

My expectation is that `CONFIG_LIBCXX_LIBCPP` should select libc++, and that the `CONFIG_GLIBCXX_LIBCPP` option should select `libstd++`.

keith-zephyr, 2024-09-10:
This looks to be a latent bug in our downstream implementation.  Our toolchain's Cmake configuration files had an error where our `lld/linker_flags.cmake` file was include to Zephyr's `ld/linker_flags.cmake` instead of the `lld` version.

closed 2024-09-10 by keith-zephyr

niedzwiecki-dawid, 2024-09-11:
Yes, the problem was on our side. Sorry for confusion