zephyrproject-rtos/zephyr #78251: nrf54l15pdk: unit address and first address in 'reg' (0x5004c000) don't match for /soc/peripheral@50000000/vpr@4c000/mailbox@1

rettichschnidi, 2024-09-10:
Building an application for the Nordic nrf54l15pdk board results in a warning generated by gen_defines.py:

```
$ west build --pristine --board nrf54l15pdk/nrf54l15/cpuapp samples/userspace/hello_world_user
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: <snip>/zephyr/samples/userspace/hello_world_user
-- CMake version: 3.25.1
-- Found Python3: /home/reto/code/3rd-party/venv-zephyrproject/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/reto/.cache/zephyr
-- Zephyr version: 3.7.99 (<snip>/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nrf54l15pdk, Revision: 0.3.0, qualifiers: nrf54l15/cpuapp
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/reto/.local/opt/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/reto/.local/opt/zephyr-sdk-0.16.8)
-- Found Dtc: /home/reto/.local/opt/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: <snip>/zephyr/boards/nordic/nrf54l15pdk/nrf54l15pdk_nrf54l15_cpuapp.dts
CMake Warning at <snip>/zephyr/cmake/modules/dts.cmake:314 (message):
  gen_defines.py succeeded with some stderr contents:

  unit address and first address in 'reg' (0x5004c000) don't match for
  /soc/peripheral@50000000/vpr@4c000/mailbox@1

Call Stack (most recent call first):
  <snip>/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  <snip>/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  <snip>/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Generated zephyr.dts: <snip>/zephyr/build/zephyr/zephyr.dts
```

This warning has "always" (at least since Zephyr 3.7.0) existed, but recently got hidded by #76472. PR #78250 will make it visible once again.

**Reproducing**

Steps to reproduce the behavior:
1. (until #78250 is merged): Cherry-pick the commit in PR #78250
2. `west build --pristine --board nrf54l15pdk/nrf54l15/cpuapp samples/userspace/hello_world_user`
3. observe the resulting warning

**Expected behavior**

No device tree warnings printed.

**Impact**

I can not build Zephyr without warnings.

nordic-piks, 2024-09-12:
Triaged internally

rettichschnidi, 2024-09-12:
Just figured out that this might be related to #72273

rettichschnidi, 2024-10-03:
> Triaged internally

Any updates on this? I'd love to finish #78253.

nordic-piks, 2024-10-03:
@masz-nordic @anangl can you help? (regarding updates)

rettichschnidi, 2024-12-04:
@nordic-piks Any updates on this?

nordic-piks, 2024-12-09:
> @nordic-piks Any updates on this?

@masz-nordic @anangl ?