zephyrproject-rtos/zephyr #78259: doc: Documentation issue in 'hardware/peripherals/i3c'  Incorrect detail for SETDASA I3C CCC usage

quic-bkulkarn, 2024-09-11:
On https://docs.zephyrproject.org/latest/hardware/peripherals/i3c.html, it is mentioned as 
**Do SETDASA to use static addresses as dynamic address if so desired**
[image]

But it should rather be as below:
**Do SETDASA to assign a dynamic address using the static address of the device**

**Environment**

* Page: `hardware/peripherals/i3c`
* Version: main
* SHA-1: 512dc9ff38d8dad84b340712e71a20185fe59819

henrikbrixandersen, 2024-09-13:
@quic-bkulkarn Could you submit a PR for fixing this?

quic-bkulkarn, 2024-09-17:
Sure, I will be submitting a PR soon. @dcpleung can you please assign this issue to me?

kartben, 2024-11-22:
fixed with #78537

closed 2024-11-22 by kartben