zephyrproject-rtos/zephyr #78263: building samples/cpp/hello_world reports "undefined reference to `_impure_ptr'" error

aaronyegx, 2024-09-11:
**Describe the bug**
We are starting to develop cpp applications based on Zephyr. But building the samples/cpp/hello_world reports "undefined reference to `_impure_ptr'" error.

**To Reproduce**
west build -b apollo4p_evb .\samples\cpp\hello_world\ -p

**Expected behavior**
 No compiling error reports.

**Impact**
 We cannot continue the development until the compiling error getting resolved.

**Logs and console output**

[1/125] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/ambiqcodes/ambiqzephyr/zephyr), build: v3.7.0-2509-g6654c2a2f629
[120/125] Linking CXX executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/ambiqcodes/ambiqzephyr/zephyr/build/zephyr/zephyr_pre0.map
cmd.exe /C "cd . && C:\ambiqcodes\radio_sw\zephyr-sdk-0.16.8\arm-zephyr-eabi\bin\arm-zephyr-eabi-g++.exe  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=C:/ambiqcodes/ambiqzephyr/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  
zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/console/libdrivers__console.a 
 zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_ambiq/mcu/apollo4p/lib..__modules__hal__ambiq__mcu__apollo4p.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/ambiqcodes/radio_sw/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -LC:/ambiqcodes/ambiqzephyr/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cmd.exe /C "cd /D C:\ambiqcodes\ambiqzephyr\zephyr\build\zephyr && C:\Tools\cmake-3.22.3-windows-x86_64\bin\cmake.exe -E true""
c:/ambiqcodes/radio_sw/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: c:\ambiqcodes\radio_sw\zephyr-sdk-0.16.8\arm-zephyr-eabi\bin\../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v7e-m/nofp\libstdc++.a(ios_init.o): in function `std::ios_base::Init::Init()':
ios_init.cc:(.text._ZNSt8ios_base4InitC2Ev+0x120): undefined reference to `_impure_ptr'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Tools\cmake-3.22.3-windows-x86_64\bin\cmake.EXE' --build 'C:\ambiqcodes\ambiqzephyr\zephyr\build'

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain: zephyr-sdk-0.16.8
 - Commit SHA or Version used: 6654c2a2f6298d3395770291626edbf3c7d3ff65 (the latest main branch)

**Additional context**

stephanosio, 2024-09-19:
FYI, this issue is currently being tracked in https://github.com/zephyrproject-rtos/sdk-ng/issues/760.

henrikbrixandersen, 2024-09-20:
I am marking this as a release blocker since it prevents using C++ in Zephyr applications using the Zephyr SDK on two supported development platforms.

stephanosio, 2024-09-26:
This issue has been fixed on the SDK side by https://github.com/zephyrproject-rtos/sdk-ng/pull/797, which will be available as part of the upcoming Zephyr SDK 0.16.9 release.

This can be closed once the main CI is updated to use the SDK 0.16.9 release.

henrikbrixandersen, 2024-09-26:
> This issue has been fixed on the SDK side by https://github.com/zephyrproject-rtos/sdk-ng/pull/797, which will be available as part of the upcoming Zephyr SDK 0.16.9 release.
> 
> 
> 
> This can be closed once the main CI is updated to use the SDK 0.16.9 release.

I take it we need to backport the usage of the new SDK version to v3.7-branch, which is also using v0.16.8?

How about v3.6-branch, which is using v0.16.5? Is that also affected?

stephanosio, 2024-09-26:
> I take it we need to backport the usage of the new SDK version to v3.7-branch, which is also using v0.16.8?

[SDK v0.16.9 will be compatible with Zephyr v3.7 and v3.6](https://github.com/zephyrproject-rtos/sdk-ng/wiki/Zephyr-SDK-Version-Compatibility-Matrix) without any changes on the Zephyr side; so, there should be nothing to "backport."

henrikbrixandersen, 2024-09-26:
> [SDK v0.16.9 will be compatible with Zephyr v3.7 and v3.6](https://github.com/zephyrproject-rtos/sdk-ng/wiki/Zephyr-SDK-Version-Compatibility-Matrix) without any changes on the Zephyr side; so, [...]

Well, the `SDK_VERSION` file content could be backported, so CI on those release branches use the fixed SDK (and downstream users have an indication of which SDK to use).

stephanosio, 2024-09-26:
> Well, the `SDK_VERSION` file content could be backported, so CI on those release branches use the fixed SDK (and downstream users have an indication of which SDK to use).

We have never done something like this in the past; but, given the criticality of this bug and that SDK v0.16.9 is compatible with v0.16.8 and v0.16.5, I suppose we can consider doing this.

stephanosio, 2024-10-22:
This issue can be closed after https://github.com/zephyrproject-rtos/zephyr/pull/80192 is merged.

PR #80321 (2024-11-04): [Backport v3.7-branch] ci: build samples/cpp/hello_world as part of the multiplatform test
Backport cd543887f41af8fcf504dded51f2346b605731d5 from #78623.

Fixes #78263
PR #80320 (2024-11-06): [Backport v3.6-branch] ci: build samples/cpp/hello_world as part of the multiplatform test
Backport cd543887f41af8fcf504dded51f2346b605731d5 from #78623.

Fixes #78263
closed 2024-10-23 by stephanosio