zephyrproject-rtos/zephyr #78308: CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM has no effect

sam-golioth, 2024-09-11:
**Describe the bug**

`CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM` has no effect. Enabling this option does not change the size of the used space in either `ext_ram_seg` or `dram0_0_seg`. This was tested with `esp32_devkitc_wrover/esp32/procpu`
**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

EricNRS, 2024-09-12:
FYI, related issues:

- https://github.com/zephyrproject-rtos/zephyr/issues/74246
- https://github.com/zephyrproject-rtos/zephyr/issues/74675
- https://github.com/zephyrproject-rtos/zephyr/issues/74284

marekmatej, 2024-09-12:
This issue is previously described here https://github.com/zephyrproject-rtos/zephyr/issues/74675

sylvioalves, 2024-09-16:
Closing as duplicated: #74675

closed 2024-09-16 by sylvioalves