zephyrproject-rtos/zephyr #78347: Build failure on em_starterkit/emsk_em7d platform with MWDT Lite toolchain

klisevich, 2024-09-12:
**Describe the bug**

I encountered build failures when compiling the hello_world sample for the em_starterkit/emsk_em7d platform using the MWDT Lite toolchain. Several warnings indicate that certain arguments are unrecognized by the toolchain, and assembler errors occur due to incompatible instructions.

**Target Platform**: em_starterkit/emsk_em7d, revision 2.3.0
**Toolchain**: DesignWare ARC C/C++ Compiler MW-Lite U-2023.06
**Diagnosis**: Attempted to verify the environment variables and toolchain path, build configuration.

**To Reproduce**

1.Set environment variables for MWDT toolchain in PS:

`$env:ARCMWDT_TOOLCHAIN_PATH = 'C:\MWLite'`
`$env:ZEPHYR_TOOLCHAIN_VARIANT = 'arcmwdt'`

2.Build the hello_world sample using the command:

`west build -b em_starterkit/emsk_em7d samples/hello_world --pristine`

3.Observe the build process and review the errors related to unrecognized arguments and assembler failures.

**Expected behavior**
The build should successfully complete without warnings or errors, producing the compiled binary for the em_starterkit/em7d platform using the MWDT Lite toolchain.

**Logs and console output**
```
(.venv) PS C:\Users\wro-s\zephyrproject\zephyr> echo $env:ZEPHYR_TOOLCHAIN_VARIANT
arcmwdt
(.venv) PS C:\Users\wro-s\zephyrproject\zephyr> echo $env:ARCMWDT_TOOLCHAIN_PATH
C:\MWLite
(.venv) PS C:\Users\wro-s\zephyrproject\zephyr>  west build -b em_starterkit/emsk_em7d samples/hello_world --pristine
-- west build: making build dir C:\Users\wro-s\zephyrproject\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/wro-s/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.30.2
-- Found Python3: C:/Users/wro-s/zephyrproject/.venv/Scripts/python.exe (found suitable version "3.11.9", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/Users/wro-s/zephyrproject/zephyr/.cache
-- Zephyr version: 3.7.99 (C:/Users/wro-s/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: em_starterkit, Revision: 2.3.0 (Active: 2.3), qualifiers: emsk_em7d
-- Found host-tools: zephyr 0.16.8 (C:/Users/wro-s/zephyr-sdk-0.16.8)
-- Found dts preprocessor: C:/Users/wro-s/zephyr-sdk-0.16.8/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc.exe (Zephyr SDK 0.16.8)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/wro-s/zephyrproject/zephyr/boards/snps/em_starterkit/em_starterkit_emsk_em7d.dts
-- Found devicetree overlay: C:/Users/wro-s/zephyrproject/zephyr/boards/snps/em_starterkit/em_starterkit_emsk_em7d_2_3.overlay
-- Generated zephyr.dts: C:/Users/wro-s/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/wro-s/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/wro-s/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: INIT_ARCH_HW_AT_BOOT (defined at arch/Kconfig:604) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: ARCH_SUPPORTS_ARCH_HW_INIT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_INIT_ARCH_HW_AT_BOOT and/or look up
INIT_ARCH_HW_AT_BOOT in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing C:/Users/wro-s/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/wro-s/zephyrproject/zephyr/boards/snps/em_starterkit/em_starterkit_emsk_em7d_defconfig'
Merged configuration 'C:/Users/wro-s/zephyrproject/zephyr/boards/snps/em_starterkit/em_starterkit_emsk_em7d_2_3_defconfig'
Merged configuration 'C:/Users/wro-s/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'C:/Users/wro-s/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/wro-s/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GNU objcopy helper for MWDT: C:/Users/wro-s/zephyr-sdk-0.16.8/arc-zephyr-elf/bin/arc-zephyr-elf-objcopy.exe (Zephyr SDK 0.16.8)
-- The C compiler identification is Clang 15.0.7
-- The CXX compiler identification is Clang 15.0.7
-- The ASM compiler identification is unknown
-- Found assembler: C:/MWLite/MetaWare/arc/bin/ccac.exe
-- Warning: Did not find file Compiler/-ASM
CMake Warning at C:/Users/wro-s/zephyrproject/zephyr/cmake/linker/arcmwdt/target.cmake:153 (message):
[... 173 lines cut ...]
[C:\MWLite\MetaWare\arc\bin\ccac.exe] (Warning) Argument `-mmpy-option=6' is unrecognized and will be passed to the linker
[C:/MWLite/MetaWare/arc/bin/asac.exe] "C:/Users/wro-s/zephyrproject/zephyr/arch/arc/core/regular_irq.S",Line 238: error: This instruction is only available with the -Xbitscan option
[C:/MWLite/MetaWare/arc/bin/asac.exe] "C:/Users/wro-s/zephyrproject/zephyr/arch/arc/core/regular_irq.S",Line 238: error: This instruction is only available with the -Xbitscan option
[C:/MWLite/MetaWare/arc/bin/asac.exe] "C:/Users/wro-s/zephyrproject/zephyr/arch/arc/core/regular_irq.S",Line 286: error: Invalid ARC core version for instruction sflag (-core3 required)
Allowable formats are:
    sflag       reg
    sflag       imm
[C:/MWLite/MetaWare/arc/bin/asac.exe] "C:/Users/wro-s/zephyrproject/zephyr/arch/arc/include\swap_macros.h",Line 497: error: Invalid ARC core version for instruction sflag (-core3 required)
Allowable formats are:
    sflag       reg
    sflag       imm
[C:/MWLite/MetaWare/arc/bin/asac.exe] 4 errors generated.
[43/133] Linking C static library zephyr\libzephyr.a
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\wro-s\zephyrproject\zephyr\build'
```

**Environment:**

OS: Windows 11 pro
Toolchain: DesignWare ARC C/C++ Compiler MW-Lite U-2023.06, Zephyr SDK 0.16.8
Zephyr Version: 3.7.99 (commit 9584008d9f14)
CMake version: 3.30.2
Python version: 3.11.9
west version: 1.2.0

evgeniy-paltsev, 2024-09-13:
Hi @klisevich,

Right now there is no MWDT support for this target, see: https://github.com/zephyrproject-rtos/zephyr/blob/1471c2f81430fd0a381c37d263a4d0ff966e0cb4/boards/snps/em_starterkit/em_starterkit_emsk_em7d.yaml#L5-L8

To add MWDT support it's required to add proper platform-related MWDT options to https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/snps/emsk/CMakeLists.txt.

You can check how it's done for other target with MWDT toolchain support: https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/snps/hsdk/CMakeLists.txt

klisevich, 2024-09-17:
Hi @evgeniy-paltsev
Thank you for your support. Could you please clarify how the MWDT support should be represented in the YAML file? I reviewed the HSDK board you mentioned but couldn't find any entries in the YAML file indicating MWDT toolkit support. Could you point me to the specific configuration?

Thanks!

ruuddw, 2024-09-17:
Hi @klisevich
It is not defined in the SoC yaml, but in the board variant file: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/snps/hsdk/hsdk.yaml

klisevich, 2024-09-17:
Hi @ruuddw 
Thank you! I see it clearly now.

klisevich, 2024-09-17:
Got it, thanks for the help! I understand the issue now and will work on it.

Closing this out.

closed 2024-09-17 by klisevich