zephyrproject-rtos/zephyr #78351: kernel: sched: schedule_api: kernel.scheduler.multiq_no_timeslicing fails

katgiadla, 2024-09-12:
**Describe the bug**
The test `tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing` fails.

Observed for `nrf54l15dk/nrf54l15/cpuflpr`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15/cpuflpr` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/kernel/sched/schedule_api/ -p nrf54l15dk/nrf54l15/cpuflpr --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-2503-g1e4fd23e58e7 ***
Running TESTSUITE threads_scheduling
===================================================================
START - test_bad_priorities
 PASS - test_bad_priorities in 0.001 seconds
===================================================================
START - test_busy_wait_cooperative
 PASS - test_busy_wait_cooperative in 0.101 seconds
===================================================================
START - test_k_thread_priority_get_init_null
 SKIP - test_k_thread_priority_get_init_null in 0.001 seconds
===================================================================
START - test_k_thread_priority_set_init_null
 SKIP - test_k_thread_priority_set_init_null in 0.001 seconds
===================================================================
START - test_k_thread_priority_set_overmax
 SKIP - test_k_thread_priority_set_overmax in 0.001 seconds
===================================================================
START - test_k_thread_priority_set_upgrade
 SKIP - test_k_thread_priority_set_upgrade in 0.001 seconds
===================================================================
START - test_k_thread_resume_init_null
 SKIP - test_k_thread_resume_init_null in 0.001 seconds
===================================================================
START - test_k_thread_suspend_init_null
 SKIP - test_k_thread_suspend_init_null in 0.001 seconds
===================================================================
START - test_k_wakeup_init_null
 SKIP - test_k_wakeup_init_null in 0.001 seconds
===================================================================
START - test_lock_preemptible
 PASS - test_lock_preemptible in 0.201 seconds
===================================================================
START - test_pending_thread_wakeup
 PASS - test_pending_thread_wakeup in 0.001 seconds
===================================================================
START - test_priority_cooperative
 PASS - test_priority_cooperative in 0.101 seconds
===================================================================
START - test_priority_preemptible
 PASS - test_priority_preemptible in 0.101 seconds
===================================================================
START - test_priority_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
 PASS - test_priority_scheduling in 0.012 seconds
===================================================================
START - test_sched_is_preempt_thread
 PASS - test_sched_is_preempt_thread in 0.001 seconds
===================================================================
START - test_sleep_cooperative
 PASS - test_sleep_cooperative in 0.101 seconds
===================================================================
START - test_sleep_wakeup_preemptible
E: 
E:  mcause: 2, Illegal instruction
E:   mtval: 0
E:      a0: 20032310    t0: 00000002
E:      a1: 00000000    t1: 0000fa00
E:      a2: 20032530    t2: 00000004
E:      a3: 20032530
E:      a4: 00001808
E:      a5: 20032ab0
E:      sp: 20033538
E:      ra: 2002a1c8
E:    mepc: 2002a1ac
E: mstatus: 00001800
E: 
E: call trace:
E:       0: sp: 20033538 ra: 2002a1ac
E:       1: sp: 2003353c ra: 20029702
E:       2: sp: 20033544 ra: 200296d0
E:       3: sp: 20033548 ra: 2002a1c8
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20032310 (unknown)
Caught system error -- reason 0 0
Fatal error was unexpected, aborting...
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA or Version used: `v3.7.0-2503-g1e4fd23e58e7`

nordic-piks, 2024-09-26:
Triaged internally

nashif, 2024-09-27:
> Triaged internally

"triaged internally" provides zero value to anybody. maybe you should just report and deal with this issue "internally".

nordic-piks, 2024-09-30:
> > Triaged internally
> 
> "triaged internally" provides zero value to anybody. maybe you should just report and deal with this issue "internally".

This was discussed with @carlescufi , the comment indicated that we discusses this issue internally.
I totally understand that this comment does not bring value to anybody, however, this was the least intrusive method to mark this issue as "viewed/triaged", so next time we can focus only at new issues.
Do you have a suggestion how can we mark this in different way? (I already checked labels but there is no one with such meaning or even similar).