zephyrproject-rtos/zephyr #78353: drivers: NXP drivers not selecting PINCTRL when needed

decsny, 2024-09-12:
This is an issue for more than NXP boards and drivers, but this tracking issue is for NXP platforms only as that is what the NXP platform maintainers will commit to fixing.

The issue is that many (maybe all) NXP boards are setting `CONFIG_PINCTRL=y` in the board conf file. This should not have to happen because any drivers that use the pinctrl should select it automatically.

Example:
https://github.com/zephyrproject-rtos/zephyr/blob/dc5f1bfb3f2eee3e3775175232d508660f5d6efe/boards/nxp/frdm_mcxn947/frdm_mcxn947_mcxn947_cpu0_defconfig#L12

The only exception I found was the `twr_ke18f` board which does use pinctrl APIs in the board code itself:
https://github.com/zephyrproject-rtos/zephyr/blob/dc5f1bfb3f2eee3e3775175232d508660f5d6efe/boards/nxp/twr_ke18f/pinmux.c#L11-L27

nordicjm, 2024-09-13:
> The only exception I found was the twr_ke18f board which does use pinctrl APIs in the board code itself:

The board's Kconfig file would need to select it in that case

decsny, 2024-09-20:
closing as duplicate of #78619

closed 2024-09-20 by decsny