zephyrproject-rtos/zephyr #78381: tests: gpio_basic_api: arduino not valid on different gpio controllers

FRASTM, 2024-09-13:
**Describe the bug**
test execution **drivers.gpio.2pin_arduino**  fails in kernel panic on some stm32 boards because the overlay selects arduino connector pins on different GPIO controllers

Please also mention any information which could help others to understand
the problem you're facing:
 - example on the nucleo stm32g474re board (and  many others too)
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? since the https://github.com/zephyrproject-rtos/zephyr/pull/77699
 - ...
-->

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

hakehuang, 2024-09-17:
I think this is ST gpio dedicated issue, it requires the gpio on same bank, right? in this case, there is an customize case with allows you to custom for your boards, like below:
`
  drivers.gpio.2pin_arduino:
    min_flash: 34
    depends_on:
      - gpio
      - arduino_gpio
    filter: dt_compat_enabled("test-gpio-basic-api") and dt_compat_enabled("arduino-header-r3")
    extra_args: "DTC_OVERLAY_FILE=arduino.overlay"
    integration_platforms:
      - frdm_k64f
    platform_exclude:
      # below boards are mikro bus not arduino
      - da14695_dk_usb
      - mimxrt595_evk/mimxrt595s/cm33

  drivers.gpio.2pin_arduino_customized:
    min_flash: 34
    depends_on:
      - gpio
      - arduino_gpio
    filter: dt_compat_enabled("test-gpio-basic-api") and dt_compat_enabled("arduino-header-r3")
    platform_allow:
      - mimxrt595_evk/mimxrt595s/cm33
`

FRASTM, 2024-09-18:
> I think this is ST gpio dedicated issue, it requires the gpio on same bank, right? 

No sure, the test main.c is implemented that way: 
```
if (in_dev != out_dev) {
		printk("FATAL: output controller %s != input controller %s\n",
		       out_dev->name, in_dev->name);
		k_panic();
```
No matter what the SOC family

FRASTM, 2024-09-18:
> @FRASTM, can you give me a board list so that I can add your platforms to the customer list?

-  disco_l475_iot1
-  nucleo_f103rb
-  nucleo_f207zg
-  nucleo_f429zi
-  nucleo_f746zg
-  nucleo_g071rb
-  nucleo_g474re
-  nucleo_h743zi
-  nucleo_l152re
-  nucleo_wb55rg
-  nucleo_wl55jc

hakehuang, 2024-09-18:
> No matter what the SOC family

NXP platform does not have such limitation, Can I add a config to disable this check?

hakehuang, 2024-09-18:
> connecting output 10 of **gpio A** to input 3 **GPIOB** coming from the arduino. overlay : it means "arduino connector pin 8 to pin 9" = D2 to D3

@FRASTM , I will add a ne instance for ST platforms with CONFIG_ONE_PORT=y

hakehuang, 2024-09-18:
@FRASTM  can you try the fixes? I do not have st boards.

PR #78654 (2024-10-17): Arduino gpio fix
st platforms require input and output on same port, add support for such constrains

#fixes: #78381
commit 43425014 (2024-10-17): tests: gpio: gpio_basic_api split input and ouput

input and output shall be split and configure by dts

fixes: #78381
closed 2024-10-17 by nashif