zephyrproject-rtos/zephyr #78405: subsys: dfu: img_util: dfu.image_util{.progressive} fail

katgiadla, 2024-09-13:
**Describe the bug**
Tests:
- `tests/subsys/dfu/img_util/dfu.image_util`
- `tests/subsys/dfu/img_util/dfu.image_util.progressive`
fail.

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/dfu/img_util -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-2572-g594bef84f84e ***
Running TESTSUITE img_util
===================================================================
START - test_check_flash
ASSERTION FAIL [0] @ ../../../../../../../zephyr/lib/utils/onoff.c:283
E: r0/a1:  0x00000004  r1/a2:  0x0000011b  r2/a3:  0x00000001
E: r3/a4:  0x00000004 r12/ip:  0x00000040 r14/lr:  0x00000871
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x000074f4
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-1928-gbd29c5c410bd`

nordic-piks, 2024-09-26:
Triaged internally

nordicjm, 2024-10-28:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/77058

closed 2024-10-28 by nordicjm