zephyrproject-rtos/zephyr #78413: samples: subsys: profiling: perf: sample.perf reports pytest timeout

katgiadla, 2024-09-13:
**Describe the bug**
Sample `samples/subsys/profiling/perf/sample.perf` reports pytest timeout.

Observed for `nrf54l15dk/nrf54l15/cpuflpr`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/profiling/perf/ -p nrf54l15dk/nrf54l15/cpuflpr --device-testing --device-serial /dev/ttyACM1 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
DEBUG:twister_harness.device.factory: Get device type "hardware"
DEBUG:twister_harness.device.hardware_adapter: Opening serial connection for /dev/serial/by-id/usb-SEGGER_J-Link_00-if00
DEBUG:twister_harness.device.hardware_adapter: Flashing device
DEBUG:twister_harness.device.hardware_adapter: Flashing command: /usr/local/bin/west flash --skip-rebuild --build-dir /zephyr/zephyr/twister-out/nrf54l15dk_nrf54l15_cpuflpr/samples/subsys/profiling/perf/sample.perf --runner nrfutil -- --erase --dev-id 
DEBUG:twister_harness.device.hardware_adapter: Flashing finishedDEBUG:twister_harness.helpers.utils: Found matching key: CONFIG_SHELL_PROMPT_UART="uart:~$ "
INFO:twister_harness.fixtures: Wait for prompt
DEBUG:twister_harness.helpers.shell: Got prompt
INFO:test_perf: send "perf record 200 99" command
ERROR:twister_harness.device.device_adapter: Read from device timeout occurredDEBUG:twister_harness.device.hardware_adapter: Closed serial connection for /dev/serial/by-id/usb-SEGGER_J-Link_if00
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-2182-ge9ff0dc35e71`

**Additional context**
Similar issue has occurred for `nrf54l15pdk/nrf54l15/cpuflpr` (last one recorded on version `v3.7.0-1999-g7cf124`).

nashif, 2024-09-13:
did that ever work before on those platforms?

nordic-piks, 2024-09-26:
@PerMac @katgiadla More logs are needed, what is the issues in the test?

nordic-piks, 2024-09-26:
Triaged internally