zephyrproject-rtos/zephyr #78418: drivers: flash: common: drivers.flash.common.default fails

katgiadla, 2024-09-13:
**Describe the bug**
Test `tests/drivers/flash/common/drivers.flash.common.default` fails.

Observed for `nrf54l15dk/nrf54l15/cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/flash/common -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-2094-gca48767be48c ***
Running TESTSUITE flash_driver
===================================================================
Test will run on device mx25r6435f@0
START - test_flash_erase
 PASS - test_flash_erase in 1.155 seconds
===================================================================
START - test_flash_fill
E: ***** USAGE FAULT *****
E:   Stack overflow (context area not valid)
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x0000186b
E: r3/a4:  0x00000000 r12/ip:  0x00000020 r14/lr:  0x0000186b
E:  xpsr:  0x00004800
E: Faulting instruction address (r15/pc): 0x0000186a
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200001e0 (test_flash_fill)
E: Halting system
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: v3.7.0-2094-gca48767be48c`

nordic-piks, 2024-09-26:
@de-nordic Please have a look

nordic-piks, 2024-09-26:
Triaged internally

de-nordic, 2024-09-26:
Check stack.

de-nordic, 2024-09-26:
OK, seems that call to `spi_transceive_dt` from `spi_nor_access`, called by `spi_nor_write` topples over. Didn't go deeper, but stack size changes change nothing.
Also the board is missing configuration which cases it to go into mx25 testing as flash device, where there is erase page size left at max val of 64k.

nordic-piks, 2024-10-11:
Test started to fail at nrf54h20dk/nrf54h20/cpuapp too.
Any update?

katgiadla, 2024-10-22:
Issue occurrs still on `nrf54h20dk/nrf54h20/cpuapp`. 

```
*** Booting Zephyr OS build v3.7.0-4790-g6f64f7755796 ***
Running TESTSUITE flash_driver
===================================================================
START - test_flash_copy
Test will run on device mram@e000000
No devices with erase requirement present
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0xe1e9000
E: r0/a1:  0x0e1e9000  r1/a2:  0x2f0124b9  r2/a3:  0x2f0124d8
E: r3/a4:  0x0e1e8fff r12/ip:  0x0e0ad5a1 r14/lr:  0x0e0ad593
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x0e0adbd0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2f011140 (test_flash_copy)
E: Halting system
```
@de-nordic , could you check this?

PR #79814 (2024-10-17): tests/flash/common: Increase ztest stack size
The commit sets CONFIG_ZTEST_STACK_SIZE to 4096.

Fixes #79801
Fixes #78418