zephyrproject-rtos/zephyr #78459: STM32U5XX devices ADC12 extended calibration fails

FrankZShen, 2024-09-16:
**Describe the bug**

Extended calibration for ADC12 produces the wrong calibration values which leads to incorrect ADC readings from the peripheral.

Issue was adding in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/75048

**To Reproduce**

Reproduction is extremely finnicky. The issue appeared and disappeared with changing code in completely unrelated places.

- CPU is at 160 MHz.
- ADC clock prescaler is 4.
- Optimisation is set for size.

**Expected behavior**

ADC readings are correct.

**Impact**

ADC readings peripheral are sometimes offset from real voltage values. While offset is consistent, it's appearance is unpredictable. Impact is bad and continued use of ADC is untenable.

**Environment (please complete the following information):**

 - Ubuntu 22.04.2 LTS
 - Zephyr SDK 0.16.8
 - Zephyr 3.7.0

**Additional context**

The issue is already fixed in the STM32 HAL driver. This issue is identical to https://github.com/STMicroelectronics/STM32CubeU5/issues/29 and was fixed in https://github.com/STMicroelectronics/STM32CubeU5/issues/29#issuecomment-1708366862. This fix was later added to release 1.4.0 of their library.

gautierg-st, 2024-09-20:
Hi @FrankZShen 
I've just pushed #78748 that reproduces what is done in the Cube. Could you test and let me know if that solves your problem?

FrankZShen, 2024-09-22:
Yes, that work fixes fit for me. 👍

gautierg-st, 2024-09-23:
Perfect, thanks for your feedback!

PR #78748 (2024-09-24): drivers: adc: stm32: fix stm32u5 extended calibration
Add Data Memory Barrier during the extended calibration of STM32U5, as it is done in STM32Cube HAL, to avoid sporadic errors during calibration that may result in measures that are offset from real values.

Fixes #78459
closed 2024-09-24 by aescolar