zephyrproject-rtos/zephyr #78501: The presence of a folder named `version` in the root of a project triggers various CMake errors (macOS)

erlingrj, 2024-09-16:
**Describe the bug**
If a folder named `version` is located at the root of a project, various CMake errors will arise.

**To Reproduce**
```
cd samples/basic/minimal
mkdir version
west build -b qemu_cortex_m3 -p always
```

The issue seems to be with the loading of the VERSION file which is to be found in at ZEPHYR_BASE/VERSION. It appears to be confused by the folder named version in the current directory.

**Logs and console output**
```
(.venv) Erlings-MBP:zephyr erling$ west build -p always -b qemu_cortex_m3 samples/basic/minimal
-- west build: making build dir /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/samples/basic/minimal
-- CMake version: 3.30.3
-- Found Python3: /Users/erling/dev/lf-zephyr-workspace/.venv/bin/python3.8 (found suitable version "3.8.20", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /Users/erling/Library/Caches/zephyr
-- Zephyr version: 3.7.0 (/Users/erling/dev/lf-zephyr-workspace/deps/zephyr)
CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:54 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:57 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:60 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:63 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:66 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:92 (math):
  math cannot parse the expression: "( << 16) + ( << 8) + ()": syntax error,
  unexpected exp_SHIFTLEFT (4).
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:93 (math):
  math cannot parse the expression: "( << 24) + ( << 16) + ( << 8) + ()":
  syntax error, unexpected exp_SHIFTLEFT (4).
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:95 (math):
  math cannot parse the expression: "ERROR": syntax error, unexpected end of
  file (5).
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/version.cmake:96 (math):
  math cannot parse the expression: "ERROR": syntax error, unexpected end of
  file (5).
Call Stack (most recent call first):
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/erling/dev/lf-zephyr-workspace/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)
```

**Environment (please complete the following information):**

- macOS Ventura 13.6
- Zephyr v3.7.0
- SDK v0.16.8

**Additional context**

nordicjm, 2024-09-17:
> The issue seems to be with the loading of the VERSION file which is to be found in at ZEPHYR_BASE/VERSION. It appears to be confused by the folder named version in the current directory.

Well no having a VERSION file in an application is actually a supported feature too. I've never heard of such a thing on an os creating a `version` folder and no-one else has reported such an issue and there's plenty of mac users, can you explain how this `version` folder is created, and assume you have a case insensitive option enabled on your OS since the default filesystem in mac is case sensitive?

erlingrj, 2024-09-17:
> Well no having a VERSION file in an application is actually a supported feature too. I've never heard of such a thing on an os creating a `version` folder and no-one else has reported such an issue [...]

Hi and thanks for your reply. It is not the OS that creates the directory named `version` it is part of an application that I am porting to Zephyr and I do not want to change it. It appears the default file system on macOS is not case-sensitive, so that is probably why I am only seeing this problem when compiling on macOS. 

Would it be possibly for the build-system to differentiate between a _file_ named `version` and a _directory_ named `version`?

nordicjm, 2024-09-17:
Have had a look and no, this is not possible. The cmake code can be changed but the file is also sourced by Kconfig, so the only option is to rename your folder, sorry

erlingrj, 2024-09-17:
OK, thanks for your help!

closed 2024-09-17 by nordicjm