zephyrproject-rtos/zephyr #78529: add delta support in zephyr

RomainPelletant, 2024-09-17:
**Is your enhancement proposal related to a problem? Please describe.**
Firmware update over the air is sometime complicated according to limited bandwidth and power consumption.
Delta firmware update is a good candidate for that kind of context.

It requires:

- Delta API with "backends" selection to add other algorithms in the future
- Add delta backend for bsdiff+heatshrink
- Patch generation as script (bsdiff+heatshrink for the first implementation)
- Add support for Heatshrink as optional module

**Describe the solution you'd like**

- Add delta API in subsys
   - Support basic function 
      - delta_configure_memory to select memory inputs and outputs
      - delta_select_backend to select delta and compression algorithm
      - delta_patch_apply to process the delta

According to memory footprint, the first backend brings is bsdiff for delta (without bzip2 compression) and heatshrink for compression.

- Add python script to generate patch (according to the selected backend): [bsdiff+heatshrink python patcher](https://github.com/kickmaker/bsdiffhs) 
- Add compression library as module (like LZ4): [Heatshrink zephyr as external module](https://github.com/kickmaker/heatshrink)
- Produce delta patch on west build

**Describe alternatives you've considered**
Link to the presentation (OSS2024)

nordicjm, 2024-09-17:
Note: heatshrink has been rejected from being included in zephyr before, unless there is a big difference in the code as per the reason given for rejection then it is likely this will get the same rejection and reason to use an external module

RomainPelletant, 2024-09-17:
@nordicjm thanks for your message. You are right, it has been declined because this module was not linked directly to any Zephyr glue.
If a delta API is implemented in subsys, it could be interesting to  provide the heatshrink compression backend as module. Whatever it is an internal or external one.
We will provide more details here in few days with a (draft) PR

igiona, 2026-04-13:
> Hi [@igiona](https://github.com/igiona), thanks for your feedback!
> 
> On my side, patch generation is immediate so no i'm not facing this kind of latency.
> 
> Could you share your environment so I can try to reproduce the issue?

@ClovisCorde  Somewhat I can't repro the issue anymore...?!
Maybe it was related to some antivirus scans of my company IT stuff... 

Anyhow, I see some "interesting discrepancies" between the `detools` patches, and the one your script produces.

The patches were generated with the exact same input binaries...
The "extension" of the file, gives away the heatshrink parameters and the tool used for the generation:

```bash
-rw-r--r--  1 user users  404911 Apr 13 09:06 my-firmware.signed.dfuota.patch.bin.detools.w10.y4
-rw-r--r--  1 user users  486499 Apr 13 09:05 my-firmware.signed.dfuota.patch.bin.bsdiffhs.w10.l4
```

```bash
-rw-r--r--  1 user users  418686 Apr 13 09:06 my-firmware.signed.dfuota.patch.bin.detools.w14.y13
-rw-r--r--  1 user users  535070 Apr 13 09:05 my-firmware.signed.dfuota.patch.bin.bsdiffhs.w14.l13
``` 

As you can see, somehow your script performs remarkably worse compared to [detools](https://github.com/eerimoq/detools), do you have an explanation for this ?
Since both tools use `bsdiff` and `heatshrink`, I would actually expect pretty much the same patch file size,... Ideally, it would be even better to have some kind of "standardized" patch format such that one can use the one or the other tool ...but this is probably not going to happen :)